Koenkk/zigbee2mqtt #20879: Connection between z2m and HA : KO on restart

jerome6994, 2024-01-18:
### What happened?

Good morning
I come to you to ask for your help.
For quite a few months now I've had a problem that I haven't been able to resolve.
I will describe my installation to you first:
HAOS (in its latest version) installed in VM under promox (also in its latest version)
I have a VM under proxmox with Docker, Portainer installed on it. And I have functional mosquitto and z2m containers with conbee 2
[image]

My issue is the following :
When I restart HA to validate a configuration or even just like that everything goes well except on the HA <-> Mosquitto <-> z2m connection

To have my functional entities in HA I have to restart z2m each time.

We are looking at the subject with several friends. There are at least 2 of us in my group who have the same problem and 2 others who don't have the problem. We compare the configurations but we don't see anything special!

We are a little desperate on this subject.

HA replied to me that he had not touched the MQTT integration so that it did not come from them but from you!
I don't know how to say it, I don't have the skills for this, I'm just asking you for a little help.

Thank you in advance

Jerome

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

 0x26580700

### Adapter

Conbee 2

### Setup

NUC / Proxmox / VM / Docker 

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13903036/log.txt)

jerome6994, 2024-01-20:
Good morning
I have just been testing my problem again for two days!
I refurbished a mosquitto (user/mdp) z2m instance with a new key to replace conbee 2. I used an SLZB-06 key in POE.
I always use a test HA instance where I have MQTT integration (in MQTT v5).
At the beginning, restarting HA went really well: that is to say that the z2m devices came back up really well (this lasted 24 hours~)
Then it gradually deteriorated, yes I did regular restarts to test the thing.
Today, I no longer have any router that does not work (HUE bulb, US sockets, blitzwlf sockets) but neither does my Xiaomi SSM-U02 switch. The only devices that have resisted for the moment are Aqara MCCGQ11LM door sensors where the states are there and correct and above all alive.

On each device, I put (retain ok, 60s retain, optimistic ok)

I don't know what is happening in z2m to arrive at this situation but I have the impression that it is lost somewhere, that its cache is too large (I have 200 devices) anyway I let the experts that you do it are because it is beyond my skills

Thank you all for your help with this bug.

JC38, 2024-01-21:
Hello,

I think I have something similar since version 1.35.x.
I have 3 Zigbee2Mqtt running as a service on different computers (2 Windows, 1 Linux). Controllers are different (Zigate V1, Popp stick, SLZB-06).
HA is only running the Mqtt integration, the broker is running on another machine.
I am using 2 Smart Home software with z2m but only HA has a problem.
The problem : after updating /patching HAOS (this requiring a restart), some Zigbee devices are displayed in HA as "unavalaible". Outside HA, these devices are still working, can be managed from Z2M UI or my other Smarthome software, but to get them back into HA, I have to go into the Z2M web UI, Settings / Tools / Restart Zigbee2MQTT and then they are back into HA.

JC38, 2024-01-21:
The minimal action to reproduce the problem (getting the Zigbee devices disabled into HA)  is to Restart HA with the Quick reload option

droans, 2024-01-23:
I've had this issue before. If HA restarted from Docker, it occasionally would not see Z2M devices but it worked fine if I just performed a simple restart inside HA. 

Unfortunately, the only solution I could find was to add Home Assistant as a dependency for Z2M in my docker-compose.yaml file.

jerome6994, 2024-01-23:
good evening
so sometimes we still have to do a complete restart to reload the entire HA configuration with the packages etc... 
 I am under HAOS in VM in proxmox and not in docker. I simply have z2m as an independent docker on another docker VM, it is not an internal addon to HA.

VOSsec, 2024-02-28:
> I've had this issue before. If HA restarted from Docker, it occasionally would not see Z2M devices but it worked fine if I just performed a simple restart inside HA.
> 
> Unfortunately, the only solution I could find was to add Home Assistant as a dependency for Z2M in my docker-compose.yaml file.

How did you do that? I have the same problem... whenever the entire server is restarted, Z2MQTT doesn't work for a short while, until I restart the Z2MQTT addon or container, after which it runs without any problems.

Can you please help me? @droans ☺️

JC38, 2024-02-28:
This is a strange behaviour from my point of view as everything regarding z2m and mqtt is running "outside" the HA appliance.
However, does someone can help for building a script that will replace these actions from the Z2M U.I 
Settings/tools/restart Zigbee2MQTT
I don't know what this Restart Zigbee2MQTT button does exactly because I can't see the z2m daemon being restarted when clicking on it.

droans, 2024-02-28:
```
zigbee2mqtt:
    image: koenkk/zigbee2mqtt:latest
    ...
    depends_on:
        homeassistant:
          condition: service_started
          restart: true
```

closed 2024-09-26 by github-actions[bot]

palves, 2024-11-28:
I see something similar.  

I have about 30 Aqara door/window sensor devices, and after a HA restart, their entities show state "unknown".
I can see that the devices are connected to the zigbee network, as I can see their status update on z2m's web interface. 
Restarting z2m "fixes" it.  

I have three z2m instances (for three separate networks/coordinators), each with their separate base_topic.  All via docker (one compose for HA, another for mosquito, and another for the three Z2Ms instances).

Is Z2M supposed to send status updates/refreshes to HA on HA restart?  Maybe HA is missing sending something on the MQTT bus?  Or is sending the wrong thing?  Nor really sure where/what to look for.  If someone guides me, I'll gladly dig further.