Koenkk/zigbee2mqtt #20882: Xiaomi ZNJLBL01LM stops responding to /set commands

andrejsoucek, 2024-01-19:
### What happened?

I've been using this Aqara roller shade companion E1 for one of my windows and the device sometimes stopped responding to z2m commands. Re-pairing always fixed the issue for another month or so. Knowing this I bought 3 more devices for the rest of the windows in the room. 

Now the situation is much worse, every day at least one of the devices stops responding. I have to re-pair the device to actually make it work wirelessly again.

My observing:
- the device works when it is linked to a router/coordinator in the network map
- after some time the device gets linked to something else, the link is not shown in the map and the device cannot be controlled wirelessly
- weird thing - it seems that using the physical buttons on the device WILL update the exposed values in the Z2M (e. g. motor running, position) but the commands sent from Z2M to the device will timeout 
```
Publish 'get' 'position' to 'FRIENDLY_NAME' failed: 'Error: Read 0x54ef4410009949c9/1 genAnalogOutput([85], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32725 - 1 - 52 - 13 - 1 after 10000ms)'
```

If the device didn't hop to another router, it would most likely work fine. :/ 

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

HA OS

andrejsoucek, 2024-01-20:
I discovered another error when I manually change position by the physical buttons:
024-01-20 07:59:20EventBus error 'OnEvent/deviceMessage': Read 0x54ef4410009937e7/1 genAnalogOutput(["presentValue"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 60244 - 1 - 102 - 13 - 1 after 10000ms)

andrejsoucek, 2024-01-29:
Looks like the issue got fixed after updating all devices to the latest firmware (OTA).

closed 2024-01-29 by andrejsoucek

zengateway, 2024-07-25:
> Looks like the issue got fixed after updating all devices to the latest firmware (OTA).
Hey! I got exactly same problem, is your

> ### What happened?
>
> I've been using this Aqara roller shade companion E1 for one of my windows and the device sometimes stopped responding to z2m commands. Re-pairing always fixed the issue for [...]

Hey! I got exactly same problem, but all of my devices already on the latest firmware. Are you still get this problem?

andrejsoucek, 2024-07-25:
It finally got fixed after updating firmware in the coordinator and switching to ember driver.