Koenkk/zigbee2mqtt #20886: LightSolutions Dimmer-Switch-ZB3.0

agoberg2, 2024-01-19:
### Link

https://www.proshop.no/Lamper/Light-Solutions-Zigbee-Dimmer-For-Wire-White/3137309

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0xf4b3b1fffef932a3","nwkAddr":40821,"manufId":4107,"manufName":"Light Solutions","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"powerSource":18,"stackVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20230318","swBuildId":"1.00","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1705662966861,"defaultSendRequestWhen":"immediate"}

### Comments

Comes up as Unsupported and no Exposes are available.

### External defintion

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['Dimmer-Switch-ZB3.0'],
    model: 'Dimmer-Switch-ZB3.0',
    vendor: 'Light Solutions',
    description: 'Automatically generated definition',
    extend: [onOff({powerOnBehavior: false})],
};

module.exports = definition;
```

agoberg2, 2024-01-19:
I got it working with this external definition:

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Dimmer-Switch-ZB3.0'],
    model: 'Dimmer-Switch-ZB3.0', // Update this with the real model of the device (written on the device itself or product page)
    vendor: 'Light Solutions', // Update this with the real vendor of the device (written on the device itself or product page)
    description: 'Wire Dimmer', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },

};

module.exports = definition;`

Koenkk, 2024-01-20:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-20 by Koenkk

esbenr, 2024-01-21:
I just got this other dimmer from LightSolutions. The problem is that it has the exact same modelID, so this converter will match those and it doesent work.
https://www.proshop.no/Lamper/Light-Solutions-WiFi-Rotating-Dimmer-for-LK-250W/3137312

Status is updated when toggled from the switch, but there is no dimming and it can only be turned off via Z2M, not on.

How do we handle models that have the same manufacturer and model ID, but different features?

I.e. my PR for adding the above dimmer wouldbe exactly the same, except more "extend"s.

esbenr, 2024-02-03:
I see that this has not been released and now my switch is detected as this despite being a totally different model.
The proposed definition is way too broad and conflicts with all devices, with the foot print 'Dimmer-Switch-ZB3.0'.

@Koenkk Please be aware that the propesed (and now released) definition is the unedited generic definition autogenerated by Z2M. It is not complete for this device at all and is conflicting with others.

Koenkk, 2024-02-03:
@esbenr only those with manufaturer name `Light Solutions` are matched.

garret, 2024-02-03:
Is it normal that this dimmer with zigbee2mqtt can only be controlled the on/off state while with the Philips hue app it can be controlled also the dimmer part? 
I am using zigbee2mqtt on a raspberry pi 3 with a conbee stick.

garret, 2024-02-03:
BIG EDIT: sorry re-read again the full thread and had to rewrite my comment. 

I just bought many dimmers as the ones linked by @esbenr These dimmers look different from the one mentioned by @agoberg2 
Despite the fact that I managed to join one device temporarily today after updating to the latest branch, then forced its removal and now I cannot re-add it again to zigbee2mqtt (I tried factory resetting the switch, put it in pairing mode again and restarting zigbee2mqtt but with no luck...), it seems that the device is wrongly recognized anyways by zigbee2mqtt as it would only show the on/off states while it should be possible also to use the dimmer function. 

Sorry if I ask but is it something planned or easily achievable? Otherwise I should return them back asap to the shop while I can. 

Just for information, I am using mosquitto/zigbee2mqtt/homeassistant on a raspberry pi 3 and conbee as zigbee antenna stick.

esbenr, 2024-02-04:
@Koenkk The one I link to also have manufacturer name Light Solution. As do many other variations of this switch with different foot print. How can you papprove a PR with the generic autogenerated example definition only identifying on "Light Solutions" They have a whole lineup of different switches.

Now all my dimmers are identified as the one @agoberg2 has, but it's a comepletely different model. So the definition of @agoberg2 's is way too broad.

Second of all the definiseon propesed by @agoberg2 is incomplete as it only exposes an on/off switch and the switch is a dimmer. It's even defined as a dimmer by "Dimmer-Switch-ZB3.0".

I suggest you revert the PR until @agoberg2 has made a proper definition for his dimmer. I'm currently trying to make one for the type I'm using, but it's difficult now when it's detected by mistake by this definition.

Koenkk, 2024-02-04:
> is incomplete as it only exposes an on/off switch and the switch is a dimmer. It's even defined as a dimmer by "Dimmer-Switch-ZB3.0".

This has already been fixed in the dev branch, besides that are there any more issues?

garret, 2024-02-04:
Thank you very much @Koenkk, I updated to the dev branch and now the switch is recognized in its main functions (on/off and dimmer). 

However, the device I and @esbenr have is recognized by Z2M as **3137308/3137309**. This is wrong as the model number of our switch is **3004482** (at least according to the [website](https://www.proshop.dk/Lamper/Light-Solutions-ZigBee-drejelysdaemper-til-LK-250W/3004482) where we both bought the switch). 

[image]

This is a picture of the light switch **3004482**
[image]

Which is different from the switches **3137308** or **3137309**.

Koenkk, 2024-02-05:
@garret I've renamed it to `3004482/3137308/3137309`

codedesperate, 2024-05-03:
Hi @Koenkk,
I have the exact same device as garrett. The rotary type dimmer. I've just switched to the Dev branch, but for me it is still being detected as the wrong device. 

[image]

Do you know what the issue is here, or how I can help to fix it? 

@garret and @esbenr, did you run into this issue with this specific device? 
https://github.com/Koenkk/zigbee2mqtt/issues/21586

garret, 2024-05-04:
@codedesperate I see the device as you in my zigbee2mqtt. I don't care if the picture is wrong or the model as well. For the moment, I am grateful that it just works as to me I have been quite disappointed in how the dimmer reacts (very slow) and also the integration is not the best (I am sure is not the fault of zigbee2mqtt). For instance, The turn on/off is very slow from home assistant or also the updating of the status in general. However, that was the only type of dimmer I could find to fit Danish households and I guess I have just to be glad it somehow functions.

codedesperate, 2024-05-05:
@garret Yeah the picture is also not really important to me. 
My issue is that on on/off requests there is a 30second window where the device won't react to any commands, because it gets a timeout error on every command send to it. I can't really use zigbee2mqtt reliably with these. Sometimes the command works, sometimes it doesn't. Wife is not too happy with this :P 

But for a product like this, only designed to be used in Danish homes, I don't think there is hope it will be supported. Even though I've been trying to avoid it, I'll buy a philips hue bridge and link it to home assistant. That seems to be the only way to make it work 100%.

exetico, 2024-05-09:
But, @codedesperate ; It would be great to have support. I mean, it's produced in China, and I bet most of the logics behind are common for other products, as mentioned in #21586: https://github.com/Koenkk/zigbee2mqtt/issues/21586#issuecomment-2021278935

Link to product:
https://www.alibaba.com/product-detail/DK-220V-240V-250W-Fuga-Standard_1600183409775.html