Koenkk/zigbee2mqtt #20888: [New device support]: _TZE204_ztqnh5cg

petepal, 2024-01-19:
### Link

https://aliexpress.com/item/1005006344138708.html?spm=a2g0o.order_list.order_list_main.23.3d991c24DUN406&gatewayAdapt=glo2pol

### Database entry

{"id":22,"type":"Router","ieeeAddr":"0xa4c1389acca76e05","nwkAddr":9012,"manufId":4417,"manufName":"_TZE204_ztqnh5cg","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1705667020052}

### Comments

Please, add support for this device.

### External defintion

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_ztqnh5cg',
    description: 'Automatically generated definition',
    extend: [],
};

module.exports = definition;
```

petepal, 2024-01-19:
Here are some entries from the log:

```
Debug 2024-01-19 15:49:06Received Zigbee message from '0xa4c1389acca76e05', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,111],"type":"Buffer"},"datatype":2,"dp":9}],"seq":29184}' from endpoint 1 with groupID 0
Debug 2024-01-19 15:49:06No converter available for 'TS0601' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,111],"type":"Buffer"},"datatype":2,"dp":9}],"seq":29184}'
Debug 2024-01-19 15:49:07Received Zigbee message from '0xa4c1389acca76e05', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,122],"type":"Buffer"},"datatype":2,"dp":9}],"seq":29440}' from endpoint 1 with groupID 0
Debug 2024-01-19 15:49:07No converter available for 'TS0601' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,122],"type":"Buffer"},"datatype":2,"dp":9}],"seq":29440}'
Debug 2024-01-19 15:49:09Received Zigbee message from '0xa4c1389acca76e05', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,123],"type":"Buffer"},"datatype":2,"dp":9}],"seq":29696}' from endpoint 1 with groupID 0
Debug 2024-01-19 15:49:09No converter available for 'TS0601' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,123],"type":"Buffer"},"datatype":2,"dp":9}],"seq":29696}'
Debug 2024-01-19 15:49:11Received Zigbee message from '0xa4c1389acca76e05', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,112],"type":"Buffer"},"datatype":2,"dp":9}],"seq":29952}' from endpoint 1 with groupID 0
Debug 2024-01-19 15:49:11No converter available for 'TS0601' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,112],"type":"Buffer"},"datatype":2,"dp":9}],"seq":29952}'
Debug 2024-01-19 15:49:12Received Zigbee message from '0xa4c1389acca76e05', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,117],"type":"Buffer"},"datatype":2,"dp":9}],"seq":30208}' from endpoint 1 with groupID 0
Debug 2024-01-19 15:49:12No converter available for 'TS0601' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,117],"type":"Buffer"},"datatype":2,"dp":9}],"seq":30208}'
Debug 2024-01-19 15:49:14Received Zigbee message from '0xa4c1389acca76e05', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,201],"type":"Buffer"},"datatype":2,"dp":9}],"seq":30464}' from endpoint 1 with groupID 0
Debug 2024-01-19 15:49:14No converter available for 'TS0601' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"dpValues":[{"data":{"data":[0,0,0,201],"type":"Buffer"},"datatype":2,"dp":9}],"seq":30464}'
```

Koenkk, 2024-01-24:
Would you mind creating a PR for this?

ademobile, 2024-01-26:
For mine, I found changing [1, 'presence', customTrueFalseConverter] to [1, 'presence', tuya.valueConverter.trueFalse1] gave me presence with Ture or False.

petepal, 2024-01-26:
@Koenkk, [done](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6960)! Changed the file according to what @ademobile said and it works like a charm.

Koenkk, 2024-01-27:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6960

closed 2024-01-27 by Koenkk

arb2040, 2024-02-02:
Hi! I've just got this device from Aliexpress and just came across this post.  Optimistically, I copied  petepal's  ts-file and thought I finally could make this sensor work.  

But, z2m did not even start!  
Can this be explained by my use of the docker-image (koenkk/zigbee2mqtt:latest) ?

petepal, 2024-02-02:
> Hi! I've just got this device from Aliexpress and just came across this post. Optimistically, I copied petepal's ts-file and thought I finally could make this sensor work.
> 
> But, z2m did not even start! Can this be explained by my use of the docker-image (koenkk/zigbee2mqtt:latest) e.g. older version?

arb2040, it should work out of the box, as it was added as `ZY-M100-S_2`'s whitelabel. Remove the custom converter, update zigbee2mqtt and it _should_ work.

arb2040, 2024-02-02:
petepal, thank you for your response!

I've removed the converter reference from configuration.yaml and restarted the z2m container.

This resulted in a bombardement of these messages:
"Zigbee2MQTT:warn  2024-02-02 19:40:17: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_ztqnh5cg'
Zigbee2MQTT:warn  2024-02-02 19:40:17: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html"

Is there a chance that the docker image is not up to the latest version of z2m?

I'm far from experienced with z2m and struggle to see what else might be the problem.

bumperboy, 2024-02-09:
Did you manage to have your device working properly?

I also have a _TZE204_ztqnh5cg and it was not detected supported by Zigbee2MQTT (1.35.3-dev commit: a5a87a79)

After using some external converters, I managed to add the device but not all the features (e.g. presence) are working correctly.

kamashkin, 2024-02-09:
> Did you manage to have your device working properly?
> 
> I also have a _TZE204_ztqnh5cg and it was not detected supported by Zigbee2MQTT (1.35.3-dev commit: [a5a87a7](https://github.com/Koenkk/zigbee2mqtt/commit/a5a87a79a80aaa261a34a8fb541f47347c4ca729))
> 
> After using some external converters, I managed to add the device but not all the features (e.g. presence) are working correctly.

this works for me

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const customTrueFalseConverter = {
    key: ['presence'],
    convertSet: (entity, key, value, meta) => {
      // Your conversion logic
      const presence = value === 1 ? true : false;
      return {state: {presence}};
    },
    convertGet: (entity, key, meta) => {
      // Convert the state value from Home Assistant to the appropriate Zigbee value
      const zigbeeValue = meta.state.presence ? 1 : 0;
      return {state: zigbeeValue};
    },
  };

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_ztqnh5cg']),
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Human Presence Sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        e.illuminance_lux(), e.presence(),
        e.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        e.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        e.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        e.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        e.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        e.numeric('fading_time', ea.STATE_SET).withValueMin(0.5).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'presence', tuya.valueConverter.trueFalse1],
            [9, 'target_distance', tuya.valueConverter.divideBy100],
            [104, 'illuminance_lux', tuya.valueConverter.raw],
            [2, 'radar_sensitivity', tuya.valueConverter.raw],
            [4, 'maximum_range', tuya.valueConverter.divideBy100],
            [3, 'minimum_range', tuya.valueConverter.divideBy100],
            [102, 'fading_time', tuya.valueConverter.divideBy10],
            [101, 'detection_delay', tuya.valueConverter.divideBy10],
        ],
    },
};

module.exports = definition;
```