Koenkk/zigbee2mqtt #20894: TuYa TS011F_plug_1 Firmware update 192 detected but does not finish process

Diegocampy, 2024-01-19:
### What happened?

I receive an update notification (on two of my sockets), I try to start the update but it doesn't finish, and at a certain point it remains in progress with an indicated time remaining n/a

### What did you expect to happen?

firmware update

### How to reproduce it (minimal and precise)

starting the update

### Zigbee2MQTT version

1.35.1-1 

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Zigbee_3.0_USB_Dongle_Plus_V2_20230525160753-if00

### Setup

home assistant in virtual machine with proxbox in minipc

Diegocampy, 2024-01-23:
Update: I'm managing to do updates in a very difficult way. I try to start the update, many, many, many times. Each time it stops after a while but by making several attempts I still manage to advance by a few 0.0% and so on, many times until I managed to finish it. Once it reaches 100% the procedure ends, even if I haven't seen any updates, other than the view of the new firmware in the info. It took me several days to manage to do one, don't be discouraged, I will have restarted the firmware update procedure I think 100-200 times.
  Now I'm doing the same with my second outlet.
Let me know if you succeeded and you too have this absurd situation

rvalitov, 2024-10-21:
I updated successfully from firmware 78 to 192. Just made a single click, no retries. My signal strength is around 170 LQI.
My info:
```
Zigbee2MQTT version 1.40.2
Coordinator type EmberZNet
Coordinator revision 7.4.4 [GA]
zigbee-herdsman-converters version 20.21.0
zigbee-herdsman version 2.1.3
```

THuebner29, 2024-10-22:
In the meantime, I have also been able to update my devices to the latest FW. But I don't know what the problem was. Maybe a newer Zigbee2Mqtt version or HA. However, the version number is still not displayed.

rvalitov, 2024-10-22:
Good. Perhaps this issue can be closed then? There's a dedicated issue #16421 that the firmware version is not displayed.

mikeg-de, 2025-01-04:
I am facing the same issue:

`z2m: Update of 'Kinderzimmer Steckdose Türe' failed (Update failed with reason: ABORT)`

Though, I am trying to push through by restarting the update process several times. It appears that the process continues where it aborted:

[image]

I am also trying to identify the remote file to reference it locally as I did when updating the IKEA Tradfri remotes that suffer from a similar issue.

https://github.com/Koenkk/zigbee2mqtt/issues/18904#issuecomment-1750197075

closed 2025-03-15 by github-actions[bot]