Koenkk/zigbee2mqtt #20895: AQARA T1 Power report is wrong or absent

jmcollin78, 2024-01-19:
### What happened?

Aqara T1 with last firmware update doesn't report the power of report a wrong power.
The installed firmware is:
[image]

For one on the switch, the power is always set even if the switch is off (and the switch is really off):
[image]

For one other the power is always 0 even if the switch is off:

[image]

Just after pairing, it works as expected, but after a while, this stop working and the issues are present.

The issue is very similar with https://github.com/Koenkk/zigbee2mqtt/issues/11480#issuecomment-1162896817 which is Closed, but still here on 1.35.1-1 and on Edge release.

This issue appears approximatively with the 1.33 release (not totally certain of the release because I lost the backup).

### What did you expect to happen?

The power should reflect the real power of the device connected.

### How to reproduce it (minimal and precise)

Switch on/off. Check power.

### Zigbee2MQTT version

Edge

### Adapter firmware version

(not found ??)

### Adapter

ezsp

### Setup

Pi 4

### Debug log

SWITCH ON:
Zigbee2MQTT:debug 2024-01-19 16:17:57: Received MQTT message on 'zigbee2mqtt/Radiateur chambre 2/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2024-01-19 16:17:57: Publishing 'set' 'state' to 'Radiateur chambre 2'

Zigbee2MQTT:info  2024-01-19 16:17:57: MQTT publish: topic 'zigbee2mqtt/Radiateur chambre 2', payload '{"consumption":0.06978899240493774,"current":0,"device_temperature":14,"energy":0.07,"last_seen":"2024-01-19T15:17:57.704Z","linkquality":176,"power":0,"power_outage_count":0,"power_outage_memory":false,"state":"OFF","switch_type":"toggle","update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":238.82}'

Zigbee2MQTT:info  2024-01-19 16:17:57: MQTT publish: topic 'zigbee2mqtt/Radiateur chambre 2', payload '{"consumption":0.06978899240493774,"current":0,"device_temperature":14,"energy":0.07,"last_seen":"2024-01-19T15:17:57.704Z","linkquality":176,"power":0,"power_outage_count":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":238.82}'

Zigbee2MQTT:debug 2024-01-19 16:17:57: Received Zigbee message from 'Radiateur chambre 2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0

Zigbee2MQTT:info  2024-01-19 16:17:57: MQTT publish: topic 'zigbee2mqtt/Radiateur chambre 2', payload '{"consumption":0.06978899240493774,"current":0,"device_temperature":14,"energy":0.07,"last_seen":"2024-01-19T15:17:57.740Z","linkquality":176,"power":0,"power_outage_count":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":238.82}'

Zigbee2MQTT:debug 2024-01-19 16:18:21: Received Zigbee message from 'Radiateur chambre 2', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":12}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-01-19 16:18:21: MQTT publish: topic 'zigbee2mqtt/Radiateur chambre 2', payload '{"consumption":0.06978899240493774,"current":0,"device_temperature":12,"energy":0.07,"last_seen":"2024-01-19T15:18:21.867Z","linkquality":176,"power":0,"power_outage_count":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","update":{"installed_version":23,"latest_version":23,"state":"idle"},"update_available":null,"voltage":238.82}'

jmcollin78, 2024-01-19:
Let me know if I can help in any way.

jmcollin78, 2024-02-03:
Hello after the "Current version: 1.35.2-1" the problem is still there.

rjaros87, 2024-02-29:
I've got the same problem. If I re-pair it helps for ~1-2 days and then the same like you mention it hangs.

rjaros87, 2024-03-07:
I discovered something :), I had a second relay with an older firmware version **21**, tested for 5 days without any issues. I've updated it to the latest version **23**, and after about 1.5 days, the relay stopped reporting. I haven't tested it with the latest firmware version 7.4+ for the coordinator and the test "ember" driver.

jmcollin78, 2024-03-08:
> I discovered something 💡 , I had a second relay with an older firmware version **21**, tested for 5 days without any issues. I've updated it to the latest version **23**, and after about 1.5 days, the relay stopped reporting. I haven't tested it with the latest firmware version 7.4+ for the coordinator and the test "ember" driver. Maybe I will test it on the weekend.

This is a fact I already expriment on my side. One AQARA T1 POWER was working as expected but after firmware update, the power didn't comes to HA after one or two days.
May be we should open an issue on Aqara forum if there is one ?

jmcollin78, 2024-03-08:
You can create a ticket here like I will do: https://static-resource.aqara.com/html/feedback.html

rjaros87, 2024-03-12:
Any response from Aqara support? I didn't receive any response :(

jmcollin78, 2024-03-12:
No response yet

jmcollin78, 2024-03-12:
I just look at my email and I got an answer from support:
```
Hi Jean-Marc,

Thank you for contacting us. 

I'm sorry to hear that you are having this problem. I have some questions for you. 

What region do you choose in the app? 

Could you please provide your Aqara user account? Not included password. 

Could you please send us a short video to show this behavior? 
 
Sincerely, 
Cecilia
```

rjaros87, 2024-03-12:
It would be nice if they provided OTA version `0.0.0_0021`, because this version worked for me 👍.
I've registered on their Aqara developer site and I've got only those information for region EU (without url to ota files):
```
{
  "code": 0,
  "requestId": "a9a7eb3e91c546dba00f8bca26649ba4.344420.17101976422032075",
  "message": "Success",
  "msgDetails": null,
  "result": [
    {
      "releaseTime": "1708655686000",
      "necessary": 0,
      "fileSize": 288146,
      "updateLog": "1.Fix known bugs",
      "state": 2,
      "firmwareVersion": "0.0.0_0030"
    },
    {
      "releaseTime": "1644466164000",
		1698329291
      "necessary": 0,
      "fileSize": 287666,
      "updateLog": "Fix the offline problem of device.",
      "state": 2,
      "firmwareVersion": "0.0.0_0023"
    },
    {
      "releaseTime": "1635488564000",
      "necessary": 0,
      "fileSize": 287046,
      "updateLog": "1. Device stability was increased.",
      "state": 2,
      "firmwareVersion": "0.0.0_0022"
    }
  ]
}
```

jmcollin78, 2024-03-12:
Do you find how to restore an old firmware (without the application and aqara hub) ?

rjaros87, 2024-03-12:
If we have the ota file, than we can use Z2M and override index.json file with "custom" firmwares :)

jmcollin78, 2024-03-13:
The last and quick answer of the support:
```
Hi Jean-Marc,

Unfortunately, the device is not compatible with Home Assistant for the time being. 
 
Sincerely, 
Cecilia
Aqara Technical Support 
```

Such a shame...

jmcollin78, 2024-03-14:
I try with the Ember adapter in Z2M. Same problem.

jmcollin78, 2024-03-18:
The answer of the support to the question "is it possible to flash the firmware of the Aqara devices": 

```
Hi Jean-Marc,

I'm sorry that there is no way to do this. 
 
Sincerely, 
Cecilia
Aqara Technical Support 
```

jmcollin78, 2024-04-02:
I tried with Z2M 1.36.1-1 and problem is still there.

jmcollin78, 2024-05-15:
The problem is fixed with Z2M 1.37.1-1 !

This is really a surprise because the release notes don't mention this but all my Aqara T1 now report the power.

Did you experience the same thing ?

rjaros87, 2024-05-15:
I will check that :)

rjaros87, 2024-05-17:
After 2 days, same behaviour, power reports 0W.
Zigbee2MQTT version [1.37.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.1) commit: [ea39d86(https://github.com/Koenkk/zigbee2mqtt/commit/ea39d86)
SkyConnect
 EZSP v13
 7.4.1.0 build 0

jmcollin78, 2024-05-22:
3 over 4 are fixed on my side. Nothing special on the one which don't report the power.
Very strange.

petravic, 2024-06-13:
This converter works with latest release and fixed the power not reporting problem for me.

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const lumi = require('zigbee-herdsman-converters/lib/lumi');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.n0agl1'],
    model: 'SSM-U01',
    vendor: 'Aqara',
    description: 'Single switch module T1 (with neutral)',
    extend: [],
    fromZigbee: [fz.on_off, fz.metering, fz.electrical_measurement, fz.device_temperature, lumi.fromZigbee.lumi_specific],
    exposes: [e.switch(), e.energy(), e.power(), e.device_temperature(), e.power_outage_memory(), e.power_outage_count(), e.switch_type(), e.voltage(), e.current()],
    toZigbee: [lumi.toZigbee.lumi_switch_type, tz.on_off, lumi.toZigbee.lumi_switch_power_outage_memory, lumi.toZigbee.lumi_led_disabled_night],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering', 'genDeviceTempCfg']);
        await reporting.onOff(endpoint);
        await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint);
        await reporting.activePower(endpoint, {min: 5, max: 600, change: 10});
        await reporting.deviceTemperature(endpoint);
        device.powerSource = 'Mains (single phase)';
        device.type = 'Router';
        device.save();
    }
};
```

rjaros87, 2024-06-13:
@petravic I believe that the final line of the converter should include module.exports = definition; :)
However, despite using the converter, the power, temperature, and consumption reports are still returning `null`. :/
What version of zigbee2mqtt and coordinator driver do you have?

jmcollin78, 2024-06-15:
Hello @petravic,

When trying your converter I have this issue in the logs:
```
[2024-06-15 11:26:18] error: 	z2m: Failed to load external converter file 'aqara-t1.js' (Cannot read properties of undefined (reading 'push'))
[2024-06-15 11:26:18] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
[2024-06-15 11:26:18] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
```

rjaros87, 2024-06-16:
@jmcollin78 I've re-paired the devices after uploading the converter, and it has been working for 3 days now. There is about a 3-minute delay in getting the `Power` value. You might be able to adjust the reporting settings to receive updates more frequently.

jmcollin78, 2024-06-16:
I will try that. Thank you

jmcollin78, 2024-06-17:
Is my configuration wrong ?
 in `config/zigbee2mqtt`:
file aqara-t1.js:
```
const {} = require("zigbee-herdsman-converters/lib/modernExtend");

const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const ota = require("zigbee-herdsman-converters/lib/ota");
const utils = require("zigbee-herdsman-converters/lib/utils");
const globalStore = require("zigbee-herdsman-converters/lib/store");
const lumi = require("zigbee-herdsman-converters/lib/lumi");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ["lumi.switch.n0agl1"],
    model: "SSM-U01",
    vendor: "Aqara",
    description: "Single switch module T1 (with neutral)",
    extend: [],
    fromZigbee: [fz.on_off, fz.metering, fz.electrical_measurement, fz.device_temperature, lumi.fromZigbee.lumi_specific],
    exposes: [
        e.switch(),
        e.energy(),
        e.power(),
        e.device_temperature(),
        e.power_outage_memory(),
        e.power_outage_count(),
        e.switch_type(),
        e.voltage(),
        e.current(),
    ],
    toZigbee: [lumi.toZigbee.lumi_switch_type, tz.on_off, lumi.toZigbee.lumi_switch_power_outage_memory, lumi.toZigbee.lumi_led_disabled_night],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff", "haElectricalMeasurement", "seMetering", "genDeviceTempCfg"]);
        await reporting.onOff(endpoint);
        await endpoint.read("haElectricalMeasurement", ["acPowerMultiplier", "acPowerDivisor"]);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint);
        await reporting.activePower(endpoint, { min: 5, max: 600, change: 10 });
        await reporting.deviceTemperature(endpoint);
        device.powerSource = "Mains (single phase)";
        device.type = "Router";
        device.save();
    },
};

module.exports = definition;
```

in `configuration.yaml` at the end of the file:
```
...
external_converters:
  - aqara-t1.js
```

jmcollin78, 2024-06-18:
Finally I get it working by clicking on the botu button here for each device :
[image]

jmcollin78, 2024-06-19:
How this fix will be integrated into the next release of Z2M ?
Is there any process to run to have this merged in the add-on directly ?

DzianisFrydliand, 2024-07-01:
I have the same question. I have Aqara DLKZMK11LM and power is not reported.

jmcollin78, 2025-01-02:
I sill experience the issue.

ddrag09, 2025-01-08:
I experience the same issue. 
I have tried both versions of the converters (with module.export) and I dont see any power or energy being reporting.

I intentionally changed a line in the converter so it would fail in order to verify that the converter is being picked up by the configuration but everything started as it should and I did not see anything in the logs for Z2M...  
Is there any way to verify that the converter file is being picked up?

maybe need to change the log level to debug?

jmcollin78, 2025-01-09:
This issue is still not fixed but I have a workaround with this converter:
```
const {} = require("zigbee-herdsman-converters/lib/modernExtend");

const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const ota = require("zigbee-herdsman-converters/lib/ota");
const utils = require("zigbee-herdsman-converters/lib/utils");
const globalStore = require("zigbee-herdsman-converters/lib/store");
const lumi = require("zigbee-herdsman-converters/lib/lumi");
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ["lumi.switch.n0agl1"],
    model: "SSM-U01",
    vendor: "Aqara",
    description: "Single switch module T1 (with neutral)",
    extend: [],
    fromZigbee: [fz.on_off, fz.metering, fz.electrical_measurement, fz.device_temperature, lumi.fromZigbee.lumi_specific],
    exposes: [
        e.switch(),
        e.energy(),
        e.power(),
        e.device_temperature(),
        e.power_outage_memory(),
        e.power_outage_count(),
        e.switch_type(),
        e.voltage(),
        e.current(),
    ],
    toZigbee: [lumi.toZigbee.lumi_switch_type, tz.on_off, lumi.toZigbee.lumi_switch_power_outage_memory, lumi.toZigbee.lumi_led_disabled_night],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ["genOnOff", "haElectricalMeasurement", "seMetering", "genDeviceTempCfg"]);
        await reporting.onOff(endpoint);
        await endpoint.read("haElectricalMeasurement", ["acPowerMultiplier", "acPowerDivisor"]);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.currentSummDelivered(endpoint);
        await reporting.activePower(endpoint, { min: 5, max: 600, change: 10 });
        await reporting.deviceTemperature(endpoint);
        device.powerSource = "Mains (single phase)";
        device.type = "Router";
        device.save();
    },
};

module.exports = definition;
```

After installation, you will have to click on those buttons to rediscover the device. Then the power works. Sometimes you will loose the power, and you just have to click on the buttons (don't known why):

[image]

Maybe one day we will have this converter directlty inserted into Zigbee2Mqtt.

jmcollin78, 2025-01-11:
Yes you have to declare it in the configuration.yaml like that:
```
...
external_converters:
  - aqara-t1.js
```
On my configuration.yaml, it is at the end of the file.

ddrag09, 2025-01-13:
yeah I got that line in the configuration.yaml and nothing is happening.

Thats why Im looking for a way to verify that the awara-t1.js file is being picked up correctly by Z2M configuration.

smartsentinelle, 2025-02-01:
Hi everybody,

Did you finally fix this issue. I have the same one and try to add en external converter as described by jmcollin78 but still not get power information.

jmcollin78, 2025-02-02:
Hello @smartsentinelle, the issue always come back so I decide to remove all Aqara in my home. I get bored of this issue which is not fixed many years after. I've got 6 aqara to sell if you are interested.
Nous switchs just work fine out of the box.

jmcollin78, 2025-04-06:
I stop using Aqara T1 and I switch to Nous which works out of the box.

closed 2025-04-06 by jmcollin78

rjaros87, 2025-04-06:
> I stop using Aqara T1 and I switch to Nous which works out of the box.

I've done the same, but @jmcollin78, if you have more sockets, plug them into each other, and then connect something that consumes more than 1000W :) Observe the difference on each switch :) However, Nous works well :)

jmcollin78, 2025-04-07:
> I've done the same, but [@jmcollin78](https://github.com/jmcollin78), if you have more sockets, plug them into each other, and then connect something that consumes more than 1000W :) Observe the [...]

What did you observe ?

rjaros87, 2025-04-08:
[image]

[image]

There are some variations in the measurements, but I believe they are still acceptable – see screenshots. For power below 200W, the variations were within ±3W.