Koenkk/zigbee2mqtt #20897: [New device support]:  Salus SQ610

Crissiii10, 2024-01-19:
### Link

https://saluscontrols.com/de/produkt/sq610-quantum-ultraflacher-raumthermostat/

### Database entry

{"friendly_name":"0x001e5e090904ce3b","supported":false} New device joined '0x001e5e090904ce3b' model SQ610 Device '0x001e5e090904ce3b' announced itself

### Comments

Is it possible to implement it?

### External defintion

```shell
Is it possible to implement it?
```

Crissii10, 2024-07-19:
Do you need more information?

closed 2025-01-07 by github-actions[bot]

mikkelke, 2026-11-02:
Salus SQ610RFNH thermostats and the KL08RF wiring centre now work in zigbee2mqtt without the Salus gateway, from zigbee2mqtt 2.14.3. Pairing guide: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/SQ610RFNH.md

I have run 7 thermostats and the wiring centre on it since mid October. If something does not work for you, open an issue and tag me.