Koenkk/zigbee2mqtt #20904: No status switch in HA after upgrade zigbee2mqtt

ikey-ru, 2024-01-20:
### What happened?

Updated zigbee2mqtt and lost the ability to control many relays
There are no required entities in the home assistant system. (No on-off status switch).
All: "This entity is no longer provided by the mqtt integration"
Updating the home asistant does not help.
For example relays SONOFF ZBMINI, TuYa WHD02, TuYa TS0121_plug and others.
I don't understand why breaking everything? 

[image]
[image]

### What did you expect to happen?

That nothing's gonna break

### How to reproduce it (minimal and precise)

Try opening the relay settings in HA

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20211222

### Adapter

zStack30x

### Setup

x86-64

closed 2024-01-20 by ikey-ru

HelloCore, 2024-01-26:
Hi @ikey-ru , I also found this problem, how did you fix it?

For me, I downgraded back to the old version 😭.

ikey-ru, 2024-01-26:
It was necessary to update the Home assistant.
Before that, most likely you will have to update the python.