Koenkk/zigbee2mqtt #20905: [New device support]: Revise the code for Xiaomi QBKG18LM (Aqara T1 double rocker without neutral)

cloudyngcloudy, 2024-01-20:
### Link

https://www.zigbee2mqtt.io/devices/QBKG18LM.html

### Database entry

n/a

### Comments

Sorry that I have wrongly referred the code for QBKG18LM as "using the existing code for QBKG19LM", but which is not true (as shown in the following issue:-  https://github.com/Koenkk/zigbee-herdsman-converters/issues/5993)

Instead, this is the code (as below) I used before to make QBKG18LM work properly.  Glad to have your update to revise the code in Z2M.  Thank you.  Sorry for the inconvenience caused.

***************

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.b2lacn01'], 
    model: 'QBKG18LM', 
    vendor: 'Xiaomi',
    description: 'Aqara T1 double rocker without neutral', 
    fromZigbee: [fz.on_off, fz.xiaomi_power, fz.xiaomi_multistate_action, fz.aqara_opple], 
    toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_switch_power_outage_memory,
            tz.xiaomi_led_disabled_night, tz.xiaomi_flip_indicator_light], 
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },

    exposes: [
            e.switch().withEndpoint('left'), e.switch().withEndpoint('right'),
            e.power().withAccess(ea.STATE), e.energy(), e.voltage(), e.flip_indicator_light(),
            e.power_outage_memory(), e.led_disabled_night(), e.device_temperature().withAccess(ea.STATE),
            e.action([
                'single_left', 'double_left', 'single_right', 'double_right']),
            e.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for left button')
                .withEndpoint('left'),
            e.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
                .withDescription('Decoupled mode for right button')
                .withEndpoint('right'),
    ],

};

module.exports = definition;

### External defintion

```shell
n/a
```

cloudyngcloudy, 2024-01-21:
After updating the external convertor, the log stated "Failed to load external converter file 'QBKG18LM.js' (xiaomi is not defined)".
Not sure how to define xiaomi in external convertor file, any idea?

Koenkk, 2024-01-22:
Add this to the top of the file: `const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');`

cloudyngcloudy, 2024-01-22:
it works thanks.

closed 2024-08-20 by github-actions[bot]