Koenkk/zigbee2mqtt #20912: Device SIN-4-FP-21_EQU stops reacting to orders after not receiving an OTA answer

twisterss, 2024-01-20:
### What happened?

I have 7 SIN-4-FP-21_EQU devices on my network. To keep a heater on, I send from HA every 7 minutes an order to switch the device ON for 15 minutes.
I works for some time, and after a variable delay (a few hours to a few days), the device stops listening to orders and ends up switching off. The device keeps sending its status and seems to work normally, but it doesn't react to new orders (I tried switching it on from the zigbee2mqtt UI as well and it did not work)

I have noticed in the logs at the time the device stops reacting that it asks for an OTA update and doesn't seem to get a reply.

I realize this might be hard to reproduce and I'm willing to troubleshoot myself, but any pointer would help :)

### What did you expect to happen?

The device should keep reacting to orders and the heater should stay on.

### How to reproduce it (minimal and precise)

Add a SIN-4-FP-21_EQU device to zigbee2mqtt, send a {"state": "ON", "on_time": 900} order every 7 minutes, wait a few hours and the device will end up reporting state OFF because it doesn't react to orders anymore.

### Zigbee2MQTT version

1.35.1 commit: 9eaaa0f

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

Docker container on a small x86-64 computer

### Debug log

Here is an extract of the logs showing the problem. The "ON" command at 2024-01-20 11:14:00 worked and the "ON" command at 2024-01-20 11:21:00 was ignored, resulting in the device switching off at 11:29.
```
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:07:00: Received MQTT message on 'zigbee2mqtt/0x187a3efffeaa9378/set' with data '{"state": "ON", "on_time": 900}'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:07:00: Publishing 'set' 'state' to '0x187a3efffeaa9378'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:08:12: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersi
on":770,"imageType":18689,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:08:12: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:08:12: Responded to OTA request of '0x187a3efffeaa9378' with 'NO_IMAGE_AVAILABLE'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:13:10: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersi
on":66562,"imageType":18705,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:13:10: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:13:10: Responded to OTA request of '0x187a3efffeaa9378' with 'NO_IMAGE_AVAILABLE'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:14:00: Received MQTT message on 'zigbee2mqtt/0x187a3efffeaa9378/set' with data '{"state": "ON", "on_time": 900}'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:14:00: Publishing 'set' 'state' to '0x187a3efffeaa9378'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:15:09: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersi
on":770,"imageType":18689,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:15:09: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:15:09: Responded to OTA request of '0x187a3efffeaa9378' with 'NO_IMAGE_AVAILABLE'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:08: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersi
on":66562,"imageType":18705,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:08: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:11: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersi
on":66562,"imageType":18705,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:11: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:14: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersi
on":66562,"imageType":18705,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:14: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:17: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":66562,"imageType":18705,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:17: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:20: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":66562,"imageType":18705,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:20:20: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:21:00: Received MQTT message on 'zigbee2mqtt/0x187a3efffeaa9378/set' with data '{"state": "ON", "on_time": 900}'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:21:00: Publishing 'set' 'state' to '0x187a3efffeaa9378'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:26:22: Received MQTT message on 'zigbee2mqtt/0x187a3efffeaa9378/set' with data '{"state": "ON", "on_time": 900}'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:26:22: Publishing 'set' 'state' to '0x187a3efffeaa9378'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:26:22: Received MQTT message on 'zigbee2mqtt/0x187a3efffeaa9378/set' with data '{"state": "ON", "on_time": 900}'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:26:22: Publishing 'set' 'state' to '0x187a3efffeaa9378'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:26:22: Received MQTT message on 'zigbee2mqtt/0x187a3efffeaa9378/set' with data '{"state": "ON", "on_time": 900}'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:26:22: Publishing 'set' 'state' to '0x187a3efffeaa9378'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:28:00: Received MQTT message on 'zigbee2mqtt/0x187a3efffeaa9378/set' with data '{"state": "ON", "on_time": 900}'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:28:00: Publishing 'set' 'state' to '0x187a3efffeaa9378'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:29:11: Received Zigbee message from '0x187a3efffeaa9378', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:29:11: Received Zigbee message from '0x187a3efffeaa9378', type 'attributeReport', cluster 'manuSpecificNodOnPilotWire', data '{"mode":3}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:30:21: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":66562,"imageType":18705,"manufacturerCode":4727}' from endpoint 1 with groupID 0
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:30:21: Device '0x187a3efffeaa9378' requested OTA
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-20 11:30:24: Received Zigbee message from '0x187a3efffeaa9378', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":66562,"imageType":18705,"manufacturerCode":4727}' from endpoint 1 with groupID 0
```

twisterss, 2024-01-23:
I'm not sure if it's the same issue because the symptom is different, but I got a crash of zigbee2mqtt with this error repeated (on the same "on with timed off" command):
```
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-23 00:49:00: Received MQTT message on 'zigbee2mqtt/0xf4b3b1fffee8235e/set' with data '{"state": "ON", "on_time": 900}'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-23 00:49:00: Publishing 'set' 'state' to '0xf4b3b1fffee8235e'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:error 2024-01-23 00:49:00: Publish 'set' 'state' to '0xf4b3b1fffee8235e' failed: 'Error: Command 0xf4b3b1fffee8235e/1 genOnOff.onWithTimedOff({"ctrlbits":0,"ontime":90
00,"offwaittime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":nu
ll,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:debug 2024-01-23 00:49:00: Error: Command 0xf4b3b1fffee8235e/1 genOnOff.onWithTimedOff({"ctrlbits":0,"ontime":9000,"offwaittime":0}, {"sendWhen":"immediate","timeout":
10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}
) failed (sendZclFrameToEndpointInternal error)
ESC[36mzigbee2mqtt      |ESC[0m     at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:471:19)
ESC[36mzigbee2mqtt      |ESC[0m     at processTicksAndRejections (node:internal/process/task_queues:95:5)
ESC[36mzigbee2mqtt      |ESC[0m     at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
ESC[36mzigbee2mqtt      |ESC[0m Zigbee2MQTT:info  2024-01-23 00:49:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xf4b3b1fffee8235e' failed: 'Error: Command 0xf4b3b1fffee8235e/1 genOnOff.onWithTimedOff({\"ctrlbits\":0,\"ontime\":9000,\"offwaittime\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (sendZclFrameToEndpointInternal error)'","meta":{"friendly_name":"0xf4b3b1fffee8235e"},"type":"zigbee_publish_error"}'

```

easyservices, 2024-01-28:
Hi there. I somehow have a similar issue with the same device however in my case I have all the power consumption info etc but I can't issue any command to it they are all rejected (timeout to be more precise). 

Do you know a working previous version of zm ? I would like to downgrade if it can solve my issue and as you have the same model you might know and tell me.

Thanks

twisterss, 2024-01-28:
Hi,
I have changed a lot of things (added/removed some devices, changed my automation) so I cannot be sure if the problem is linked to the z2m version. But I didn't see the issue with version 1.34.0.
I still see the issue sometimes. I've temporarily fixed it by adding automatic zigbee2mqtt restarts when the problem happens. I'm still trying to understand what happens.

easyservices, 2024-01-30:
Hi

Thanks for the feedback. The 1.34.1 is bugged and reversing to the 1.34.0 solved my issue.

Regards

jclebedeff, 2024-01-31:
Hi,
same issue here

Zigbee2MQTT version
1.35.1

Zbdongle E - Zigbee (EZSP) 7.3.1.0 build 176

grrosminet, 2024-02-02:
Devrait être résolu par 1.35.2

twisterss, 2024-02-02:
Really? Awesome ! I’ll test that asap. 
I didn’t find an obvious fix in the changelog. Is it one of the EZSP changes?

twisterss, 2024-03-10:
It seems fixed, closing.

closed 2024-03-10 by twisterss