Koenkk/zigbee2mqtt #20936: VES-ZB-WAL-012 Battery 200%

tobiasg021, 2024-01-22:
### What happened?

The battery is on 200% 
[image]
[image]

### What did you expect to happen?

100% or lower 

### How to reproduce it (minimal and precise)

Connect to Zigbee2mqtt 

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

Sonoff dongle P 

### Setup

Zigbee2mqtt as Home assistant addons on Intel Nuc

Koenkk, 2024-01-22:
@martyn-vesternet could you check this?

closed 2024-08-28 by github-actions[bot]

tobiasg021, 2025-11-03:
1.5 years later, the converter no longer works. When will the change be implemented in series production? See log.

z2m: Invalid external converter 'VES-ZB-WAL-012_as_button_controller.js' was ignored and renamed to prevent interference with Zigbee2MQTT. (Cannot read properties of undefined (reading 'options')) 

PS:  warning: 	z2m: External converters are not meant for long term usage, but for local testing after which a pull request should be created to add out-of-the-box support for the device