Koenkk/zigbee2mqtt #20937: TS0601 thermostat always heating, even when it’s closed

Jannomag, 2024-01-22:
### What happened?

Hi,
I have four TS0601 thermostat valves.
Two of them are working fine while the other two have an issue with closing the valve.
When they reach the temperature set point they’re "closing", I can hear them. But when their state is closed the radiator is still on and the valve isn’t closed completely. I need to set "force:close" in order to close the valve completely.

I already tried to set the PID setting to 1 on the thermostat, as I found out googling around.
I also found that some people removed the plastic ring inside the screw connector. I don’t have these at all.

Do you guys have any ideas how to fix this?

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

0x26720700

### Adapter

ConBee II

### Setup

Add-on on Home Assistant, HAOS via VirtualBox VM in Ubuntu Server 20.04

git-tail-d, 2024-01-24:
Hello,

some thermostat have nor enough power to close the valve completely. On some it can help to adjust the running time of the motor.
It is an issue of the thermostat and not from z2m.

Which TRV does you have exactly?

Jannomag, 2024-01-24:
It’s a TuYa _TZE200_cwnjrr72 TS0601_thermostat.

When I close it manually via Zigbee2MQTT interface it closes the valve completely, so the motor ain’t the problem

closed 2024-08-22 by github-actions[bot]

JacekKac, 2025-03-05:
any solutions to this? i have exactly the same problem