Koenkk/zigbee2mqtt #20939: IMOU ZigBee PIR Motion Sensor

array81, 2024-01-22:
### Link

https://www.imoulife.com/en/product/detail/imou-home-security-system

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0x84b4dbfffed5c855","nwkAddr":31366,"manufId":21845,"manufName":"MultIR","powerSource":"Battery","modelId":"ZP1-EN","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"ZP1-EN","manufacturerName":"MultIR","powerSource":3,"zclVersion":3,"appVersion":15,"hwVersion":1,"dateCode":"20220318","swBuildId":"IR100_v1.2.0"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffed1f24b","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":9}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffed1f24b","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":15,"hwVersion":1,"dateCode":"20220318","swBuildId":"IR100_v1.2.0","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1705931522443,"defaultSendRequestWhen":"immediate"}

### Comments

I'm trying to add support for an IMOU PIR motion sensor.
I tried adding the voltage level (since I understand that the Tuya sensor supports it) but I got NULL as the result. Also I guess you could add "Occupancy timeout" but I haven't figured out how.
This is the Tuya sensor which looks like the IMOU one: https://www.zigbee2mqtt.io/devices/TS0202_1.html
This is the image of the IMOU sensor: https://www.bpm-power.com/cdn-cgi/image/quality=90,fit=scale-down,f=auto/var/catalog/orig/3134641.jpg

If anyone has the skills to improve the extension I could do the tests.

### External defintion

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy= require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZP1-EN'],
    model: 'ZP1-EN',
    vendor: 'IMOU',
    description: 'IMOU ZigBee PIR Motion Sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1], // <-- added here all clusters reported from zigbee
    toZigbee: [], // <-- add here all clusters to send commands to zigbee
    exposes: [e.occupancy(), e.battery_low()], // <-- this will define which fields will be exposed in the definition message to configure a frontend (e.g. the Z2M frontend, Home Assistant, Domoticz)
};

module.exports = definition;
```

array81, 2024-03-02:
This code is better, now battery level works. Not voltage.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const legacy= require('zigbee-herdsman-converters/lib/legacy');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZP1-EN'],
    model: 'ZP1-EN',
    vendor: 'IMOU',
    description: 'IMOU ZigBee PIR Motion Sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1_with_timeout, fz.battery, fz.ignore_basic_report],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.linkquality(), e.battery(), e.battery_voltage()],
    configure: async (device, coordinatorEndpoint, logger) => {
       const endpoint = device.getEndpoint(1);
       await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
       await reporting.batteryPercentageRemaining(endpoint);
    }
};

module.exports = definition;
```

Koenkk, 2024-03-02:
Looks good, if battery % works, no need to care about voltage, could you make a pull request?

array81, 2024-03-03:
> Looks good, if battery % works, no need to care about voltage, could you make a pull request?

I'm not a expert so I need make a pull request on zigbee-herdsman-converters right?
I need think I need change https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/src/converters/fromZigbee.ts

Koenkk, 2024-03-03:
a new file has to be added in the devices folder called imou.ts with the contents of your converter

array81, 2024-03-09:
> a new file has to be added in the devices folder called imou.ts with the contents of your converter

Thanks, I have create a new convert for IMOU Siren.
It seems to work correctly, however it uses "warning" and to turn the siren on/off you need to use mqtt code.

https://github.com/Koenkk/zigbee2mqtt/issues/21716

array81, 2024-03-09:
I sent the "pull request" I hope it goes well

closed 2024-10-06 by github-actions[bot]