Koenkk/zigbee2mqtt #20948: [New device support]: TS004F Switch 4-Gang

sango74, 2024-01-22:
### Link

https://fr.aliexpress.com/item/1005006151543213.html?spm=a2g0o.order_list.order_list_main.23.69515e5blLr7QI&gatewayAdapt=glo2fra

### Database entry

{"id":25,"type":"EndDevice","ieeeAddr":"0xa4c138c35b135c1e","nwkAddr":30785,"manufId":4417,"manufName":"_TZ3000_0ht8dnxj","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[1,3,4,6,4096,0],"outClusterList":[3,4,5,6,8,4096,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_0ht8dnxj","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1705945987447,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
possible to add this device?
THANKS :)

2024-01-22 19:10:54Received Zigbee message from '0xa4c138c35b135c1e', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
2024-01-22 19:10:55Received Zigbee message from '0xa4c138c35b135c1e', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
2024-01-22 19:10:58Received Zigbee message from '0xa4c138c35b135c1e', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
2024-01-22 19:10:59Received Zigbee message from '0xa4c138c35b135c1e', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
2024-01-22 19:11:00Received Zigbee message from '0xa4c138c35b135c1e', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
2024-01-22 19:11:01Received Zigbee message from '0xa4c138c35b135c1e', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":51,"transtime":10}' from endpoint 1 with groupID 0
2024-01-22 19:11:06Received Zigbee message from '0xa4c138c35b135c1e', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":51,"transtime":10}' from endpoint 1 with groupID 0
2024-01-22 19:11:08Received Zigbee message from '0xa4c138c35b135c1e', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":500,"minimum":153,"stepmode":1,"stepsize":50,"transtime":10}' from endpoint 1 with groupID 0
2024-01-22 19:11:17Received Zigbee message from '0xa4c138c35b135c1e', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":51,"transtime":10}' from endpoint 1 with groupID 0
2024-01-22 19:11:21Received Zigbee message from '0xa4c138c35b135c1e', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":51,"transtime":10}' from endpoint 1 with groupID 0
2024-01-22 19:11:25Received Zigbee message from '0xa4c138c35b135c1e', type 'commandStep', cluster 'genLevelCtrl', data '{"stepmode":1,"stepsize":51,"transtime":10}' from endpoint 1 with groupID 0
2024-01-22 19:11:26Received Zigbee message from '0xa4c138c35b135c1e', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":500,"minimum":153,"stepmode":3,"stepsize":50,"transtime":10}' from endpoint 1 with groupID 0

### External defintion

```shell
const {batteryPercentage, onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS004F'],
    model: 'TS004F',
    vendor: '_TZ3000_0ht8dnxj',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), onOff({powerOnBehavior: false}), identify()],
};

module.exports = definition;
```

hnykda, 2024-01-25:
I'm in the same camp, can help!

hnykda, 2024-01-26:
Seems that it's already in, maybe two weeks ago? https://github.com/Koenkk/zigbee-herdsman-converters/pull/6888 Will try

hnykda, 2024-01-26:
Yes, it is.

Works on zigbee2mqtt edge  (just the  image is wrong)

[image]

I propose closing this issue

sango74, 2024-01-27:
Hello,
I'm new to manually adding devices.
Is it possible to have an explanatory tutorial?
My Zigbee2MQTT runs on a debian machine.

Good day,

hnykda, 2024-01-27:
You don't need to add manual converter. Someone has done that already, but it's not in the official stable released version of z2m yet. You can either wait or run your zigbee2mqtt from the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html )

closed 2024-08-26 by github-actions[bot]