Koenkk/zigbee2mqtt #20955: OTA check error with supported device (Tuya BSD29_1)

guaycuru, 2024-01-23:
### What happened?

I have three Tuya BSD29_1 Smart Plugs, which according to the [docs ](https://www.zigbee2mqtt.io/devices/BSD29_1.html) do support OTA updates. But when checking, I receive this error:

> Failed to check if update available for '0x70b3d52b6004e6af' (Failed to find endpoint which support OTA cluster)

### What did you expect to happen?

I expected the OTA check to succeed since it's a supported device

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Home Assistant SkyConnect

### Setup

Add-on on Home Assitant OS running on Raspberry Pi 3B

lzzsalvo, 2024-01-30:
same issue

HughODwyer90, 2024-02-01:
Same issue

lzzsalvo, 2024-02-03:
@Koenkk

Koenkk, 2024-02-20:
Fixed, although I don't know iff all versions support OTA. If it reports about missing OTA check https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

guaycuru, 2024-02-27:
Cool, thank you! Now it seems that `zigbee2mqtt` needs to `zigbee-herdsman-converters` to 18.36.0 in `package.json` right?

Koenkk, 2024-02-27:
yes, but just update to the latest dev branch

guaycuru, 2024-03-01:
Oh but I'm using Home Assistant. Not sure how I would update Zigbee2MQTT to the latest dev branch

HughODwyer90, 2024-03-01:
> Oh but I'm using Home Assistant. Not sure how I would update Zigbee2MQTT to the latest dev branch

You need to use ZigBee2Mqtt edge version in Home Assistant I believe.

guaycuru, 2024-03-02:
Ok so today a new Zigbee2MQTT showed up and I updated it.
Now I'm on `zigbee_herdsman_converters_version 18.42.0`.
Clicking on "Check for new Updates" does not trigger the original error message, but instead "Device didn't respond to OTA request".
All Tuya BSD29_1 devices display this behavior.
I tried restarting the devices and Zigbee2MQTT but to no avail.
Should I open a separate issue for this?

Koenkk, 2024-03-02:
Then I would expect this particular version does not support OTA updates.

charsiufan, 2024-03-03:
> Ok so today a new Zigbee2MQTT showed up and I updated it. Now I'm on `zigbee_herdsman_converters_version 18.42.0`. Clicking on "Check for new Updates" does not trigger the original error message, [...]

I have a similar issue.

Arcaios, 2024-03-10:
Same issue... any solution?

MrSmithCZ, 2024-06-08:
Same issue. Hopefully some help will come soon.

davidclaverie, 2024-06-28:
Same issue. Dont have solution yet?

daubman, 2024-09-26:
Found this when experiencing the same issue with two different Tuya BSD29_1 plugs - is there a way to debug further that would be helpful? Are there details that would explain "this particular version" beyond "BSD29_1"?

closed 2025-01-07 by github-actions[bot]