Koenkk/zigbee2mqtt #20956: Xiaomi SRTS-A01 - invalid sensor_temp UI behaviour

and7ey, 2024-01-23:
### What happened?

[image]
The Sensor temp is displayed as number, but when you try to change its value, its never changed (remains `null` always) - when changed thru interface, thru mqtt or with set number. 

Actually, local_temperature is changed. 

### What did you expect to happen?

Sensor temp should be either changed or not displayed as slider & text input fields.
The doc should be updated that this field is never updated.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20210708

### Adapter

Sonoff dongle

### Setup

HAOS 2024.1.5

### Debug log

Debug 2024-01-23 14:34:50Received MQTT message on 'zigbee2mqtt/thermostat_nursery/set/sensor_temp' with data '25.3'
Debug 2024-01-23 14:34:50Publishing 'set' 'sensor_temp' to 'thermostat_nursery'
Debug 2024-01-23 14:34:54Received Zigbee message from 'thermostat_nursery', type 'attributeReport', cluster 'aqaraOpple', data '{"638":1,"640":1}' from endpoint 1 with groupID 0
Debug 2024-01-23 14:34:54zigbee-herdsman-converters:aqara_trv: Unhandled key 640 = 1
Info 2024-01-23 14:34:54MQTT publish: topic 'zigbee2mqtt/thermostat_nursery', payload '{"away_preset_temperature":"15.0","battery":100,"calibrate":null,"calibrated":true,"child_lock":"LOCK","device_temperature":25,"internal_heating_setpoint":21,"linkquality":87,"local_temperature":25.3,"occupied_heating_setpoint":21,"power_outage_count":1,"preset":"manual","schedule":"OFF","schedule_settings":"mon,sun|0:00,21.0|12:00,17.0|18:00,21.0|23:59,21.0","sensor":"external","sensor_temp":null,"setup":false,"system_mode":"heat","update":{"installed_version":2590,"latest_version":2590,"state":"idle"},"update_available":null,"valve_alarm":false,"valve_detection":"OFF","voltage":2900,"window_detection":"ON","window_open":false}'
Debug 2024-01-23 14:34:54Received Zigbee message from 'thermostat_nursery', type 'attributeReport', cluster 'aqaraOpple', data '{"65522":{"data":[170,113,13,36,18,162,5,65,10,0,0,21,141,0,1,157,27,152,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2024-01-23 14:34:54zigbee-herdsman-converters:aqara_trv: Unhandled key 65522 = �q $�A ���

DaN66O, 2024-01-23:
The same problem

and7ey, 2024-07-22:
+

almirus, 2024-10-16:
still here