Koenkk/zigbee2mqtt #20959: [Wrong device]: Moes BHT-002-GCLZB

pwd-github, 2024-01-23:
### Link

https://moeshouse.com/products/zigbee-programmable-thermostat-006

### Model

ZHT-006-GA-WH-MS

### Description

Zigbee Smart Thermostat Gas/Water Heater/Room Thermostat Digital Programmable 

### Vendor

Moes

### Picture (link)

https://moeshouse.com/cdn/shop/products/zigbee-smart-thermostat-gaswater-heater-room-thermostat-digital-programmable-855434.jpg

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0x003c84fffef2dac7","nwkAddr":39558,"manufId":4098,"manufName":"_TZE200_aoclfnxz","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0007","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_aoclfnxz","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1706022540698,"defaultSendRequestWhen":"immediate"}

### Notes

The difference between this one and the one it currently thinks it is, is that this doesn't have a system_mode cool, so it's throwing a lot of errors

uniquePWD, 2024-07-22:
Still valid

uniquePWD, 2024-11-11:
Still valid

uniquePWD, 2025-01-11:
This is still valid

uniquePWD, 2025-03-13:
Still valid

uniquePWD, 2025-05-13:
Still an issue

uniquePWD, 2025-05-15:
@Koenkk did I file this wrong? It's been over a year since I filed it and it's slipped through the cracks

Koenkk, 2025-05-16:
I've removed the `cool` `running_state`!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-05-16 by Koenkk

uniquePWD, 2025-05-19:
Thank you so much