Koenkk/zigbee2mqtt #20962: Ubisys H1: Setting local_temperature_offset fails due to wrong value being used

m0wlheld, 2024-01-23:
### What happened?

I have an [Ubisys H1](https://www.zigbee2mqtt.io/devices/H1.html) heating regulator, connected to Z2M via Slaesh's coordinator stick. Device's firmware version is [1.5.0](https://www.ubisys.de/en/support/firmware/change-logs-h1/) .

When trying to change the `local_temperature_offset` value via Z2M UI, the following error is raised:

```
Publish 'set' 'local_temperature_offset' to '0x001fee0000009838' failed: 'RangeError [ERR_OUT_OF_RANGE]: Write 0x001fee0000009838/1 hvacThermostat({"ubisysTemperatureOffset":450}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= -128 and <= 127. Received 450)'
```

The offset value entered in the UI was 4.5, but it was obviously multiplied by 100. Same happens when publishing an MQTT message directly or changing the attribute via slider in HA.

Also: Both `local_temperature_offset` and `remote_temperature_duration` are shown as "Unknown" in HA. 

### What did you expect to happen?

Value from Z2M UI or HA UI stored onto the device and reported properly. 

### How to reproduce it (minimal and precise)

Use Z2M UI, open the device's detail page and either move the slider or enter a value directly for "Local temperature offset"

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652RB stick

### Setup

plain dockerized, connected via Mosquitto to HA container

sjorge, 2024-01-23:
I have reached out to Ubisys, I based this on the draft tech reference I got from them.

sjorge, 2024-01-25:
I've not heard back on this yet, but scale 10 seems to work. I'll double check tonight if it matches with whats on the UI and open a PR.

closed 2024-01-25 by Koenkk