Koenkk/zigbee2mqtt #20963: [Feature request]: Ubisys H1 Firmware 1.5.1 is available

m0wlheld, 2024-01-23:
### Is your feature request related to a problem? Please describe

Ubisys H1 Heating Regulator firmware 1.5.1 is available for download and it seems to fix an important bug, introduced with 1.5.0 . However, it is not yet available by Z2M for OTA.

* [Firmware 1.5.1 Download](https://www.ubisys.de/wp-content/uploads/10F2-7B2D-0000-0001-01510430-m7b-h1.ota.zigbee)
* [Firmware Changelog](https://www.ubisys.de/en/support/firmware/change-logs-h1/)

### Describe the solution you'd like

Firmware 1.5.1 to become available for OTA update.

### Describe alternatives you've considered

Stay with 1.5.0

### Additional context

n/a

sjorge, 2024-01-24:
I was able to upgrade just fine?
[image]

What error are you getting?

m0wlheld, 2024-01-24:
No error, it's just not offered for update ("no update available"). Are you using -dev version of Z2M?

sjorge, 2024-01-24:
Yes, from a couple of days ago `1.35.1-dev commit: [0007db42](https://github.com/Koenkk/zigbee2mqtt/commit/0007db42)`

m0wlheld, 2024-01-24:
If so, this "feature request" should resolve itself with the next "stable" release, by start of February. I will wait.

closed 2024-01-24 by m0wlheld

sjorge, 2024-01-24:
Double check the debug log, it finds the image and it seems to have the latest one in the mqtt message too:

```
debug 2024-01-24 18:01:23: OTA: Is new image available for '0x001fee00000097b5' (H1), latest meta '{"hardwareVersionMin":0,"hardwareVersionMax":1,"fileVersion":22086704,"url":"http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B2D-0000-0001-01510430-m7b-h1.ota.zigbee"}'
```

So I think it should indeed just show up next release.