Koenkk/zigbee2mqtt #20964: [New device support]: QS-zigbee-S08-16A-RF

wanico31, 2024-01-23:
### Link

https://www.aihomestyle.com/products/wifi-switch-module-with-rf-receiver/

### Database entry

"ieeeAddr":"0xa4c138787fad91ac"

### Comments

Hello,
the Zigbee device is not supported in Home Assistante Zigbee2MQTT. 
Zigbee Model: TS000F / Zigbee Manufacturer: _TZ3000_dlhhrhs8
Thanks for your help.

### External defintion

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'],
    model: 'TS000F',
    vendor: '_TZ3000_dlhhrhs8',
    description: 'Automatically generated definition',
    extend: [onOff({powerOnBehavior: false})],
};

module.exports = definition;
```

wanico31, 2024-01-23:
I managed to make it work with the following lines added:
fromZigbee: ['genOnOff','state'], // <-- added here all clusters reported from zigbee
toZigbee: ['genOnOff'], // <-- add here all clusters to send commands to zigbee

Koenkk, 2024-01-25:
Can you pos the working converter?

wanico31, 2024-01-25:
here it is:
```
//ver: 2024-01-24
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'],
    model: 'TS000F',
    vendor: '_TZ3000_dlhhrhs8',
    description: 'My super switch!',
    extend: [onOff({powerOnBehavior: false})],
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

Koenkk, 2024-01-27:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-01-27 by Koenkk