Koenkk/zigbee2mqtt #20968: Xiaomi Aqara Knob H1 ZNXNKG02LM - incorrect and non-functional attributes in reports (not during actions)

uncle-fed, 2024-01-23:
### What happened?

This is **not** about messages which are being generated when the switch is interacted with (pressed / rotated). I deliberately did not want to add to the [currently open issue](https://github.com/Koenkk/zigbee2mqtt/issues/19995) that deals with `action`-related properties. The issues described below are about report messages that get generated automatically by the switch, roughly every 55 minutes (without touching it).

### What did you expect to happen?

I would be reasonable to expect that zigbee2mqtt messages do NOT contain the following attributes that are automatically reported every 55 minutes: 

- `device_temperature`
- `power_outage_count`
- `state`

Those attributes are not documented as being valid. So either the docs need to be updated or the behaviour changed. While I have no real knowledge about the `device_temperature` it seems to be most likely bogus because it always transmit the value of `25` 
 which just cannot be right. I have two devices in separate rooms with significantly different temperatures but both transmit 25, on top of that the real temperature is never above 21 degrees currently in the warmest room. Then again, I can see, as I have other Aqara devices (contact sensor, presence sensor, buttons etc. that they all seem to report bogus device temperatures. So I guess it is a more general issue that this particular device.

The `power_outage_count` _may_ be somehow valid, but it makes me doubt it as well. I purchased two switches together and they came in as new/sealed. One reports the value of `38`, the other one `5668`. I just find it very hard to believe that somebody disconnected the batteries over 5 thousand times on that new switch. Still, the value does seem to increase if batteries are replaced. There are two CR2032 batteries in this device and the counter seems to increase even only one of them is removed (while device keeps functioning).

`state` is most definitely the attribute that should never be reported in the `event` mode (not sure about the `command` mode). The the `state` never appears in payload when interacting with the button.  It only gets reported together with the above two attributes every 55 minutes as being 'OFF'.

See the debug log below that shows the moment when the report message comes in (without pressing any button).

### Zigbee2MQTT version

1.35.1 commit: 3c962042

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus-P

### Setup

Plain Z2M on Pi 4

### Debug log

```
Zigbee2MQTT:debug 2024-01-23 20:01:07: Received Zigbee message from 'home/switch/office1/desk', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,23,12,3,40,25,4,33,168,67,5,33,37,22,6,36,3,0,0,0,0,8,33,20,1,10,33,0,0,12,32,1,100,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-01-23 20:01:07: ZNXNKG02LM: Processed buffer into data {"1":3095,"3":25,"4":17320,"5":5669,"6":3,"8":276,"10":0,"12":1,"100":0}
Zigbee2MQTT:debug 2024-01-23 20:01:07: ZNXNKG02LM: unknown key 6 with value 3
Zigbee2MQTT:debug 2024-01-23 20:01:07: ZNXNKG02LM: Processed data into payload {"voltage":3095,"battery":100,"device_temperature":25,"power_outage_count":5668,"state":"OFF"}
Zigbee2MQTT:debug 2024-01-23 20:01:07: ZNXNKG02LM: Processed data into payload {"voltage":3095,"battery":100,"device_temperature":25,"power_outage_count":5668,"state":"OFF"}
Zigbee2MQTT:info  2024-01-23 20:01:07: MQTT publish: topic 'z2m/home/switch/office1/desk', payload '{"battery":100,"device_temperature":25,"last_seen":"2024-01-23T19:01:07.965Z","linkquality":91,"power_outage_count":5668,"state":"OFF","voltage":3095}'
2024-01-23T19:01:08.479Z zigbee-herdsman:controller:endpoint Read 0x287681fffeec5847/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-23T19:01:08.479Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x287681fffeec5847:4061/1 (0,0,1)
2024-01-23T19:01:08.480Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":4061,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":251,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,166,0,0,0]}}
2024-01-23T19:01:08.480Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,221,15,1,1,0,0,251,0,30,5,16,166,0,0,0,174]
2024-01-23T19:01:08.490Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2024-01-23T19:01:08.490Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2024-01-23T19:01:08.490Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2024-01-23T19:01:08.490Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2024-01-23T19:01:08.491Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-23T19:01:08.494Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,251,61]
2024-01-23T19:01:08.494Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,251,61]
2024-01-23T19:01:08.495Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,251] - 61
2024-01-23T19:01:08.495Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":251}
2024-01-23T19:01:08.495Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-23T19:01:08.504Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,221,15,0,80]
2024-01-23T19:01:08.504Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,221,15,0,80]
2024-01-23T19:01:08.504Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [221,15,0] - 80
2024-01-23T19:01:08.505Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":4061,"relaycount":0,"relaylist":[]}
2024-01-23T19:01:08.505Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-23T19:01:08.558Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,0,0,221,15,1,1,0,25,0,253,165,77,0,0,8,8,166,1,0,0,0,32,8,221,15,29,71]
2024-01-23T19:01:08.559Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,0,221,15,1,1,0,25,0,253,165,77,0,0,8,8,166,1,0,0,0,32,8,221,15,29,71]
2024-01-23T19:01:08.559Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,0,0,221,15,1,1,0,25,0,253,165,77,0,0,8,8,166,1,0,0,0,32,8,221,15,29] - 71
2024-01-23T19:01:08.559Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":4061,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":25,"securityuse":0,"timestamp":5088765,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[8,166,1,0,0,0,32,8]}}
2024-01-23T19:01:08.561Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":166,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":0,"status":0,"dataType":32,"attrData":8}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":4061,"endpoint":1,"linkquality":25,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-23T19:01:08.563Z zigbee-herdsman:controller:endpoint DefaultResponse 0x287681fffeec5847/1 0(1, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-23T19:01:08.564Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

WoRsTiG, 2024-03-02:
I've had exactly the same issue, i've got aqara d1's that work as intended. And now i've got 2 new aqara h1's with no response. So when actuating from z2m itself it reports back in mqtt, but physical interaction doesn't result in any z2m logs or mqtt events at all...

Here's the state info for my 2 switch H1 in the z2m state dump:

"0x54ef441000ae0a95": {
    "date_code": "11-10-2021",
    "definition": {
        "description": "Smart wall switch H1 EU (with neutral, double rocker)",
        "exposes": [
            {
                "endpoint": "left",
                "features": [
                    {
                        "access": 7,
                        "description": "On/off state of the switch",
                        "endpoint": "left",
                        "label": "State",
                        "name": "state",
                        "property": "state_left",
                        "type": "binary",
                        "value_off": "OFF",
                        "value_on": "ON",
                        "value_toggle": "TOGGLE"
                    }
                ],
                "type": "switch"
            },
            {
                "endpoint": "right",
                "features": [
                    {
                        "access": 7,
                        "description": "On/off state of the switch",
                        "endpoint": "right",
                        "label": "State",
                        "name": "state",
                        "property": "state_right",
                        "type": "binary",
                        "value_off": "OFF",
                        "value_on": "ON",
                        "value_toggle": "TOGGLE"
                    }
                ],
                "type": "switch"
            },
            {
                "access": 5,
                "category": "diagnostic",
                "description": "Instantaneous measured power",
                "label": "Power",
                "name": "power",
                "property": "power",
                "type": "numeric",
                "unit": "W"
            },
            {
                "access": 1,
                "description": "Sum of consumed energy",
                "label": "Energy",
                "name": "energy",
                "property": "energy",
                "type": "numeric",
                "unit": "kWh"
            },
            {
                "access": 7,
                "description": "Decoupled mode for left button",
                "endpoint": "left",
                "label": "Operation mode",
                "name": "operation_mode",
[... 198 lines cut ...]
            "scenes": []
        },
        "242": {
            "bindings": [],
            "clusters": {
                "input": [],
                "output": [
                    "greenPower"
                ]
            },
            "configured_reportings": [],
            "scenes": []
        }
    },
    "friendly_name": "Switch_Achterdeur",
    "ieee_address": "0x54ef441000ae0a95",
    "interview_completed": true,
    "interviewing": false,
    "manufacturer": "LUMI",
    "model_id": "lumi.switch.n2aeu1",
    "network_address": 57883,
    "power_source": "Mains (single phase)",
    "supported": true,
    "type": "Router"
}

thestraycat, 2024-06-11:
As the 'command' mode has now been detailed to control "binding" the Aqara ZNXNKG02LM to a end device has anyone managed to sucessfully bind it to a light say? 

I cant find any reference to this device and binding other than it seems to expose the mode and option to do so. And the manuals in paper form (no pdf that ive found) and in chinese!

Can anyone detail the steps to bind it to an end device?

closed 2025-01-07 by github-actions[bot]

thestraycat, 2025-02-01:
Still interested in this!

dimitry42, 2025-04-28:
This doesn't seem to be fixed.