Koenkk/zigbee2mqtt #20969: Zigbee messages are sent out of order

Flavien, 2024-01-23:
### What happened?

The Zigbee messages are sent by Z2M out of order from the order they are consumed from MQTT.

### What did you expect to happen?

See detailed logs below, the Zigbee messages should be sent in the same order as they are received from the MQTT broker. As you can see below, the messages got received in order `1`, `2`, `3`, `4` but the Zigbee messages were sent in order `1`, `3`, `4`, `2`.

Ideally, the order should be preserved.

### How to reproduce it (minimal and precise)

Here is an extract of my Zigbee2MQTT logs:

```
 1  2024-01-23T17:00:17.523629503Z Zigbee2MQTT:debug 2024-01-23 17:00:17: Received MQTT message on 'zigbee2mqtt/entrance/set' with data '{"scene_recall":1}'
 2  2024-01-23T17:00:17.525128068Z Zigbee2MQTT:debug 2024-01-23 17:00:17: Publishing 'set' 'scene_recall' to 'entrance'
 3  2024-01-23T17:00:17.772864147Z Zigbee2MQTT:info  2024-01-23 17:00:17: Successfully recalled group scene
 4  2024-01-23T17:00:17.779407543Z Zigbee2MQTT:info  2024-01-23 17:00:17: MQTT publish: topic 'zigbee2mqtt/entrance/ceiling', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"last_seen":"2024-01-23T15:08:24+00:00","linkquality":138,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":null}'

 5  2024-01-23T17:00:17.994246776Z Zigbee2MQTT:debug 2024-01-23 17:00:17: Received MQTT message on 'zigbee2mqtt/entrance/set' with data '{"scene_recall":2}'
 6  2024-01-23T17:00:17.995912782Z Zigbee2MQTT:debug 2024-01-23 17:00:17: Publishing 'set' 'scene_recall' to 'entrance'

 7  2024-01-23T17:00:18.874496782Z Zigbee2MQTT:debug 2024-01-23 17:00:18: Received MQTT message on 'zigbee2mqtt/entrance/set' with data '{"scene_recall":1}'
 8  2024-01-23T17:00:18.876433375Z Zigbee2MQTT:debug 2024-01-23 17:00:18: Publishing 'set' 'scene_recall' to 'entrance'
 9  2024-01-23T17:00:19.168387535Z Zigbee2MQTT:info  2024-01-23 17:00:19: Successfully recalled group scene
 10 2024-01-23T17:00:19.170650493Z Zigbee2MQTT:info  2024-01-23 17:00:19: MQTT publish: topic 'zigbee2mqtt/entrance/ceiling', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"last_seen":"2024-01-23T15:08:24+00:00","linkquality":138,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":null}'
 
 11 2024-01-23T17:00:19.557180937Z Zigbee2MQTT:debug 2024-01-23 17:00:19: Received MQTT message on 'zigbee2mqtt/entrance/set' with data '{"state":"OFF"}'
 12 2024-01-23T17:00:19.557249454Z Zigbee2MQTT:debug 2024-01-23 17:00:19: Publishing 'set' 'state' to 'entrance'
 13 2024-01-23T17:00:19.777748749Z Zigbee2MQTT:info  2024-01-23 17:00:19: MQTT publish: topic 'zigbee2mqtt/entrance/ceiling', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"last_seen":"2024-01-23T15:08:24+00:00","linkquality":138,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":null}'
 
 14 2024-01-23T17:00:20.399203034Z Zigbee2MQTT:info  2024-01-23 17:00:20: Successfully recalled group scene
 15 2024-01-23T17:00:20.402557453Z Zigbee2MQTT:info  2024-01-23 17:00:20: MQTT publish: topic 'zigbee2mqtt/entrance/ceiling', payload '{"brightness":77,"color":{"h":32,"hue":32,"s":82,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"last_seen":"2024-01-23T15:08:24+00:00","linkquality":138,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16785162,"latest_version":16785162,"state":"idle"},"update_available":null}'
```

- Lines `1 - 4`, we send a `{"scene_recall":1}` message to the group, the scene recall is correctly applied.

- Line `5 - 6` is where the problem is, we also send a `{"scene_recall":2}`, which is received and processed by Zigbee2MQTT as evidenced by this log entry, however the scene is no immediately recalled.

- Line `7 - 10`, before the scene 2 has the time to get recalled, we send another message with `{"scene_recall":1}` to the group. As you can see on line 9 and 10, the scene is successfully recalled (the light visible recalled the scene).

- Line `11 - 13`, we send `{"state":"OFF"}`, and at that point the light correctly turns off.

- The anomaly shows up on line `14 -15`, the scene recall from lines `5 - 6` is now being applied late, and the light turns back on.

Note this all happens very quickly (in the span of 3 seconds).

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Z2M on Docker on Raspberry Pi 4

Flavien, 2024-01-24:
Doesn't [this](https://github.com/Koenkk/zigbee-herdsman/blob/9b3dfc2b6865d6907a995ad97db44eba08266191/src/utils/queue.ts#L16-L40) have a bug, or am I misreading it @Koenkk / @mikekap ?

    public async execute<T>(func: () => Promise<T>, key: string | number = null): Promise<T> {
        const job : Job = {key, running: false, start: null};
        // Minor optimization/workaround: various tests like the idea that a job that is
        // immediately runnable is run without an event loop spin. This also helps with stack
        // traces in some cases, so avoid an `await` if we can help it.
        this.jobs.push(job);
        if (this.getNext() !== job) {
            await new Promise((resolve): void =>  {
                job.start = (): void => {
                    job.running = true;
                    resolve(null);
                };
                this.executeNext();
            });
        } else {
            job.running = true;
        }

        try {
            return await func();
        } finally {
            this.jobs.splice(this.jobs.indexOf(job), 1);
            this.executeNext();
        }
    }

The first job to come in will be set as `running = true`, and gets suspended at `await func()`. Then assuming it doesn't finish running before the next message is received through MQTT, this function will re-enter with the new job, which gets added to `this.jobs`, `getNext()` returns `null` so `job.running` is set to `true`, and `await func()` is executed, even though the first job hasn't finished yet?

mikekap, 2024-01-25:
If `getNext()` returns `null`, `getNext() !== job` will be true - so we await the promise to get a slot on the queue. The else clause doesn't get taken the second time.

closed 2024-08-23 by github-actions[bot]