Koenkk/zigbee2mqtt #20980: Tz.Converter and Fz.Converter cannot be used in Custom converter

Nailik, 2024-01-24:
### What happened?

Hi,

i tried creating a custom converter to test if it works correctly before opening a pull request.
However i need to access  `Tz.Converter` and `Fz.Converter` but it can't be access because 
`import {Definition, Fz, Tz, KeyValue}` from 'zigbee-herdsman-converters/lib/types';
results in `cannot use import statement outside a module`.

### What did you expect to happen?

Being able to import necessary definitions

### Zigbee2MQTT version

-

### Adapter firmware version

-

### Adapter

-

### Setup

-

Koenkk, 2024-01-25:
You don't need to use these in the external converter, (so remove `satisfies Tz.Converter` from it)

closed 2024-01-25 by Koenkk

Koenkk, 2024-01-26:
You don't need it in a `.js` file, `Tz.Converter` is just typing information for `.ts` files only. There is no downside of not using it.

Nailik, 2024-01-26:
Thanks looking at this https://github.com/Koenkk/zigbee-herdsman-converters/issues/5635 helped me.