Koenkk/zigbee2mqtt #20982: GLEDOPTO not updating state correctly when changed by external device

shawly, 2024-01-24:
### What happened?

I've bound my IKEA STYRBAR to my GLEDOPTO GL-C-007-2ID LED controllers RGB endpoint so I can turn it on and off.

But when I turn it on or off the state is not reflected on the device in Z2M or in MQTT.

Interestingly, a log message with the change is produced and the LED controllers state does change to on, but the RGB endpoints (ID 11) genOnOff cluster stays off.

Debug log message:
```
Debug 2024-01-24 15:42:43Received Zigbee message from 'Living room LED Controller', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
```

And the controllers state:
```json
{
    "brightness": 162,
    "brightness_rgb": 49,
    "brightness_white": 162,
    "color": {
        "h": 24,
        "hue": 24,
        "s": 100,
        "saturation": 100,
        "x": 0.555,
        "y": 0.422
    },
    "color_mode": "xy",
    "color_mode_rgb": "xy",
    "color_rgb": {
        "hue": 24,
        "saturation": 100,
        "x": 0.555,
        "y": 0.422
    },
    "color_temp": 532,
    "color_temp_rgb": 532,
    "last_seen": "2024-01-24T15:42:43+01:00",
    "linkquality": 72,
    "state": "ON",
    "state_rgb": "OFF",
    "state_white": "OFF"
}
```

### What did you expect to happen?

That the state change is reflected in the state_rgb property

### How to reproduce it (minimal and precise)

Bind STYRBAR to LED controllers RGB or White endpoint
Turn it on or off

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

zzh

### Setup

Docker

### Debug log

```
zigbee2mqtt  | Zigbee2MQTT:debug 2024-01-24 15:50:15: Received Zigbee message from 'STYRBAR', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:15: MQTT publish: topic 'zigbee2mqtt/STYRBAR', payload '{"action":"off","battery":85,"last_seen":"2024-01-24T15:50:15+01:00","linkquality":91,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:15: MQTT publish: topic 'zigbee2mqtt/STYRBAR', payload '{"action":"","battery":85,"last_seen":"2024-01-24T15:50:15+01:00","linkquality":91,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:15: MQTT publish: topic 'zigbee2mqtt/STYRBAR/action', payload 'off'
zigbee2mqtt  | Zigbee2MQTT:debug 2024-01-24 15:50:16: Received Zigbee message from 'Living room LED Controller', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 11 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:16: MQTT publish: topic 'zigbee2mqtt/Living room LED Controller', payload '{"brightness":162,"brightness_rgb":49,"brightness_white":162,"color":{"h":24,"hue":24,"s":100,"saturation":100,"x":0.555,"y":0.422},"color_mode":"xy","color_mode_rgb":"xy","color_rgb":{"hue":24,"saturation":100,"x":0.555,"y":0.422},"color_temp":532,"color_temp_rgb":532,"last_seen":"2024-01-24T15:50:16+01:00","linkquality":72,"state":"OFF","state_rgb":"OFF","state_white":"OFF"}'
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:16: MQTT publish: topic 'zigbee2mqtt/Living room LED Controller/rgb', payload '{"brightness":49,"color":{"hue":24,"saturation":100,"x":0.555,"y":0.422},"color_mode":"xy","color_temp":532,"state":"OFF"}'
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:16: MQTT publish: topic 'zigbee2mqtt/Living room LED Controller/white', payload '{"brightness":162,"state":"OFF"}'
zigbee2mqtt  | Zigbee2MQTT:debug 2024-01-24 15:50:17: Received Zigbee message from 'STYRBAR', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:17: MQTT publish: topic 'zigbee2mqtt/STYRBAR', payload '{"action":"on","battery":85,"last_seen":"2024-01-24T15:50:17+01:00","linkquality":83,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:17: MQTT publish: topic 'zigbee2mqtt/STYRBAR', payload '{"action":"","battery":85,"last_seen":"2024-01-24T15:50:17+01:00","linkquality":83,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":null}'
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:17: MQTT publish: topic 'zigbee2mqtt/STYRBAR/action', payload 'on'
zigbee2mqtt  | Zigbee2MQTT:debug 2024-01-24 15:50:18: Received Zigbee message from 'Living room LED Controller', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:18: MQTT publish: topic 'zigbee2mqtt/Living room LED Controller', payload '{"brightness":162,"brightness_rgb":49,"brightness_white":162,"color":{"h":24,"hue":24,"s":100,"saturation":100,"x":0.555,"y":0.422},"color_mode":"xy","color_mode_rgb":"xy","color_rgb":{"hue":24,"saturation":100,"x":0.555,"y":0.422},"color_temp":532,"color_temp_rgb":532,"last_seen":"2024-01-24T15:50:18+01:00","linkquality":69,"state":"ON","state_rgb":"OFF","state_white":"OFF"}'
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:18: MQTT publish: topic 'zigbee2mqtt/Living room LED Controller/rgb', payload '{"brightness":49,"color":{"hue":24,"saturation":100,"x":0.555,"y":0.422},"color_mode":"xy","color_temp":532,"state":"OFF"}'
zigbee2mqtt  | Zigbee2MQTT:info  2024-01-24 15:50:18: MQTT publish: topic 'zigbee2mqtt/Living room LED Controller/white', payload '{"brightness":162,"state":"OFF"}'
```

shawly, 2024-07-23:
Still an issue

shawly, 2024-12-30:
No stale

jannesaarinen, 2024-12-30:
I have the same issue.

Koenkk, 2024-12-30:
You can configure reporting for the `genOnOff` `onOff` attribute via the reporting tab

closed 2025-03-10 by github-actions[bot]