Koenkk/zigbee2mqtt #20983: LEDVANCE 74746 OTA - Warning "Images currently unavailable for device" and missing data

sirjmann92, 2024-01-24:
### What happened?

Zigbee2MQTT.io shows the Sylvania LEDVANCE 74746 LED Panels have OTA support (see https://www.zigbee2mqtt.io/devices/74746.html). OTA tab is missing information for these devices and attempting to check for update throws an error. I've tried restarting HA, Z2M, Mosquitto, and reconfiguring devices. 

I don't recall seeing these devices show up in the OTA list until 1.35 and this issue has been happening with every iteration since, including both latest stable and latest-dev (Docker).

The devices show up in the OTA list, however there are no firmware versions or build dates in their respective columns in the OTA table:
[image]

Clicking "Check for new updates" results in error mesage:
```
Error 2024-01-23 17:54:42 Failed to check if update available for 'LED Panel 01' (OTA: Device didn't respond to OTA request)
```
And I'm also seeing this related warning in the logs:
```
Warning 2024-01-23 18:20:56 OTA: Images currently unavailable for device 'Panel Light 2x2 TW', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":1,"manufacturerCode":4489,"imageType":11,"fileVersion":1057809}
```

Zigbee2MQTT via Docker
- Zigbee2MQTT version 1.35.0 (tried both 1.35.0 commit: b5a90e9 and 1.35.0-dev commit: 14ac9a8)
- Coordinator type zStack3x0
- Coordinator revision 20230507
- Frontend version 0.6.150
- Zigbee-herdsman-converters version 18.15.1
- Zigbee-herdsman version 0.32.6

Home Assistant via Docker
- Core 2024.1.3
- Frontend 20240104.0

Mosquitto broker via Docker (latest)

UZG-01 Coordinator
- Model: | UZG-01
- ESP32 Firmware ver: | 0.1.5 (2023-09-06 12:01:02.723464)
- ESP32 version: | ESP32-D0WD
- CC2652P version: | CC2652PP1FRGZR

### What did you expect to happen?

Data on OTA table should be complete and OTA check should not fail or throw error (even if OTA update is not available).

### How to reproduce it (minimal and precise)

1. Allow an LEDVANCE 74746 device to join the Zigbee network and make sure it interviewed and configured properly
2. Navigate to OTA tab in Zigbee2MQTT
3. Observe missing data for Build Date and Firmware Version columns
4. Click "Check for new updates" on OTA tab for LEDVANCE 74746 devices
5. Observe error in UI after timeout
6. Check logs - observe error and warning messages

### Zigbee2MQTT version

1.35.1-dev commit: 9f3b102

### Adapter firmware version

20230507

### Adapter

UZG-01 (zStack3x0)

### Setup

HA, Z2M, Mosquitto, all in Docker containers on a Synology DS918+

### Debug log

Clicking "Check for new updates" results in error mesage:
```
Error 2024-01-23 17:54:42 Failed to check if update available for 'LED Panel 01' (OTA: Device didn't respond to OTA request)
```
And I'm also seeing this related warning in the logs:

```
Warning 2024-01-23 18:20:56 OTA: Images currently unavailable for device 'Panel Light 2x2 TW', hardwareVersion '0', manufacturerName LEDVANCE, {"fieldControl":1,"manufacturerCode":4489,"imageType":11,"fileVersion":1057809}
```

nmbgeek, 2024-01-28:
I am seeing similar behavior with 2 Sengled E12-N11 bulbs

mvoss96, 2024-03-08:
Im also getting the error on the newest version of zigbee2mqtt:
Warning 2024-03-08 10:27:23OTA: Images currently unavailable for device 'Lightify Switch Mini', hardwareVersion '1', manufacturerName OSRAM, {"fieldControl":0,"manufacturerCode":4364,"imageType":109,"fileVersion":17498112}'
Info 2024-03-08 10:27:23No update available for '0x000d6f00109c53ae'

breucode, 2024-04-14:
I have the same problem with an OSRAM Flex RGBW strip.
I am on zigbee2mqtt 1.36.1

`Warning 2024-04-14 11:49:12OTA: Images currently unavailable for device 'Flex RGBW', hardwareVersion '1', manufacturerName OSRAM, {"fieldControl":0,"manufacturerCode":4364,"imageType":2,"fileVersion":16909063}'`

sirjmann92, 2024-06-02:
This is still an issue.

Still not able to update the firmware on these LEDVANCE 74746 LED panel lights, even though they show up in the OTA list, but no current firmware version is available (blank). 

Issue was reported on version 1.35.1 back in January and we're on 1.37.1 now.

https://www.zigbee2mqtt.io/devices/74746.html indicates they are OTA compatible with Z2M.

Can someone please either remove this device from the OTA list or correct it so that it reports the current firmware version and allows for updating?

Koenkk, 2024-06-02:
Seems no OTA's are available for this device, disabled it.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-02 by Koenkk

sirjmann92, 2024-06-02:
Thank you, @Koenkk - Might be worth removing the OTA compatibility from the z2m.io site as well so it doesn't cause confusion? Honestly, I'd rather it work properly, obviously (show current version in OTA and correctly report no updates are available, rather than blank (null) data and errors) but this will cut down on confusion until it's able to be repaired (if ever).

Thanks again!

Koenkk, 2024-06-03:
Docs will be updated with the next z2m release.