Koenkk/zigbee2mqtt #20991: [New device support]: Add support for Tuya TS0601 - _TZE204_mby4kbtq (MOES - ZigBee Smart Gas Leakage Detector Combustible Sensor)

moipcr, 2024-01-24:
### Link

https://moeshouse.com/products/zigbee-gas-leakage-detector

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c138c284ab9aa3","nwkAddr":16425,"manufId":4417,"manufName":"_TZE204_mby4kbtq","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_mby4kbtq","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1706124177464,"defaultSendRequestWhen":"immediate"}

### Comments

I follow the guide but I can´t get Tuya endpoints for this device. I try to simulate with custom definition other MOES Smoke Detector but can´t get any sensor value only link level. Attached custom js for any help.

[custom.zip](https://github.com/Koenkk/zigbee2mqtt/files/14042879/custom.zip)

### External defintion

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_mby4kbtq',
    description: 'Automatically generated definition',
    extend: [],
};

module.exports = definition;
```

MrEcosse, 2024-04-04:
This should do it

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_mby4kbtq']),
    model: 'ZSS-QY-GL-U-EN',
    vendor: 'TuYa',
    description: 'Tuya Zigbee Gas Sensor 3',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.gas(), e.binary('preheat_status', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'), tuya.exposes.faultAlarm(), e.binary('alarm_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Alarm test switch'), tuya.exposes.silence()],
        meta: {
            tuyaDatapoints: [
                [1, 'gas', tuya.valueConverter.trueFalse0],
                [10, 'preheat_status', tuya.valueConverter.trueFalse0],
                [11, 'fault_alarm', tuya.valueConverter.trueFalse1],
                [13, 'alarm_switch', tuya.valueConverter.onOff],
                [16, 'silence', tuya.valueConverter.raw],
            ],
	},
};

module.exports = definition;
```

MrEcosse, 2024-04-05:
With Z2M 1.36.1 use the converter below that is working fine with my unit

```
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
const zigbeeHerdsmanUtils = require('zigbee-herdsman-converters/lib/utils');
const exposes = zigbeeHerdsmanConverters['exposes'] || require("zigbee-herdsman-converters/lib/exposes");
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters || zigbeeHerdsmanConverters.fromZigbee;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters || zigbeeHerdsmanConverters.toZigbee;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_mby4kbtq']),
    model: 'ZSS-QY-GL-U-EN',
    vendor: 'TuYa',
    description: 'Tuya Zigbee Gas Sensor 3',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.gas(), e.binary('preheat_status', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'), tuya.exposes.faultAlarm(), e.binary('alarm_switch', ea.STATE_SET, 'ON', 'OFF').withDescription('Alarm test switch'), tuya.exposes.silence()],
        meta: {
            tuyaDatapoints: [
                [1, 'gas', tuya.valueConverter.trueFalse0],
                [10, 'preheat_status', tuya.valueConverter.trueFalse0],
                [11, 'fault_alarm', tuya.valueConverter.trueFalse1],
                [13, 'alarm_switch', tuya.valueConverter.onOff],
                [16, 'silence', tuya.valueConverter.raw],
            ],
	},
};

module.exports = definition;
```

Koenkk, 2024-04-06:
@MrEcosse could you create a [pull request](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts) to add out-of-the-box support for this device?

pflugj, 2024-07-19:
I have some of these devices and used the above converter
The only thing that's missing, is the correct picture.
Looking at your supported device db, I found:
TS0601_gas_sensor_1
If I can help to add the picture than please let me know what I can do do to help

Koenkk, 2024-07-21:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-21 by Koenkk

pflugj, 2024-07-21:
Thanks a lot :-)

GuyFromMars365, 2024-12-23:
Running Home Assistant Add-on: Zigbee2MQTT, Current version: 1.42.0-2, but my Moes ZSS-QY-GL-U still hasn't been recognized. Has this PR been merged into the version I'm running?

maunz1337, 2025-01-13:
Hello there,

i am also not able to add this device to my network by default:
[image]

What should/can i do?

tvrabec, 2025-05-24:
Having very same issue, did anyone solve that? Device is recognised in ZHA, but no entity available.