Koenkk/zigbee2mqtt #20992: Regression with Develco SPLZB-131

paranoidi, 2024-01-24:
### What happened?

After some update (I do not know which) all my SPLZB-131 sockets started reporting power consumption incorrectly where are previously they worked perfectly.

### What did you expect to happen?

Reported "Instantaneous measured power" should be correct. It seems to be off by orders of magnitude.

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Add-on on home assistant OS

fschaal, 2024-03-12:
Same here. My PC and 2 monitors are connected to the plug but It's saying power of 0.11W.

[image]

oerik, 2024-05-30:
I had the same, removing the device and pair it again solved the wrong power measurement.

paranoidi, 2024-06-05:
Issue was solved by upgrading the sockets firmware.

closed 2024-06-05 by paranoidi