Koenkk/zigbee2mqtt #20993: Zigbee2mqtt freeze after error command publish set state

Mab306, 2024-01-24:
### What happened?

I have two problems and solution for that. Hope someone helps me. 
 1. problem but everthing works:

Log : 
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]

2. Problem error and Zigbee2mqtt id freeze must restart addon:
 
Log:
Zigbee2MQTT:error 2024-01-24 20:47:13: Publish 'set' 'state' to 'LED Stripe Küche' failed: 'Error: Command 0xa4c13883d2104a19/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 20983 - 1 - 28 - 6 - 11 after 10000ms)'

Have repair that device but sometimes is another device. If that device not work okay but why is Zigbee2mqtt freezing after that error?  After freeze nothing work all lights stay last state. Must restart Zigbee2mqtt after that work evething. 

### What did you expect to happen?

Device not Response 

### How to reproduce it (minimal and precise)

If more lights go on at the same Time

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

EZSP v12

### Adapter

Skyconnect

### Setup

Home assistant OS Intel nuc

tonyamtar, 2024-01-25:
Hello 

Same error for me 

Publish 'set' 'state' to '0x54ef4410003679a3' failed: 'Error: Command 0x54ef4410003679a3/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'

Mab306, 2024-01-25:
@tonyamtar after that error your Zigbee2mqtt add on freezing too?

tonyamtar, 2024-01-25:
Yes, after that nothing works. Only a restart works. I'm not at home this evening I'm going to try changing the port of the device

tonyamtar, 2024-01-26:
I noticed that when I restart my Proxmox VM it works better again but for just one day. After restarting, the RAM gradually increases until it reaches 26 out of 27GB available. From then on I am at maximum everything works except ZIGBEE2MQTT

Mab306, 2024-01-26:
@Koenkk could u maybe help i try a lot of but couldnt find solution against frezzing from addon.

Koenkk, 2024-01-28:
- Is this still present with the latest  [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?
- If yes, provide the herdsman debug log of this
See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Mab306, 2024-01-28:
I have downgrade my firmware from skyconnect dongle to firmware: NabuCasa_SkyConnect_EZSP_v7.1.1.0_ncp-uart-hw_115200.gbl

First Problem with:

Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]

seems to be solved. :-) the second problen must be wait a little time i have downgrade for 20 mins. I had the newst firmware for my dongle maybe isnt stable for zigbee2mqtt.

Nerivec, 2024-01-29:
@Mab306 First problem is just old code that needed to be removed. You should no longer see it in Z2M's next release (let me know if you do). It doesn't have any impact on Z2M in any case.
Second might be related to https://github.com/Koenkk/zigbee2mqtt/issues/20868? One of you two mind checking your machine's logs to see if the adapter is crashing?
_PS: If you had firmware 7.4.0 you need EDGE for it to work properly (until next release)._

Mab306, 2024-01-29:
@Nerivec first Problem is gone but good to know.
 Second Problem still availabl. I think its the same Problem like other in #20868. I have Download Firmware from skyconnect to 7.1.1 but still freezing now i try Edge with herdsman debug on. 

Really is skyconnect not good or why i have this problems ?

Nerivec, 2024-01-29:
You said HAOS, right? So, if you go in Settings>Logs and you switch to "Host" in top right dropdown, do you see anything talking about "USB"?
What firmware are you using? The ones from Nabu Casa, darkxst, other?

Mab306, 2024-01-29:
@Nerivec  I didnt See anything about usb. Have restart and look again same didnt See anything about usb in log "host".
I think must be nabu case. I Flash  the firmware myself over Web flasher.  i take this Firmware: 

[image]

Mab306, 2024-01-30:
@Koenkk @Nerivec Now i have frezzing and nothing work. Only restart add on

herdsman:controller:endpoint Command 0xcc86ecfffec1d61b/1 genLevelCtrl.moveToLevelWithOnOff({"level":38,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-30T15:49:26.187Z zigbee-herdsman:controller:endpoint Command 0xcc86ecfffec1d61b/1 genLevelCtrl.moveToLevelWithOnOff({"level":38,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-30T15:49:26.274Z zigbee-herdsman:controller:endpoint Command 0xcc86ecfffec1d61b/1 genLevelCtrl.moveToLevelWithOnOff({"level":38,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-30T15:49:26.364Z zigbee-herdsman:controller:endpoint Command 0xcc86ecfffec1d61b/1 genLevelCtrl.moveToLevelWithOnOff({"level":38,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[image]

Nerivec, 2024-01-30:
You said you were going to try EDGE with herdsman debug, did you get those logs?

Mab306, 2024-01-30:
Yes

closed 2024-08-28 by github-actions[bot]