Koenkk/zigbee2mqtt #20999: [New device support]: Ewelink Zigbee GU10 LED Bulb

RezzZ, 2024-01-25:
### Link

https://fr.aliexpress.com/item/1005006099930203.html?spm=a2g0o.order_list.order_list_main.5.5f795e5bHoQ1qo&gatewayAdapt=glo2fra

### Database entry

{"id":85,"type":"Router","ieeeAddr":"0xb40ecfd093f80000","nwkAddr":47252,"manufId":4742,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"CK-BL702-AL-01(7009_Z102LG03-1)","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,61184,64529,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"CK-BL702-AL-01(7009_Z102LG03-1)","manufacturerName":"eWeLink","powerSource":0,"zclVersion":8,"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20230105","swBuildId":"1.2.3"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":142,"colorTempPhysicalMax":500,"colorMode":2,"currentHue":29,"currentSaturation":7,"colorTemperature":346}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{"scenes":{"0_9":{"name":"relax","state":{"color_mode":"color_temp","state":"ON","brightness":144,"color":{"hue":27,"saturation":91},"color_temp":447}},"5_9":{"name":"dimmed","state":{"color_mode":"color_temp","state":"ON","brightness":77,"color":{"saturation":91,"hue":27,"h":27,"s":91},"color_temp":447}},"6_9":{"name":"nightlight","state":{"color_mode":"color_temp","state":"ON","brightness":2,"color":{"saturation":91,"hue":27,"h":27,"s":91},"color_temp":447}},"1_9":{"name":"read","state":{"color_mode":"color_temp","state":"ON","brightness":234,"color":{"saturation":78,"hue":33,"h":33,"s":78},"color_temp":346}},"2_9":{"name":"bright","state":{"color_mode":"color_temp","state":"ON","brightness":254,"color":{"saturation":78,"hue":33,"h":33,"s":78},"color_temp":346}},"3_9":{"name":"concentrate","state":{"color_mode":"color_temp","state":"ON","brightness":254,"color":{"saturation":42,"hue":42,"h":42,"s":42},"color_temp":233}},"4_9":{"name":"energize","state":{"color_mode":"color_temp","state":"ON","brightness":254,"color":{"saturation":8,"hue":209,"h":209,"s":8},"color_temp":156}},"99_9":{"name":"off","state":{"color_mode":"color_temp","state":"OFF","brightness":254,"color":{"saturation":8,"hue":209,"h":209,"s":8},"color_temp":156}},"0_8":{"name":"relax","state":{"color_mode":"color_temp","state":"ON","brightness":144,"color":{"x":0.5026,"y":0.4153,"hue":27,"saturation":91},"color_temp":447}},"1_8":{"name":"read","state":{"color_mode":"color_temp","state":"ON","brightness":234,"color":{"x":0.445,"y":0.4067,"hue":33,"saturation":78},"color_temp":346}},"2_8":{"name":"bright","state":{"color_mode":"color_temp","state":"ON","brightness":254,"color":{"x":0.445,"y":0.4067,"hue":33,"saturation":78},"color_temp":346}},"3_8":{"name":"concentrate","state":{"color_mode":"color_temp","state":"ON","brightness":254,"color":{"saturation":42,"hue":42,"h":42,"s":42,"x":0.3685,"y":0.369},"color_temp":233}},"4_8":{"name":"energize","state":{"color_mode":"color_temp","state":"ON","brightness":254,"color":{"saturation":8,"hue":209,"h":209,"s":8,"x":0.315,"y":0.325},"color_temp":156}},"5_8":{"name":"dimmed","state":{"color_mode":"color_temp","state":"ON","brightness":77,"color":{"x":0.5026,"y":0.4153,"hue":27,"saturation":91},"color_temp":447}},"6_8":{"name":"nightlight","state":{"color_mode":"color_temp","state":"ON","brightness":2,"color":{"x":0.5026,"y":0.4153,"hue":27,"saturation":91},"color_temp":447}},"99_8":{"name":"off","state":{"color_mode":"color_temp","state":"OFF","brightness":2,"color":{"x":0.3059,"y":0.316,"hue":218,"saturation":17},"color_temp":142}}}}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20230105","swBuildId":"1.2.3","zclVersion":8,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1706162011741,"defaultSendRequestWhen":"immediate"}

### Comments

Not 100% sure if this is a "new" device but the device already registered in z2m for this [modelid](https://www.zigbee2mqtt.io/devices/CK-BL702-AL-01.html#tuya-ck-bl702-al-01)  is a different light bulb than the gu10 spots I purchased.
Also, even though the bulbs are recognized, they "go" offline after 15/20 minutes but are actually still responding to all commands send from homeassistant...
If more info is required please let me know.

### External defintion

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['CK-BL702-AL-01(7009_Z102LG03-1)'],
    model: 'CK-BL702-AL-01(7009_Z102LG03-1)',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[142,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
};

module.exports = definition;
```

RezzZ, 2024-01-25:
2024-01-25 10:04:11Failed to ping 'Kitchen Rail5' (attempt 1/1, Read 0xb40ecfd11b0d0000/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8303 - 1 - 204 - 0 - 1 after 10000ms))
here a warning that constantly pops up for all my 6 spots

alex-jung, 2024-02-18:
+1 

I have the same issues with GU10 bulbs

oraichu, 2024-02-27:
Was planning on getting some of these, so +1 from me as well.

Jop5298, 2024-03-14:
I have ordered and received the same spots but I am not able to detect them at all. However, they are detected almost instant in the Philips Hue app. Any idea how this can?

mikesrus, 2024-03-16:
which channel do you use in z2m? I have 2 such lamps and they are not visible in any way in z2m, but they are perfectly connected in yandex stantion 2

Jop5298, 2024-03-16:
I stopped Zigbee2MQTT, disconnected my zigbee dongle, reconnected it again, started Zigbee2MQTT and it could find the spot. So it looks like it was a problem on my device.

Koenkk, 2024-03-17:
I've added the device, not sure about the connectivity issues.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

ncodee, 2024-05-09:
I'm having the same issue with these bulbs, commands are being sent via z2m, but it takes a good 10 or more seconds the bulb to respond. LQI 220+, but Z2M throwing errors and extremely slow.

Happy to provide logs, if needed

Any fix on this please?

Billos, 2024-05-18:
Hello,

Do you know if it's the same device as this one ? https://fr.aliexpress.com/item/1005006155842902.html?spm=a2g0o.order_list.order_list_main.11.debe5e5bZUNwg1&gatewayAdapt=glo2fra

milkywade, 2024-05-18:
Think so idd

Billos, 2024-05-18:
> I have ordered and received the same spots but I am not able to detect them at all. However, they are detected almost instant in the Philips Hue app. Any idea how this can?

I have the same issue, can't detect it with my Z2M

Br3Ak3R, 2024-06-09:
+1

Same issues here

Bought this https://de.aliexpress.com/item/1005006492947103.html

janfuhlrott, 2024-07-27:
Bought same Bulb, connects only directly to Coordinator, showing as router but does no routing und ist recognized as Tuya...

closed 2025-01-07 by github-actions[bot]