Koenkk/zigbee2mqtt #21001: zigbee2mqtt show LQI 255 with Sonoff - Multipan Protocol

LightMoon, 2024-01-25:
### What happened?

False 255 LQI using Multipan Addon - Devices are ok but the LQI showing 255 for all devices near or far from coordinator.

### What did you expect to happen?

Expecting to see realistic LQI for Zigbee nodes 

### How to reproduce it (minimal and precise)

- Coordinator: Silicon Labs EFR32MG21 - Sonoff ZBDongle-E
- Firmware:  [MultiPAN RCP Firmware ZBDongle-E](https://raw.githubusercontent.com/darkxst/silabs-firmware-builder/main/firmware_builds/zbdonglee/rcp-uart-802154-v4.3.1-zbdonglee-460800.gbl)
- 

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

v4.3.1

### Adapter

Sonoff ZBDongle-E

### Setup

Add-on on Home Assistance OS

### Debug log

#https://github.com/Koenkk/zigbee-herdsman/issues/738
#https://github.com/home-assistant/addons/issues/3158
@Drealine

Drealine, 2024-01-25:
Agree. Not resolved at the moment. Waiting too

g1mp11, 2024-07-10:
Having the same issue at all devices, no matter which brand.

iaxexo, 2024-12-05:
Following since i do have the same issue. On the map LQI seems to be meaningful
[image]
but on the table it is struck to 255 for each device
[image]

closed 2025-04-09 by github-actions[bot]

DangerDan83, 2025-08-12:
I have a ConBee II Stick and it#s the same. All devices (5 different Brands) show 255, but on the map it's the real value.
Version: 2.6.0-1