Koenkk/zigbee2mqtt #21004: Problem (re-) pairing IKEA E1743

bipsendk, 2024-01-25:
### What happened?

Z2M lost connectivity to an IKEA E1743 switch.
Tried to re-pair without luck.
Replaced battery in switch, did a force remove in Z2M and tried to pair agin -keep gettin " device has not successfully been paired"

### What did you expect to happen?

Device to be paired and discovered correctly.

### How to reproduce it (minimal and precise)

Press pairing button on IKEA switch 4 times
enable pairing in Z2M
Z2M reports start interviewing 
Z2M fails pairing with device has not successfully been paired 

DeviceId in log: 0xccccccfffed18919

### Zigbee2MQTT version

1.32.2-1 

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee P stick

### Setup

Add-on for HassIO - running as a VM with USB passthrough to the Zigbee dongle.

### Debug log

`info  2024-01-25 13:15:23: Zigbee: allowing new devices to join.
info  2024-01-25 13:15:23: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"zcpes-1"}'
info  2024-01-25 13:15:40: Starting interview of '0xccccccfffed18919'
info  2024-01-25 13:15:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xccccccfffed18919","ieee_address":"0xccccccfffed18919","status":"started"},"type":"device_interview"}'
debug 2024-01-25 13:15:43: Received Zigbee message from 'smartplug_p51s', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":4,"rmsCurrent":61,"rmsVoltage":227}' from endpoint 1 with groupID 0
info  2024-01-25 13:15:43: MQTT publish: topic 'zigbee2mqtt/smartplug_p51s', payload '{"child_lock":"UNLOCK","current":0.06,"energy":19.29,"indicator_mode":"off/on","linkquality":162,"power":4,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
debug 2024-01-25 13:15:43: Received Zigbee message from 'smartplug_p51s', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,1929]}' from endpoint 1 with groupID 0
info  2024-01-25 13:15:43: MQTT publish: topic 'zigbee2mqtt/smartplug_p51s', payload '{"child_lock":"UNLOCK","current":0.06,"energy":19.29,"indicator_mode":"off/on","linkquality":165,"power":4,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":227}'
debug 2024-01-25 13:15:46: Received Zigbee message from 'utility_meter_dishwasher', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":230}' from endpoint 1 with groupID 0
info  2024-01-25 13:15:46: MQTT publish: topic 'zigbee2mqtt/utility_meter_dishwasher', payload '{"child_lock":"LOCK","current":0,"energy":517.19,"indicator_mode":"off/on","linkquality":69,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":230}'
debug 2024-01-25 13:15:46: Received Zigbee message from 'utility_meter_dishwasher', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,51719]}' from endpoint 1 with groupID 0
info  2024-01-25 13:15:46: MQTT publish: topic 'zigbee2mqtt/utility_meter_dishwasher', payload '{"child_lock":"LOCK","current":0,"energy":517.19,"indicator_mode":"off/on","linkquality":66,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":null,"voltage":230}'
debug 2024-01-25 13:15:46: Received Zigbee message from 'Marie loftlampe', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2024-01-25 13:15:46: Successfully pinged 'Marie loftlampe' (attempt 1/2)
debug 2024-01-25 13:15:55: Received Zigbee message from 'smartplug_hevo', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":192}' from endpoint 1 with groupID 0
debug 2024-01-25 13:15:55: Device '0xccccccfffed18919' announced itself
info  2024-01-25 13:15:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xccccccfffed18919","ieee_address":"0xccccccfffed18919"},"type":"device_announce"}'
error 2024-01-25 13:16:01: Failed to interview '0xccccccfffed18919', device has not successfully been paired
info  2024-01-25 13:16:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xccccccfffed18919","ieee_address":"0xccccccfffed18919","status":"failed"},"type":"device_interview"}'
debug 2024-01-25 13:16:02: Received MQTT message on 'homeassistant/sensor/P51S-satellite/availability' with data 'online'
debug 2024-01-25 13:16:11: Received Zigbee message from 'PIR-soveværelse', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,60,12,3,40,24,4,33,0,0,5,33,2,0,8,33,10,1,10,33,145,186,12,32,1,19,32,0,20,32,0,100,16,0,101,33,179,0,105,32,30,106,32,2,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2024-01-25 13:16:11: RTCGQ14LM: Processed buffer into data {"1":3132,"3":24,"4":0,"5":2,"8":266,"10":47761,"12":1,"19":0,"20":0,"100":0,"101":179,"105":30,"106":2,"107":0}
debug 2024-01-25 13:16:11: RTCGQ14LM: unknown key 19 with value 0
debug 2024-01-25 13:16:11: RTCGQ14LM: unknown key 20 with value 0
debug 2024-01-25 13:16:11: RTCGQ14LM: Processed data into payload {"voltage":3132,"battery":100,"device_temperature":24,"power_outage_count":1,"illuminance":179,"detection_interval":30,"motion_sensitivity":"medium","trigger_indicator":false}
debug 2024-01-25 13:16:11: RTCGQ14LM: Processed data into payload {"voltage":3132,"battery":100,"device_temperature":24,"power_outage_count":1,"illuminance":179,"detection_interval":30,"motion_sensitivity":"medium","trigger_indicator":false}
info  2024-01-25 13:16:11: MQTT publish: topic 'zigbee2mqtt/PIR-soveværelse', payload '{"battery":100,"detection_interval":30,"device_temperature":24,"illuminance":179,"linkquality":87,"motion_sensitivity":"medium","occupancy":false,"power_outage_count":1,"trigger_indicator":false,"update":{"installed_version":10,"latest_version":10,"state":"idle"},"update_available":null,"voltage":3132}'
debug 2024-01-25 13:16:11: Received MQTT message on 'homeassistant/sensor/P51S/P51S_battery/P51S_battery_charge_status/state' with data '0'
debug 2024-01-25 13:16:11: Received MQTT message on 'homeassistant/sensor/P51S/P51S_battery/P51S_battery_full_charge_lifetime/state' with data '-1'
debug 2024-01-25 13:16:11: Received MQTT message on 'homeassistant/sensor/P51S/P51S_battery/P51S_battery_charge_remaining_percentage/state' with data '66'
debug 2024-01-25 13:16:11: Received MQTT message on 'homeassistant/sensor/P51S/P51S_battery/P51S_battery_charge_remaining/state' with data '-1'
debug 2024-01-25 13:16:11: Received MQTT message on 'homeassistant/sensor/P51S/P51S_battery/P51S_battery_powerline_status/state' with data 'Online'
`

bipsendk, 2024-01-25:
Found a work-around (at least this time):

- Delete switch from Z2M
- Restart Z2M
- Pair Switch

Just strange, that this is required for it to work...

wormvortex, 2024-01-25:
Doesn’t work for me. Tried it numerous times.

mbbert, 2024-02-11:
Same problem here. Nothing that I tried so far did work. 3 of my 4 switches have this issue now.

Failed to interview '0x0123456789ABCDEF', device has not successfully been paired

Z2M version: 1.35.2
E1743 version: 24.4.6

pcb1962, 2024-02-14:
I did an OTA update on about 12 of these switches a few days ago (2.5 hours each!). After the update they were all reporting the battery at 1% which I understand is a known issue. To fix this I did a force delete and re-pair of each one. All of them managed to re-pair, a few of them took several attempts (failed interview) and I had to put new batteries in some of them to get them to pair, even when the battery was reading 60%+ before the upgrade. Some of them came back with no battery field in the mqtt message but settled down after a few hours. Now 2 or 3 days later most of them are showing battery at 1% again, and the low battery red light is showing when they're clicked. Most of these are paired via IKEA repeaters and outlets which might give them a better chance, but some are paired with Sonoff stick or TI coordinator. I've seen reports elsewhere of these draining the battery in a few days after upgrade, with no solution. If that's the way it is I shall have to retire them and replace with something from Aqara.

mbbert, 2024-06-02:
I sucessfully re-paired my switch again by downgrading to Z2M version 1.33.1.

Today I upgraded to version 1.38.0 and tried the pairing again, but without any luck. I also tried the new (at least for me) Interview button. But I got the same error there.

Then I gave an old docker container a try, which I still had lying around, containing version 1.33.1. I removed the device, which worked on 2nd try and paired it again. I pressed the buttons from time to time during this process. It worked on the first attempt!

Here's the protocol:

```
Zigbee2MQTT:info  2024-06-02 16:04:58: Removing device '0x04cd15fffe0cfd68' (block: false, force: false)
Zigbee2MQTT:error 2024-06-02 16:05:08: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device '0x04cd15fffe0cfd68' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'
Zigbee2MQTT:info  2024-06-02 16:05:08: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{},"error":"Failed to remove device '0x04cd15fffe0cfd68' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)","status":"error","transaction":"8trub-1"}'
Zigbee2MQTT:info  2024-06-02 16:05:21: Removing device '0x04cd15fffe0cfd68' (block: false, force: false)
Zigbee2MQTT:info  2024-06-02 16:05:26: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload ''
Zigbee2MQTT:info  2024-06-02 16:05:26: Successfully removed device '0x04cd15fffe0cfd68' (block: false, force: false)
Zigbee2MQTT:info  2024-06-02 16:05:27: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":false,"id":"0x04cd15fffe0cfd68"},"status":"ok","transaction":"8trub-2"}'
Zigbee2MQTT:info  2024-06-02 16:06:04: Device '0x04cd15fffe0cfd68' joined
Zigbee2MQTT:info  2024-06-02 16:06:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe0cfd68","ieee_address":"0x04cd15fffe0cfd68"},"type":"device_joined"}'
Zigbee2MQTT:info  2024-06-02 16:06:05: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:06:05.092Z"}'
Zigbee2MQTT:info  2024-06-02 16:06:05: Starting interview of '0x04cd15fffe0cfd68'
Zigbee2MQTT:info  2024-06-02 16:06:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe0cfd68","ieee_address":"0x04cd15fffe0cfd68","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2024-06-02 16:06:05: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:06:05.908Z"}'
Zigbee2MQTT:info  2024-06-02 16:06:11: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:06:11.792Z"}'
Zigbee2MQTT:info  2024-06-02 16:06:11: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:06:11.810Z"}'
Zigbee2MQTT:error 2024-06-02 16:06:26: Failed to interview '0x04cd15fffe0cfd68', device has not successfully been paired
Zigbee2MQTT:info  2024-06-02 16:06:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe0cfd68","ieee_address":"0x04cd15fffe0cfd68","status":"failed"},"type":"device_interview"}'
Zigbee2MQTT:info  2024-06-02 16:06:28: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:06:28.424Z"}'
Zigbee2MQTT:info  2024-06-02 16:06:28: Starting interview of '0x04cd15fffe0cfd68'
Zigbee2MQTT:info  2024-06-02 16:06:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe0cfd68","ieee_address":"0x04cd15fffe0cfd68","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2024-06-02 16:06:29: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:06:29.230Z"}'
Zigbee2MQTT:info  2024-06-02 16:06:29: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:06:29.339Z"}'
Zigbee2MQTT:info  2024-06-02 16:06:30: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x04cd15fffe0cfd68","ieee_address":"0x04cd15fffe0cfd68"},"type":"device_announce"}'
Zigbee2MQTT:info  2024-06-02 16:06:32: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:06:32.701Z","linkquality":29}'
Zigbee2MQTT:info  2024-06-02 16:06:32: Successfully interviewed '0x04cd15fffe0cfd68', device has successfully been paired
Zigbee2MQTT:info  2024-06-02 16:06:32: Device '0x04cd15fffe0cfd68' is supported, identified as: IKEA TRADFRI ON/OFF switch (E1743)
Zigbee2MQTT:info  2024-06-02 16:06:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TRADFRI ON/OFF switch","exposes":[{"access":5,"description":"Remaining battery in %, can take up to 24 hours before reported.","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["on","off","brightness_move_down","brightness_move_up","brightness_stop"]},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E1743","options":[{"access":2,"description":"Set to false to disable the legacy integration (highly recommended), [... cut ...]
Zigbee2MQTT:info  2024-06-02 16:08:04: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:08:04.468Z","linkquality":32}'
Zigbee2MQTT:info  2024-06-02 16:08:05: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffe0cfd68', payload '{"last_seen":"2024-06-02T14:08:04.468Z","linkquality":32,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"}}'
```

After that, I upgraded again to the latest version, and bound it to an IKEA LED2002G5 without any issues.

closed 2024-12-30 by github-actions[bot]