Koenkk/zigbee2mqtt #21008: Permit join does not work on some devices. How do we find the cause?

mrmaximas, 2024-01-25:
### What happened?

I was experimenting with xiaomi ZNCZ02LM outlet and found that not all routers on my network go into pairing mode.

### What did you expect to happen?

I expected that the new device can be paired with any router in my network, switched to pairing mode individually. With some routers like [ZWSM16-1-Zigbee](https://www.zigbee2mqtt.io/devices/ZWSM16-1-Zigbee.html#avatto-zwsm16-1-zigbee) or [ZWSM16-2-Zigbee](https://www.zigbee2mqtt.io/devices/ZWSM16-2-Zigbee.html#avatto-zwsm16-2-zigbee) even at a distance of 10 cm from the new paired device can not pass interviews. Visually nothing happens. It looks like the individually selected router is not in pairing mode. At the same time, the new device perfectly paired with other routers in the network.

### How to reproduce it (minimal and precise)

You should try pairing the new device with a individually router device in the network.

### Zigbee2MQTT version

1.35.1-dev commit: 51897288

### Adapter firmware version

20230922

### Adapter

Sonoff ZBDONGLE-P

### Setup

Mac OS catalina latest

Koenkk, 2024-01-26:
The [coordinator check](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-coordinator-check) might help.

mrmaximas, 2024-01-26:
> The [coordinator check](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-coordinator-check) might help.

how to do it via web ui?

mrmaximas, 2024-01-26:
> The [coordinator check](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-coordinator-check) might help.

{
  "data": {
    "missing_routers": []
  },
  "status": "ok"
}

Koenkk, 2024-01-27:
This may then be some issues of the specific router, to figure this out traffic needs to be [sniffed](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)

closed 2024-01-27 by mrmaximas