Koenkk/zigbee2mqtt #21013: [New device support]: Moes TS0601 _TZE200_icka1clh

Blendi, 2024-01-25:
### Link

https://www.aliexpress.com/item/1005005801861031.html?spm=a2g0o.order_list.order_list_main.25.50b65e5bNhNdKV&gatewayAdapt=glo2fra

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"0x6c5cb1fffe98d021","nwkAddr":21894,"manufId":4098,"manufName":"_TZE200_icka1clh","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"2�B-f2�B-\u00122�B-\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_icka1clh","stackVersion":0,"dateCode":"","appVersion":65,"hwVersion":1,"powerSource":3}},"manuSpecificTuya_2":{"attributes":{}},"haDiagnostic":{"attributes":{}},"genIdentify":{"attributes":{}},"genScenes":{"attributes":{"count":0,"currentScene":0}},"genLevelCtrl":{"attributes":{}},"genBinaryInput":{"attributes":{}},"genMultistateInput":{"attributes":{}},"closuresShadeCfg":{"attributes":{}},"closuresDoorLock":{"attributes":{}},"closuresWindowCovering":{"attributes":{}},"manuSpecificTuya_3":{"attributes":{}},"manuSpecificUbisysDeviceSetup":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1706032659886,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,

I bough this new device on aliexpress, and before buying I tried to make sure this device is compatible with zigbee2mqtt. From the reviews on the product looked like it was supported and people could connect the roller blind to homeassistant. 

Most probably this product is a new version and they have changed things a bit with zigbee. Unfortunatelly I dont have a tuya gateway otherwise I would have tried to follow the guide on (https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats)

I tired below external definition with different exposes and tuyaDatapoints but nothing worked, until at the end I just commented exposes and tuyaDatapoints completely. I copied the 'exposes' and tuyaDatapoints from existing tuya external definitions for roller blinders already known in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts

I just throw things at the wall and see what sticks. 

### External defintion

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_icka1clh',
        },
    ],
    model: 'TS0601_roller_blind',
    vendor: 'TuYa',
    description: 'AM43 Roller Blind',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [ 
        // Here you should put all functionality that your device exposes
		//e.text('work_state', ea.STATE),
        //e.cover_position().setAccess('position', ea.STATE_SET),
        //e.battery(),
        //e.enum('opening_mode', ea.STATE_SET, ['tilt', 'lift']).withDescription('Opening mode'),
        //e.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Motor side'),
        //e.enum('set_upper_limit', ea.STATE_SET, ['start', 'stop']).withDescription('Learning'),
        //e.enum('factory_reset', ea.STATE_SET, ['SET']).withDescription('Remove limits'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
		    //[1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})],
            //[2, 'position', tuya.valueConverter.coverPositionInverted],
            //[3, 'position', tuya.valueConverter.coverPositionInverted],
            //[4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})],
            //[7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            //[13, 'battery', tuya.valueConverter.raw],
            //[101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            //[102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
            //[107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
};

module.exports = definition;
```

Blendi, 2024-02-02:
I have no idea what I did but it is now partially working. I just went over different places and copied part of codes. Only issue I have is that buttons orientation is wrong. I press button with arrow facing down and the roller blinds open. I press the opposite button facing up the blind/curtains roll down. There should be somewhere an option to invert the function of the buttons but anyway so far I am happy with the progress. Battery percentage also is not updating. That might be my next goal.

In the meantime I bought a cheap tuya gateway from aliexpress, in order to extract the tuyadatapoints later.

my external js code looks like below.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_icka1clh',
        },
    ],
    model: 'AM43-0.45/40-ES-EB',
    vendor: 'Moes',
    description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
	onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [ 
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options')
                .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0)
                    .withValueMax(255)
                    .withDescription('Motor speed'))
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
		    [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})],
            [2, 'position', tuya.valueConverter.coverPositionInverted],
            [3, 'position', tuya.valueConverter.coverPositionInverted],
            [4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
            [107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
};

module.exports = definition;`

woloss, 2024-02-06:
Thanks for your code. As for up/down thing, there is `Change Direction` in instruction, I guess it should flip buttons.

Blendi, 2024-02-06:
Hello,

Actually I was wrong. In Homeassistant and Zigbee2mqtt addon, orientation works fine. I press up arrow and the blinds rise. I press stop and they stop. I press down arrow in homeassistant, blind goes down. Also I can control how many percent I want my blind to be opened, 20-30% etc. All is good except the battery percentage, that I cannot see in Homeassistant or zigbee2mqtt addon/docker.

The orientation is wrong only when I press the physical buttons on the device, but that is a minor issue for me as I don’t plan to control it from the buttons on the device. I will update here in case I find the solution for the physical buttons and change their orientation to match with blind movements.

woloss, 2024-02-06:
Looks like there was no battery state in previous revision or it wasn't found. https://github.com/Koenkk/zigbee2mqtt/issues/17363

arrikhan, 2024-06-24:
Innocent question, can the code from ESPHome for the bluetooth model help identify where to find the battery state and illuminance (light sensor)? These models look identical (I have both) other than the protocol to communicate.  ESPHome integration of AM43 exposes battery level and light brightness.

Code is here --> https://github.com/esphome/esphome/tree/dev/esphome/components/am43

Dengar66, 2024-07-02:
I was working on setting this one up as well and came across your issue already opened. This is what I've got currently. I noticed you were missing motor speed specifically. Have you tested the opening mode or setting limits through Zigbee yet?

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_icka1clh',
        },
    ],
    model: 'AM43-0.45/40-ES-EB',
    vendor: 'Moes',
    description: 'Roller blind/shades drive motor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.battery(), // No data
        exposes.enum('motor_direction', ea.STATE_SET, ['normal', 'reversed']).withDescription('Set the motor direction'),
        exposes.numeric('motor_speed', ea.STATE_SET).withValueMin(0).withValueMax(255).withDescription('Motor speed').withUnit('rpm'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'CLOSE': tuya.enum(2), 'STOP': tuya.enum(1), 'OPEN': tuya.enum(0)})],
            [2, 'position', tuya.valueConverter.coverPositionInverted],
            [3, 'position', tuya.valueConverter.coverPositionInverted],
            [4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})], // Have not tested
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})], // Have not tested
            [13, 'battery', tuya.valueConverter.raw], // Not getting data back for this
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(1), 'stop': tuya.enum(0)})],
            [105, 'motor_speed', tuya.valueConverter.raw],
            [107, 'factory_reset', tuya.valueConverter.setLimit], // Have not tested
        ],
    },
    extend: [
    ],
};

module.exports = definition;
`

closed 2025-01-07 by github-actions[bot]

dkahn93, 2025-03-29:
I just got a Moes Roller Blind TS0601 _TZE200_icka1clh and the up, down, and stop work fine, but I'm in the same situation where I'd really like to see on HASS what the battery status is. Has anyone figured this out yet?

I also got it with a solar panel with it that reportedly is supposed to detect the amount of sun it's getting, but I see no status/entity for this either.

Blendi, 2025-03-29:
I was wondering the same @dkahn93 about the light sensor. Since it is provided in the box the device somehow should report on its status via mqtt. But we don’t see any data about it. And not knowing the battery status is really annoying. From time to time the device suddenly goes offline. And you realize it only when you want to control the device or when your last automation was not executed.

dasmith83, 2025-06-10:
> I have the same problem as everyone else. There is no battery status and no lighting sensor.

What external converter code are you using? I'm finding that the above are throwing me errors? 

I find the below works...however the position doesn't seem to be detected properly in Z2M, open/close work fine but that is all...

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_icka1clh'}],
        model: 'AM43-0.45/40-ES-EB',
        vendor: 'MOES',
        description: 'Tuya curtain motor',
        fromZigbee: [fz.tuya_cover],
        toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
        exposes: [e.cover_position().setAccess('state', 2)],
    },
];`

DavidAntonin, 2025-06-10:
> What external converter code are you using? I'm finding that the above are throwing me errors?

I have no external converter

dasmith83, 2025-06-10:
> > What external converter code are you using? I'm finding that the above are throwing me errors?
> 
> I have no external converter

Oh ok...mine suggested it was unsupported without adding the external converter with code above...guess I have a slightly new version of the device perhaps.

bubez81, 2025-07-06:
Hello, sorry can't understand, there is a solution for the battery problem? With an external converter? I'm a little bit confused right now