Koenkk/zigbee2mqtt #21014: [New device support]: Gewiss GWA1513

anthopocket, 2024-01-25:
### Link

https://www.gewiss.com/fr/fr/products/product.1000000.1008169.GWA1513

### Database entry

2024-01-25 22:18:54Publish 'set' 'state' to '0x0015bc001a007842' failed: 'Error: Command 0x0015bc001a007842/1 genOnOff.on({}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 63487 - 1 - 140 - 6 - 11 after 10000ms)' Debug 2024-01-25 22:18:54Error: Command 0x0015bc001a007842/1 genOnOff.on({}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 63487 - 1 - 140 - 6 - 11 after 10000ms) at Timeout._onTimeout (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7)

### Comments

device gewiss GWA1513

### External defintion

```shell
const {onOff, batteryPercentage, temperature, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['GWA1511_MotionSensor'],
    model: 'GWA1511_MotionSensor',
    vendor: 'Gewiss',
    description: 'Automatically generated definition',
    extend: [onOff({powerOnBehavior: false}), batteryPercentage(), temperature(), identify()],
};

module.exports = definition;
```

anthopocket, 2024-02-18:
hi,

it's the same of

https://www.zigbee2mqtt.io/devices/MOSZB-141.html

Name and model are changed for factory.

Koenkk, 2024-02-18:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-18 by Koenkk

anthopocket, 2024-02-18:
it doesnt ok.

try it :
https://www.zigbee2mqtt.io/devices/WISZB-120.html
or
https://www.zigbee2mqtt.io/devices/WISZB-121.html

on the net, i find 

`Gewiss offers a wide range of other ZigBee based devices such as smoke detectors, window/door sensors. Those devices however are rebranded Develco devices and supported by the Develco ZigBee integration plugin.

`

anthopocket, 2024-02-18:
it's not ok

log
 2024-02-18 21:35:02MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x0015bc001a007842"},"error":"Device '0x0015bc001a007842' cannot be configured","status":"error","transaction":"vikqm-3"}'

anthopocket, 2024-02-18:
it's ok
good job
thanksyou

Koenkk, 2024-02-19:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

anthopocket, 2024-02-25:
hi,

it's ok 

when are the new stable ?

Koenkk, 2024-02-25:
1 March