Koenkk/zigbee2mqtt #21015: [New device support]: TS0601 _TZE200_lvkk0hdg (EPTTECH Tank Level Monitor Zigbee)

rvdlinden, 2024-01-25:
### Link

https://www.aliexpress.com/item/1005006395402636.html

### Database entry

{"id":66,"type":"EndDevice","ieeeAddr":"0x0ceff6fffe4bdda2","nwkAddr":39698,"manufId":4098,"manufName":"_TZE200_lvkk0hdg","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�ME-\u0019%NE-\u0019aNE-j�NE-j\u000fPE-j�KE-j�KE-\u0019>LE-i","65506":31,"65508":0,"65534":0,"appVersion":65,"modelId":"TS0601","manufacturerName":"_TZE200_lvkk0hdg","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1706198663760,"defaultSendRequestWhen":"immediate"}

### Comments

I have followed all the documentation on how to add a new unsupported device, but I got stuck on the configuration file. Only 3 out of the 7 Tuya Data Points are working (the read-only points), but the other 4 data points for configuration of the device are seen in Tuya IOT, but refused to be configured in Zigbee2MQTT

The attached external definition file is an attempt to setup the device, but that all I got and do not know how to continue. 

The working Data Points and also the failing Data Points are all described in the definition file for easy reference.

NOTE: There is also an identical device, but then it's a Wifi based device. In the Smart Life app both devices are fully working (all 7 Data Points), but in HA (either via Zigbee2MQTT or LocalTuya) only 3 out of 7 Data Points can be used.
See for the TLC2206 Wifi device https://www.aliexpress.com/item/1005005758585356.html

Both devices have equal Data Points
`{"1":"Liquid level status","2":"Liquid level depth","7":"Set Max","8":"Set Mini","19":"Installation height","21":"Sensor to full line","22":"Liquid level ratio"}`

### External defintion

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
  fingerprint: [
    {
      modelID: 'TS0601',
      manufacturerName: '_TZE200_lvkk0hdg',
    },
  ],
  model: 'TS0601_tlc2206zb',
  vendor: 'TuYa',
  whiteLabel: [
    {
      vendor: 'EPTTECH',
      model: 'TLC2206-ZB',
    },
  ],
  description: 'EPTTECH Tank Level Monitor Zigbee',
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  onEvent: tuya.onEventSetTime,
  configure: tuya.configureMagicPacket,
  exposes: [
    e.numeric('liquid_level_percent', ea.STATE)
      .withUnit('%')
      .withDescription('Liquid level percentage'),
    e.numeric('liquid_depth', ea.STATE)
      .withUnit('m')
      .withDescription('Liquid Depth'),
    e.enum('liquid_state', ea.STATE, ['low', 'normal', 'high'])
      .withDescription('Liquid State'),
    // =============== Found but non functional ===============
    /*
    e.numeric('max_set', ea.STATE_SET)
      .withUnit('%')
      .withDescription('Liquid max percentage')
      .withValueMin(0)
      .withValueMax(100)
      .withValueStep(1),
    e.numeric('mini_set', ea.STATE_SET)
      .withUnit('%')
      .withDescription('Liquid minimal percentage')
      .withValueMin(0)
      .withValueMax(100)
      .withValueStep(1),
    e.numeric('installation_height', ea.STATE_SET)
      .withUnit('mm')
      .withDescription('Height from sensor to tank bottom')
      .withValueMin(100)
      .withValueMax(2500)
      .withValueStep(1),
    e.numeric('liquid_depth_max', ea.STATE_SET)
      .withUnit('mm')
      .withDescription('Height from sensor to liquid level')
      .withValueMin(100)
      .withValueMax(1000)
      .withValueStep(1),
    */
  ],
  meta: {
    tuyaDatapoints: [
      [1, 'liquid_state', tuya.valueConverterBasic.lookup({'low': 1, 'normal': 0, 'high': 2})],
      [2, 'liquid_depth', tuya.valueConverter.divideBy100],
      [22, 'liquid_level_percent', tuya.valueConverter.raw],
      // =============== Found but non functional ===============
      /*
      [7, 'max_set', tuya.valueConverter.raw],
      [8, 'mini_set', tuya.valueConverter.raw],
      [19, 'installation_height', tuya.valueConverter.raw],
      [21, 'liquid_depth_max', tuya.valueConverter.raw],
      */
    ],
  },
};

module.exports = definition;
```

NDNELSON, 2024-02-27:
It is working now ?

rvdlinden, 2024-02-27:
I only have it working via local tuya and using a tuya zigbee gateway. 
Directly adding the device to Home Assistant via a zigbee dongle only shows the
the depth, percentage and state values, but all configurable fields like height, min and max are failing.

NDNELSON, 2024-02-28:
Have you tried setup by the new way tuya in the first February home assistant release ? https://www.home-assistant.io/integrations/tuya/

rvdlinden, 2024-02-28:
Yes, I use that, but the device is still unsupported 

[image]

blouinp, 2024-04-12:
I have the same problem.... unsupported :(

blouinp, 2024-04-13:
> I only have it working via local tuya and using a tuya zigbee gateway. Directly adding the device to Home Assistant via a zigbee dongle only shows the the depth, percentage and state values, but all configurable fields like height, min and max are failing.

HI how you did it?

rvdlinden, 2024-04-13:
So far I can only use it reliable using a tuya zigbee hub and add the zigbee water tank meter to ir as a subdivide.

In Home Assistant using this localtuya fork https://xzetsubou.github.io/hass-localtuya/

With the information in this thread you can configure the various entities in localtuya

TonioClos, 2024-05-07:
Hey! did you still trying to do it directly with zigbee2mqtt or just stayed with local tuya?

rvdlinden, 2024-05-08:
Still local tuya

TonioClos, 2024-05-09:
Okay I´ll stick with HA without local tuya and try then, thank you!

gregorobreza, 2024-05-18:
Hi, I am also also interested in adding support for this device trough Zigbee2MQTT. Same as you, I am struggling with other 4 datapoints. For all four I am getting an error  "No converter available". 
I would love to contribute If someone can point me in the right direction how to even debug in such cases.

TheRealWaldo, 2024-05-21:
Like everyone here, I'd like to get this working and added as an officially supported device!

gregorobreza, 2024-05-25:
I thing I finally figure it out. It took a while for me to better understand how to work with zigbee-herdsman-conversters since it is not really well documented and I am relatively new to Zigbee2MQTT. 

I just added custom ```toZigbee``` converter at the top of the file and used ```tuya.sendDataPointValue()```. This [article](https://medium.com/@omaslyuchenko/hello-zigbee-world-part-18-custom-zigbee2mqtt-external-converter-1a666db52d44) on medium really helped me understand converters better. And of course searching trough examples of supported devices to se how its done for similar devices.

So here is my current converter:

```js
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const {} = require("zigbee-herdsman-converters/lib/modernExtend");
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const utils = require("zigbee-herdsman-converters/lib/utils");

const dp = {
  instalationHeight: 19,
  liquidDepthMax: 21,
  maxSet: 7,
  miniSet: 8,
};

const tzdatapoints = {
  key: [
    "installation_height",
    "liquid_depth_max",
    "max_set",
    "mini_set",
  ],
  convertSet: async (entity, key, value, meta) => {
    switch (key) {
      case "installation_height": {
        await tuya.sendDataPointValue(entity, dp.instalationHeight, value);
        break;
      }
      case "liquid_depth_max": {
        await tuya.sendDataPointValue(entity, dp.liquidDepthMax, value);
        break;
      }
      case "max_set": {
        await tuya.sendDataPointValue(entity, dp.maxSet, value);
        break;
      }
      case "mini_set": {
        await tuya.sendDataPointValue(entity, dp.miniSet, value);
        break;
      }
    }
  },
};

const definition = {
  // Since a lot of TuYa devices use the same modelID, but use different datapoints
  // it's necessary to provide a fingerprint instead of a zigbeeModel
  fingerprint: [
    {
      // The model ID from: Device with modelID 'TS0601' is not supported
      // You may need to add \u0000 at the end of the name in some cases
      modelID: "TS0601",
      // The manufacturer name from: Device with modelID 'TS0601' is not supported.
      manufacturerName: "_TZE200_lvkk0hdg",
    },
  ],
  model: "TS0601_tlc2206zb",
  vendor: "TuYa",
  whiteLabel: [
    {
      vendor: "EPTTECH",
      model: "TLC2206-ZB",
    },
  ],
  description: "EPTTECH Tank Level Monitor Zigbee",
  fromZigbee: [tuya.fz.datapoints],
  // toZigbee: [tuya.tz.datapoints],
  toZigbee: [tzdatapoints],
  onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
  configure: tuya.configureMagicPacket,
  exposes: [
    // Here you should put all functionality that your device exposes
    e
      .numeric("liquid_level_percent", ea.STATE)
      .withUnit("%")
      .withDescription("Liquid level percentage"),
    e
      .numeric("liquid_depth", ea.STATE)
      .withUnit("m")
      .withDescription("Liquid Depth"),
    e
      .enum("liquid_state", ea.STATE, ["low", "normal", "high"])
      .withDescription("Liquid State"),

    e
      .numeric("installation_height", ea.STATE_SET)
      .withUnit("mm")
      .withDescription("Height from sensor to tank bottom")
      .withValueMin(100)
      .withValueMax(3000)
      .withValueStep(1),
    e
      .numeric("mini_set", ea.STATE_SET)
      .withUnit("%")
      .withDescription("Liquid minimal percentage")
      .withValueMin(0)
      .withValueMax(100)
      .withValueStep(1),
    e
      .numeric("max_set", ea.STATE_SET)
      .withUnit("%")
      .withDescription("Liquid max percentage")
      .withValueMin(0)
      .withValueMax(100)
      .withValueStep(1),
    e
      .numeric("liquid_depth_max", ea.STATE_SET)
      .withUnit("mm")
      .withDescription("Height from sensor to liquid level")
      .withValueMin(100)
      .withValueMax(2000)
      .withValueStep(1),
  ],
  meta: {
    // All datapoints go in here
    tuyaDatapoints: [
      [22, "liquid_level_percent", tuya.valueConverter.raw],
      [2, "liquid_depth", tuya.valueConverter.divideBy100],
      [
        1,
        "liquid_state",
        tuya.valueConverterBasic.lookup({ low: 1, normal: 0, high: 2 }),
      ],
      [19, "installation_height", tuya.valueConverter.raw],
      [7, "max_set", tuya.valueConverter.raw],
      [8, "mini_set", tuya.valueConverter.raw],
      [21, "liquid_depth_max", tuya.valueConverter.raw],
    ],
  },
  extend: [
    // A preferred new way of extending functionality.
  ],
};

module.exports = definition;
```

I would appreciate if someone else could test configuration and give feedback. Then we could maybe move further and suggest it as an officially supported device.

rvdlinden, 2024-05-25:
I have removed my tuya gateway and local tuya in HA and connected the TLC ZB directly to HA again.

So far it works like a charm

petos, 2024-05-25:
@gregorobreza 
I can test, just tell me what to paste and where,...

TonioClos, 2024-05-31:
Working perfectly, thank you so much for the code n_n @gregorobreza

ocigam69, 2024-06-10:
Hello,
@gregorobreza Thanks for the code. Very good job!
I have a problem. I’ve created the file and placed it next to the configuration.yaml file of Z2M, but when configuring it in the external converters, I encounter an error: `external_converters must be array`
I’ve also added the following entry to the configuration.yaml file of Z2M, but it still hasn’t worked:

```
external_converters:
  - bidon.js
```

ocigam69, 2024-06-10:
After many tests, I just needed to leave a blank line at the end of the file. 😊
Thanks @gregorobreza !!!

gregorobreza, 2024-06-10:
Hm interesting 🤔. Happy to hear that you solved it!👍

SentenciaLaMalice, 2024-06-12:
Hi!. Somebody knows if this new device is going to be integrated in the native Z2M in a close future?

I have just purchased it and I don't feel very confortable yet touching the guts of Z2M so, if it's comming to Z2M, I will just wait a little. 

Thanks in advance!.

ivanfmartinez, 2024-06-14:
Received sensor today, and was detected fine by @gregorobreza definition  https://github.com/Koenkk/zigbee2mqtt/issues/21015#issuecomment-2131168644

SentenciaLaMalice, 2024-06-17:
I decided to give a try to @gregorobreza code a try (thank you very much sir) and it worked great. The sensor started reporting perfectly. But, sadly, as soon as I configured the levels of my setup using the "Expose" tab on Z2M the device does not report the level anymore. I was not able to fix it, I keep getting zero per cent :(

SentenciaLaMalice, 2024-06-17:
Just to add that I just noticed on the debug logs of Z2M that the device is sending a huge ammount of messages per minute with its current status. Didn't count them but I estimate 100 message per minute easily. I let one log line here for if it helps

2024-06-17 23:00:49MQTT publish: topic 'zigbee2mqtt/nivel_gasoil', payload '{"installation_height":1045,"linkquality":87,"liquid_depth":0,"liquid_depth_max":100,"liquid_level_percent":0,"liquid_state":"low","max_set":95,"mini_set":20}'

jtscott, 2024-06-17:
> Just to add that I just noticed on the debug logs of Z2M that the device is sending a huge ammount of messages per minute with its current status.

I noticed this as well, the Availability reflects this as the device is always 'just now' and doesn't seem to quit. Would be nice to be able to set the report interval if possible.

gregorobreza, 2024-06-19:
> I decided to give a try to @gregorobreza code a try (thank you very much sir) and it worked great. The sensor started reporting perfectly. But, sadly, as soon as I configured the levels of my setup [...]

Thats strange... Are you sure your configured levels make sense and are set correctly based on sensor instructions?

gregorobreza, 2024-06-19:
> I noticed this as well, the Availability reflects this as the device is always 'just now' and doesn't seem to quit. Would be nice to be able to set the report interval if possible.

I am not really sure that device support setting reporting interval. Device behaves in the same manner when used with tuya hub and it seem this is expected behaviour. But definitely worth to dig a bit deeper if even possible.

ivanfmartinez, 2024-08-01:
After some time installed I made more tests and adjusted the definition. But it still have problems.

The parameters that are defined before as min and max, are related to the state, and I have changed the names.
low_set and high_set now
when percentage is below low_set state becames "low" and percentage over high_set state becomes "high"

I did not understand exactly how that convertSet works because the sendDataPointValue for low_set uses dp.highSet and sendDataPointValue for high_set uses dp.lowSet to work correctly.

The instalation_height is working, changing it changed the liquid_level correctly.

The distance from sensor "liquid_depth_max" are not working, changing it does not affect the liquid level.

And percentage is not correct because of this non working parameter, percentage should consider this parameter to be calculated correctly.

Someone can help to fix this and make sensor correct ?

```js
// based on 
//    https://github.com/Koenkk/zigbee2mqtt/issues/21015#issuecomment-2131168644
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const {} = require("zigbee-herdsman-converters/lib/modernExtend");
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const utils = require("zigbee-herdsman-converters/lib/utils");

const dp = {
  instalationHeight: 19,
  liquidDepthMax: 21,
  lowSet: 7,
  highSet: 8,
};

const tzdatapoints = {
  key: [
    "installation_height",
    "liquid_depth_max",
    "low_set",
    "high_set",
  ],
  convertSet: async (entity, key, value, meta) => {
    switch (key) {
      case "installation_height": {
        await tuya.sendDataPointValue(entity, dp.instalationHeight, value);
        break;
      }
      case "liquid_depth_max": {
        await tuya.sendDataPointValue(entity, dp.liquidDepthMax, value);
        break;
      }
      case "low_set": {
        // Using highSet here is strange but works, inverting does not work correctly
        await tuya.sendDataPointValue(entity, dp.highSet, value);
        break;
      }
      case "high_set": {
        // Using lowSet here is strange but works, inverting does not work correctly
        await tuya.sendDataPointValue(entity, dp.lowSet, value);
        break;
      }
    }
  },
};

const definition = {
  // Since a lot of TuYa devices use the same modelID, but use different datapoints
  // it's necessary to provide a fingerprint instead of a zigbeeModel
  fingerprint: [
    {
      // The model ID from: Device with modelID 'TS0601' is not supported
      // You may need to add \u0000 at the end of the name in some cases
      modelID: "TS0601",
      // The manufacturer name from: Device with modelID 'TS0601' is not supported.
      manufacturerName: "_TZE200_lvkk0hdg",
    },
  ],
  model: "TS0601_tlc2206zb_external",
  vendor: "TuYa",
  whiteLabel: [
    {
      vendor: "EPTTECH",
      model: "TLC2206-ZB",
    },
  ],
  description: "EPTTECH Tank Level Monitor Zigbee (external)",
  fromZigbee: [tuya.fz.datapoints],
  // toZigbee: [tuya.tz.datapoints],
  toZigbee: [tzdatapoints],
  onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
  configure: tuya.configureMagicPacket,
  exposes: [
    // Here you should put all functionality that your device exposes
    e
      .numeric("liquid_level_percent", ea.STATE)
      .withUnit("%")
      .withDescription("Liquid level percentage"),
    e
      .numeric("liquid_depth", ea.STATE)
      .withUnit("m")
      .withDescription("Liquid Depth"),
    e
      .enum("liquid_state", ea.STATE, ["low", "normal", "high"])
      .withDescription("Liquid State"),
    e
      .numeric("installation_height", ea.STATE_SET)
      .withUnit("mm")
      .withDescription("Height from sensor to tank bottom")
      .withValueMin(100)
      .withValueMax(3000)
      .withValueStep(1),
    e
      .numeric("low_set", ea.STATE_SET)
      .withUnit("%")
      .withDescription("Liquid percentage to set low state (below this value)")
      .withValueMin(0)
      .withValueMax(100)
      .withValueStep(1),
    e
      .numeric("high_set", ea.STATE_SET)
      .withUnit("%")
      .withDescription("Liquid percentage to set high state (above this value)")
      .withValueMin(0)
      .withValueMax(100)
      .withValueStep(1),
    e
      .numeric("liquid_depth_max", ea.STATE_SET)
      .withUnit("mm")
      .withDescription("Height from sensor to liquid level")
      .withValueMin(100)
      .withValueMax(2000)
      .withValueStep(1),
  ],
  meta: {
    // All datapoints go in here
    tuyaDatapoints: [
      [22, "liquid_level_percent", tuya.valueConverter.raw],
      [2, "liquid_depth", tuya.valueConverter.divideBy100],
      [1, "liquid_state", tuya.valueConverterBasic.lookup({ low: 1, normal: 0, high: 2 }),],
      [19, "installation_height", tuya.valueConverter.raw],
      [7, "low_set", tuya.valueConverter.raw],
      [8, "high_set", tuya.valueConverter.raw],
      [21, "liquid_depth_max", tuya.valueConverter.raw],
    ],
  },
  extend: [
    // A preferred new way of extending functionality.
  ],
};

module.exports = definition;

```

rvdlinden, 2024-08-01:
I have the Wifi version and also the zigbee version of the tank level monitor. 

When I initially used the Wifi device via Tuya app and Tuya, all settings worked as how they were described in the manual of the TLC.

When I later bought the zigbee version, inside the package was the same manual as from the Wifi version, but it worked completely different. 

I'm pretty sure that although the device looks identical and you'd only expect the connection to be different, they are actually different devices with completely different firmware and therefore also different behavior.

DrTrotty, 2024-08-05:
> So far I can only use it reliable using a tuya zigbee hub and add the zigbee water tank meter to it as a subdevice.
> 
> In Home Assistant using this localtuya fork https://xzetsubou.github.io/hass-localtuya/
> 
> With the information in this thread you can configure the various entities in localtuya

Hello. May I know how it works stably? If I disconnect it from the power supply - approx. For 20 minutes, for example - my gateway or my water meter, the application still sees it as online, the Home assistant lost it after the first power cut...:C.
I also have a wifi water meter in localtuya..

DrTrotty, 2024-08-06:
> @gregorobreza <https://github.com/gregorobreza> this works perfectly for
> me. Thank you for doing it. Are you going to open a pull request for it?

I connected the meter to the zigbee network. I could not add the external
configuration file in the device properties. I created a .js file with your
code. I wrote in the zigbee config file what I saw in the official
description. Nothing happened. I cried... do I need to install a converter?
If so, how? Thank you for taking care of me!

Tisztelettel: Végvári Oszkár
              Tel.: +36203441154

Mark Roberts ***@***.***> (időpont: 2024. aug. 6., K, 19:00)
ezt írta:

markxroberts, 2024-08-06:
The external converter needs to be in the same folder as your ```configuration.yaml``` file
The format of the configuration file is important too:
```
external_converters:
  - filename.js
```
The code that worked for me is from this post: https://github.com/Koenkk/zigbee2mqtt/issues/21015#issuecomment-2131168644

DrTrotty, 2024-08-07:
Thanks. I'm looking forward to it. I apply.

Petos Safarik ***@***.***> ezt írta (időpont: 2024. aug. 7.,
Sze, 6:47):

> Hello,
> it's configuration.yaml, but not HA's but ZB's. Where HA's configuration
> file is, go to zigbee2mqtt directory. There is also configuration.yaml.
> HERE is the right spot you are going to work.
>
>    1. create new file with name water-level.js and paste there the config
>    mentioned in this discussion
>    2. edit configuration.yaml and on the very TOP of the file paste:
>
> external_converters: - water-level.js
>
>    3. Save. Restart ZB component
>    4. profit

gregorobreza, 2024-08-07:
> @gregorobreza this works perfectly for me. Thank you for doing it. Are you going to open a pull request for it?

I am quite busy at the moment but yes I am planning to. I am also planning to do some more tests on my water tank.

ivanfmartinez, 2024-08-07:
> > @gregorobreza this works perfectly for me. Thank you for doing it. Are you going to open a pull request for it?
> 
> I am quite busy at the moment but yes I am planning to. I am also planning to do some more tests on my water tank.

@gregorobreza  when doing tests please check the parameter **liquid_depth_max**, for me it does not work because the **liquid_depth** and **liquid_level_percent** does not change after I change **liquid_depth_max**.

DrTrotty, 2024-08-07:
Thank you very much for your help! It worked. Can I delete the first two lines from the config.yaml in this case?
external_converters: 
  - water-level.js
A thousand thanks again!!! I'm also going to test the maximum level...perhaps this value cannot be fixed in the external configuration?

petos, 2024-08-08:
>  Can I delete the first two lines from the config.yaml in this case?

If you want to undo, then you can either delete these 2 lines OR comment out (next time you wouln't have to figure it out again. Or just make it empty:
`external_converters: [ ]`

DrTrotty, 2024-08-08:
> > Can I delete the first two lines from the config.yaml in this case?
> 
> If you want to undo, then you can either delete these 2 lines OR comment out (next time you wouln't have to figure it out again. Or just make it empty: `external_converters: [ ]`

Thanks.

DrTrotty, 2024-08-08:
Unfortunately, the liquid_depth_max setting doesn't work for me either.

jesse-kaufman, 2024-08-08:
Changing liquid_depth_max doesn't seem to do anything when using the device via localtuya and a Tuya hub, so I wonder if that isn't a firmware issue

ivanfmartinez, 2024-08-09:
I have contacted the Manufaturer ang got the device protocol, I will only have time to check on sunday, but maybe  someone else can verify it before.

[lowpower_protocol_lvkk0hdg.pdf](https://github.com/user-attachments/files/16559654/lowpower_protocol_lvkk0hdg.pdf)

DrTrotty, 2024-08-10:
I checked the diary. The sesor reports every 1 second.

DrTrotty, 2024-08-11:
I don't know, someone indicated, setting the maximum liquid level doesn't work for me with the tuya application either. No matter how much I change, the value of the percentage does not change

jesse-kaufman, 2024-08-11:
> I don't know, someone indicated, setting the maximum liquid level doesn't work for me with the tuya application either. No matter how much I change, the value of the percentage does not change

Yup, I ended up just disabling everything except the current depth in meters, setup a template sensor to calculate water level %, and setup my own alerts separate from the min/max settings on the device itself 

I also set debounce to 1 and added liquid_depth and linkquality to the debounce_ignore ... the device is much less chatty now .. it sends updates anywhere from 2-20 seconds apart, rather than at least one per second (if not more)

DrTrotty, 2024-08-13:
> template sensor

Hi. If it is not a big request, could you publish the code of the template sensor. It would save me a lot of time...I would appreciate it.

jesse-kaufman, 2024-08-13:
Here's the template code I'm using .. it takes the moving average (a filter sensor), converts to inches, and then uses input_number entities for min/max depth to calculate the percent .. in my case, I wanted it to show 0% when there's about 2.5 inches of liquid in my tank, so that's what I set min_depth to

```jinja
{% set depth = states('sensor.flower_water_level_sensor_current_depth_2m_moving_average') | float(-1) * 39.37 %}
{% set max_depth = states('input_number.flower_water_level_sensor_max_water_depth') | float(-1) %}
{% set min_depth = states('input_number.flower_water_level_sensor_min_water_depth') | float(-1) %}

{% if depth > 0 %}
  {% set percent = (((depth - min_depth) / (max_depth - min_depth)) * 100) | round %}
  {{ ([0, percent, 100]|sort)[1] }}
{% endif %}
```

DrTrotty, 2024-08-13:
Thank you very much!

DrTrotty, 2024-08-16:
I bought the device from Aliexpress and asked for a refund.

c4m0-lab, 2024-09-07:
thanks for this - its awsome! 
Its my first unkown device. 

I only have the problem that no image is displayed and when the power is "?". 
Does anyone know how to get this clean?

after this I need to look how to get the device not to be so chatty :)

DrTrotty, 2024-09-08:
Hello. I have been corresponding with the manufacturer for some time. Does anyone have liquid_depth_max setting working? If so, did you buy it on Ali Express or directly from the manufacturer? The person I correspond with says it works. He also sent a video...let's say, it doesn't show whether the device is Wi-Fi, He says it's zigbee. That's our firmware version.

DrTrotty, 2024-09-08:
This value does not work on my phone in the Tuya application either. Would it be possible to use a different gateway? There are miracles...

c4m0-lab, 2024-09-08:
It was a long way yesterday.. but I think that works for me..
its zigbee and buyed by amazon.
But I am still in the testing.. 

[image]

DrTrotty, 2024-09-08:
When doing tests please check the parameter *liquid_depth_max*, for me it
does not work because the *liquid_depth* and *liquid_level_percent* does
not  after I change *liquid_depth_max*.

Tisztelettel: Végvári Oszkár
              Tel.: +36203441154

c4m0-lab ***@***.***> (időpont: 2024. szept. 8., V, 9:39) ezt
írta:

c4m0-lab, 2024-09-08:
first test worked. 
I cannot empty it, so I have to wait until it does it by itself (1-2 weeks). But by testing before the installation, it took some time but did it. I had to enter the value a few times one after the other until he took it.

DrTrotty, 2024-09-09:
In case anyone else is interested...more correspondence with the manufacturer, they swear they checked the stock, all their zigbees devices work perfectly...maybe I bought a fake on ali express? When I get to the end, I'll order one from them, and I'll write about the progress.

c4m0-lab, 2024-09-14:
after testing it a while it works :) 

the only problem is it seems to send every second an update ...
don't know how to fix that..

Vinsssounet, 2024-09-16:
Hi there !

I'd like to purchase this device to use with Home Assistant.

I'm reading different things on this thread.

It seems the liquid depth max setting does not work for some people, but there's a workaround by using the current depth and calculating the percentage with a custom sensor ?

Are some people actually able to use thé device as intended ?

jesse-kaufman, 2024-09-16:
Yes, max depth doesn’t change any of the sensor readings but it really isn’t necessary when you can create your own template sensor to calculate the % .. I’m using the device regularly, rely on it to notify me when the water is low, and it’s been working great. The max depth thing not working doesn’t affect my setup at all

Vinsssounet, 2024-09-16:
Thank you, I'm gonna give it a a go 😉

DrTrotty, 2024-09-21:
Hello. Correspondence...correspondence, I received the new device. I tried it with the tuya app, so far all settings are working perfectly. Zigbee2mqtt is coming next week...

DrTrotty, 2024-09-23:
Hello. All values ​​of the new device work correctly even under HA. :D Finally!

[64 more replies]

closed 2025-12-06 by github-actions[bot]