Koenkk/zigbee2mqtt #21016: [New device support]: GEWISS 1513

anthopocket, 2024-01-25:
### Link

https://www.gewiss.com/fr/fr/products/product.1000000.1008169.GWA1513

### Database entry

Debug 2024-01-25 22:26:29Received Zigbee message from '0x0015bc001e004416', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2017-12-07 03:07"}' from endpoint 35 with groupID 0 Info 2024-01-25 22:26:29MQTT publish: topic 'zigbee2mqtt/0x0015bc001e004416', payload '{"device":{"dateCode":"2017-12-07 03:07","friendlyName":"0x0015bc001e004416","ieeeAddr":"0x0015bc001e004416","manufacturerID":6548,"manufacturerName":"Gewiss","networkAddress":11016,"powerSource":"Battery","type":"EndDevice","zclVersion":1},"last_seen":"2024-01-25T21:26:29.626Z","linkquality":116}' Debug 2024-01-25 22:26:29Received Zigbee message from '0x0015bc001e004416', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 35 with groupID 0 Info 2024-01-25 22:26:29MQTT publish: topic 'zigbee2mqtt/0x0015bc001e004416', payload '{"device":{"dateCode":"2017-12-07

### Comments

GEWISS 1513

### External defintion

```shell
const {batteryPercentage, onOff, temperature, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['GWA1513_WindowSensor'],
    model: 'GWA1513_WindowSensor',
    vendor: 'Gewiss',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), onOff({powerOnBehavior: false}), temperature(), identify()],
};

module.exports = definition;
```

anthopocket, 2024-02-18:
hi

it's the same of

https://www.zigbee2mqtt.io/devices/WISZB-138.html

the name and model change.

thanks you

Koenkk, 2024-02-18:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/21014

Koenkk, 2024-02-18:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-18 by Koenkk

anthopocket, 2024-02-18:
it doesnt ok.

try it :
https://www.zigbee2mqtt.io/devices/WISZB-120.html
or
https://www.zigbee2mqtt.io/devices/WISZB-121.html

on the net, i find

`Gewiss offers a wide range of other ZigBee based devices such as smoke detectors, window/door sensors. Those devices however are rebranded Develco devices and supported by the Develco ZigBee integration plugin.

anthopocket, 2024-02-18:
[image]

anthopocket, 2024-02-18:
it's ok
good job
thanks you