Koenkk/zigbee2mqtt #21018: Error when changing Measurement Pool Interval in Tuya BSD29_1

felipebraz, 2024-01-26:
### What happened?

I'm trying to change an specific configuration (Measurement Pool Interval) of the my Tuya BSD29_1

[image]

After I added a value on this field this error start appearing: "Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()"

I already tried to go back to original value (empty) and filling the mentioned field (Power calibration) but this message keeps constantly appearing, even going back to the default values (all empty)

[image]

### What did you expect to happen?

The pool interval being updated.

### How to reproduce it (minimal and precise)

Change the value on this field.

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

20221226

### Adapter

Sonoff zigbee 3.0 usb dongle plus

### Setup

x86-64 containerized

felipebraz, 2024-01-26:
Oh and changing the value the Measurement Pool Interval doesn't affect the update frequency.

nARN, 2024-02-12:
I believe the error is not specific to the polling interval. It is caused by the empty power calibration field, right under polling. I suspect you focused on it after changing the polling interval and then unfocused, which is when, apparently, the UI attempt to update the configuration.

In this case the problem is that upon update an empty field is treated as an empty string instead of "use the default value", which it represented originally.

nARN, 2024-02-12:
Regarding the apparent lack of reaction to your changes, it seems that you need to restart zigbee2mqtt to apply the new polling interval. I filed #21371 for that.

Skallwar, 2024-02-12:
I noticed that anything under 30 seconds will not report anything back anymore

nARN, 2024-02-12:
@Skallwar That's not my experience. After a restart, values as low as 1 second work like a charm. I actually see updates every second.

FWIW, my plug is model `TS011F`, manufacturer `_TZ3000_okaz9tjs`, app version `100`, which makes it `TS011F_plug_3`, IIUC.

Skallwar, 2024-02-12:
I did not find the app version but other than that I have the same plug. I will try to put it closer to my usb dongle, maybe that's the issue, I have a 79 which is a tad low

Skallwar, 2024-02-12:
Okay my bad, that was just a bad connection

closed 2024-08-12 by felipebraz