Koenkk/zigbee2mqtt #21024: Zigbee network crashing

ckegels, 2024-01-26:
### What happened?

every day my zigbee network will crash ( i am using a home assistant yellow with its silicon labs zigbee chip)
It started happening after installing a zigbee thermostat made by moes.
after checking logs it seems to be spamming an insane amount, about 10 times a second.
any ways to stop this device from doing that?

### What did you expect to happen?

i would like to have a stable network

### How to reproduce it (minimal and precise)

plug in thermostat

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

add on home assistant

### Debug log

[log1.txt](https://github.com/Koenkk/zigbee2mqtt/files/14064450/log1.txt)

Nerivec, 2024-01-29:
https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#stop-message-flooding

Guessing this is that issue?

ckegels, 2024-01-29:
> https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#stop-message-flooding
> 
> Guessing this is that issue?

Thanks for the reply, i tried debounce 1 but that didnt stop it from flooding my zigbee network wich it than still crashes. but i will give it another go. thanks

kokireg, 2024-02-12:
I have the same problem with moes bht-006

rom169, 2024-05-17:
Same problem here since using Moes BHT-006 which are recognized as Moes BHT-002-GCLZB.

closed 2025-01-07 by github-actions[bot]