Koenkk/zigbee2mqtt #21027: Failed to configure Xiaomi WSDCGQ12LM

difrost, 2024-01-26:
### What happened?

Hi,

just switched today to the latest dev build (due to fixes that were included for other devices) while all devices work correctly the Xiaomi sensor is throwing following error after boot:
```
Zigbee2MQTT:error 2024-01-26 12:48:01: Failed to configure 'BSensor', attempt 3 (Error: {"address":11944,"clusterId":32801,"sequence":60} after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:663:43)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:566:13
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:1675:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```
I'm setting my own precision settings on this sensor. 

A grep for the BSensor (excluding initial boot part):
```
Zigbee2MQTT:info  2024-01-26 12:26:24: MQTT publish: topic 'zigbee2mqtt/BSensor', payload '{"battery":99,"device_temperature":7,"humidity":63,"linkquality":255,"power_outage_count":8,"pressure":1005,"temperature":8.8,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2999}'
Zigbee2MQTT:info  2024-01-26 12:38:40: Configuring 'BSensor'
Zigbee2MQTT:info  2024-01-26 12:38:40: MQTT publish: topic 'zigbee2mqtt/BSensor', payload '{"battery":99,"device_temperature":7,"humidity":63,"linkquality":255,"power_outage_count":8,"pressure":1005,"temperature":9.3,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2999}'
Zigbee2MQTT:error 2024-01-26 12:38:50: Failed to configure 'BSensor', attempt 1 (Error: {"address":11944,"clusterId":32801,"sequence":43} after 10000ms
Zigbee2MQTT:info  2024-01-26 12:42:30: Configuring 'BSensor'
Zigbee2MQTT:info  2024-01-26 12:42:30: MQTT publish: topic 'zigbee2mqtt/BSensor', payload '{"battery":99,"device_temperature":7,"humidity":63,"linkquality":255,"power_outage_count":8,"pressure":1004,"temperature":9.3,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2999}'
Zigbee2MQTT:error 2024-01-26 12:42:40: Failed to configure 'BSensor', attempt 2 (Error: {"address":11944,"clusterId":32801,"sequence":54} after 10000ms
Zigbee2MQTT:info  2024-01-26 12:47:48: Configuring 'BSensor'
Zigbee2MQTT:info  2024-01-26 12:47:48: MQTT publish: topic 'zigbee2mqtt/BSensor', payload '{"battery":96,"device_temperature":9,"humidity":60,"linkquality":255,"power_outage_count":8,"pressure":1004,"temperature":9.6,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2994}'
Zigbee2MQTT:error 2024-01-26 12:48:01: Failed to configure 'BSensor', attempt 3 (Error: {"address":11944,"clusterId":32801,"sequence":60} after 10000ms
Zigbee2MQTT:info  2024-01-26 12:51:19: MQTT publish: topic 'zigbee2mqtt/BSensor', payload '{"battery":96,"device_temperature":9,"humidity":60,"linkquality":255,"power_outage_count":8,"pressure":1004,"temperature":9.6,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2994}'
Zigbee2MQTT:info  2024-01-26 12:56:48: MQTT publish: topic 'zigbee2mqtt/BSensor', payload '{"battery":96,"device_temperature":9,"humidity":60,"linkquality":255,"power_outage_count":8,"pressure":1004,"temperature":9.8,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2994}'
Zigbee2MQTT:info  2024-01-26 13:11:03: MQTT publish: topic 'zigbee2mqtt/BSensor', payload '{"battery":96,"device_temperature":9,"humidity":60,"linkquality":255,"power_outage_count":8,"pressure":1004,"temperature":9.8,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2994}'
```

### How to reproduce it (minimal and precise)

I've done a restart of the service but so far no error. Originally it also popped up 30 min after restart to latest-dev.

### Zigbee2MQTT version

docker.io/koenkk/zigbee2mqtt@sha256:1a7de41c99c8cb188c6508d4bd3dfef4a51bc14c0b740b599f5ebf17e2a023bd

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle Plus-E

### Setup

x86-64, Deployment in K3s

satzmann666, 2024-02-05:
Regarding the `No converter available for 'sensor_temp'` issue, in my case with exactly the same messages in logs I found that the topic for posting external temperature to Aqara TRVs ([Aqara SRTS-A01](https://www.zigbee2mqtt.io/devices/SRTS-A01.html)) has been changed.

It used to be `zigbee2mqtt/FRIENDLY_NAME/set/sensor_temp`, but changed to `zigbee2mqtt/FRIENDLY_NAME/set/external_temperature_input`

In case you have automations set up that post current temperature value from sensor to a TRV you need to change the automations or the blueprint they implement.

difrost, 2024-02-07:
> Regarding the `No converter available for 'sensor_temp'` issue, in my case with exactly the same messages in logs I found that the topic for posting external temperature to Aqara TRVs ([Aqara [...]

Thanks. I found that out when I was trying to debug the original issue and found the commit that changed the endpoint for external temperature. 

Regarding the "Failed to configure ..." this issue is still present in 1.35.2. If I'll find some free quants of time I'll try to investigate that further but so far no luck - I'm not a JS/TS dev.

cpaius, 2024-02-23:
Same issue as #21154 by the looks of it.

closed 2024-09-21 by github-actions[bot]