Koenkk/zigbee2mqtt #21030: Z2M addon uses 2GB of memory (excessive and growing RAM usage)

CrazyCoder, 2024-01-26:
### What happened?

Observing high memory usage for the Z2M addon on HA Yellow device (4GB total RAM).

1.67GB / 3.75GB (44% of RAM used by the addon alone).

After addon restart, the memory usage is normal (150MB, <5%), but grows over time.

### What did you expect to happen?

The memory usage should be reasonable and not grow over time.

### How to reproduce it (minimal and precise)

Run addon on HA with 50+ zigbee devices for a day?

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

Add-on on HA Yellow

### Debug log

There is nothing related in the logs.

kaigame, 2024-02-04:
I had this issue also, but since updating to 1.35.2-1, memory usage has been pretty stable.

Koenkk, 2024-03-12:
Could you check if this is solved with the [new ember](https://github.com/Koenkk/zigbee2mqtt/discussions/21462) adapter?

kaigame, 2024-03-12:
I'll try and check this evening, as I'm having the same issue with the new release, downgraded at the moment for stability.

kaigame, 2024-03-20:
Sorry for the late reply, had nothing but trouble with the ember adapter, but since switching back to EZSP with the 7.4.1 firmware it's been pretty stable.

CrazyCoder, 2024-03-20:
Interesting. Ember seems to be working fine for me, memory usage has increased from 2.3% to 3.4% over a week, and I did not notice any problems so far.

Need more time to see if it really fixes the original memory leak.

kaigame, 2024-03-26:
As soon i start it, i get this:

[14:12:07] INFO: Preparing to start...
[14:12:07] INFO: Socat not enabled
[14:12:08] INFO: Starting Zigbee2MQTT...
======== Ember Adapter Starting ========
======== EZSP starting ========
======== ASH NCP reset ========
======== ASH starting ========
Received frame with CRC error
Received frame with CRC error
======== ASH connected ========
======== ASH started ========
======== EZSP started ========
[STACK STATUS] Network up.
[INIT TC] NCP network matches config.
[CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[STACK STATUS] Network opened.
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=110 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=111 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=112 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=113 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=114 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=115 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=116 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=117 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=118 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=119 Len=7].

Not sure where i went wrong

kaigame, 2024-03-26:
added:
rtscts: false
to my config, and now i get:

[14:17:07] INFO: Starting Zigbee2MQTT...
======== Ember Adapter Starting ========
======== EZSP starting ========
======== ASH NCP reset ========
======== ASH starting ========
Received frame with CRC error
Received frame with CRC error
======== ASH connected ========
======== ASH started ========
======== EZSP started ========
[STACK STATUS] Network up.
[INIT TC] NCP network matches config.
[CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[STACK STATUS] Network opened.
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "43292".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "46430".
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=152 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=154 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=155 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=157 Len=7].
~x~> [ZCL to=25638] Failed to send request with status=NO_BUFFERS.
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=164 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=166 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=167 Len=7].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=168 Len=7].
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "54719".
Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "25638".
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=149 Len=40].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=158 Len=38].
The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=158 Len=38].

smily555, 2024-06-08:
I am still on EZSP and int work with no problems,   if I change to EMBER I also get this after a minute:

2024-06-08 21:02:27~x~> [ZCL to=31523] Failed to send request with status=NO_BUFFERS.
Error 2024-06-08 21:02:27Default response to 0xa4c138dc0db2e5a7 failed
Error 2024-06-08 21:02:27The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=97 Len=37].
Error 2024-06-08 21:02:27The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=125 Len=7].
Error 2024-06-08 21:02:27~x~> [ZCL to=56333] Failed to send request with status=NO_BUFFERS.
Error 2024-06-08 21:02:27Default response to 0xa4c138c754029313 failed
Error 2024-06-08 21:02:28The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=128 Len=7].
Error 2024-06-08 21:02:29The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=101 Len=39].
Error 2024-06-08 21:02:29The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=131 Len=7].
Error 2024-06-08 21:02:29The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=102 Len=39].
Error 2024-06-08 21:02:30The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=102 Len=38].
Error 2024-06-08 21:02:30The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=132 Len=7].
Error 2024-06-08 21:02:30The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=135 Len=7].
Error 2024-06-08 21:02:30The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=136 Len=7].
Error 2024-06-08 21:02:31The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=104 Len=20].
Error 2024-06-08 21:02:32The NCP has run out of buffers, causing general malfunction. Remediate network congestion, if present. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=138 Len=7].
...

I am afraid to  update Z2M to latest versions, as EZSP will become obsolete. 
Zigbee2MQTT version    1.37.1-dev commit: [1a12c2f](https://github.com/Koenkk/zigbee2mqtt/commit/1a12c2f)
Coordinator type   EZSP v13
Coordinator revision. 7.4.1.0 build 0

Any solutions?

closed 2024-12-30 by github-actions[bot]