Koenkk/zigbee2mqtt #21031: Zigbee2MQTT devices are not displayed in Home Assistant

H2FSpawn, 2024-01-26:
### What happened?

After Installing HA on a VM in Proxmox i installed Mosquitto broker. Then i installed the addon Zigbee2MQTT.  In Z2M i added a Phillips Hue lamp for testing and worked well (i also gave the light a frindly name). I can see and change the state in Z2M. But the device will not displayed in Home Assistant.

Here you can see the protocoll from Mosquitto broker:
[image]

It looks for me that a connection to the Z2M is etablished but then there is something wrong.

Z2M Hostname: 45df7312-zigbee2mqtt

I reinstalled both (Z2M and Mboker) serveral times with the same result.

What could be the problem? 

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

20240104.0

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,Zigbee Gateway TI CC2652P + CP2102N

### Setup

HA VM on ProxmoxVE

moipcr, 2024-01-26:
Are you sure that your broker has any authentification? Same user/password and host must be indicated in configuration.yaml. Check installation section. Other question, disconnection logs are in broker logs or in Z2M or both?

H2FSpawn, 2024-01-27:
I found a solution with your qustions. I used a youtube video for install and configuration of Z2M. There he don't used a user in his configuration. After creating a user and setup the Integration "MQTT" manually it worked.
[image]

Here a short summary of my installation maybe it helps someone else:
Create user: Config-> Persons -> TAB User-> create user
Install Add-on "Mosquito broker"
Install Integration "MQTT"
- Server: core-mosquitto
- Port: 1883
- user: (created user)
- Password: (password)

Install Add-on "Zigbee2MQTT"

- Add Respository https://github.com/zigbee2mqtt/hassio-zigbee2mqtt
- choose Zigbee2MQTT and install
- enable watchdog
- and Start

Now you can search for devices.

Hope that helps someone.

closed 2024-01-27 by H2FSpawn