Koenkk/zigbee2mqtt #21035: HS1SA HEIMAN smoke not on dashboard

mirandandrob, 2024-01-27:
### What happened?

Added HS1SA to devices but not populating in dashboard.
 work fine with ZHA apart from double beep due to 12 firmware. (Beeps randomly during fax or night have 7 units)
 Only show 2 exposes out of 4 in devices.
 Currently have a conbee2 running zha and a ezsp running z2m.
 Any ideas on what I can do to make them populate

### Zigbee2MQTT version

1.35.1-1

### Adapter firmware version

0264

### Adapter

Ezsp

### Setup

X86 add on

Koenkk, 2024-01-28:
Please create issues related to the frontend in this repository: https://github.com/nurikk/zigbee2mqtt-frontend

closed 2024-01-28 by Koenkk

mirandandrob, 2024-01-28:
Issue resolved with reset of smoke detectors and keeping awake during handshake to dongle.
Multiple tries.
Many thanks for your hard work Koenkk