Koenkk/zigbee2mqtt #21036: LEGRAND 064879

JulienLIESM, 2024-01-27:
### What happened?

The LEGRAND 064879, connected socket pairs to the network, and 10 seconds later disconnects automatically, this problem goes around in circles.
It works perfectly with the 1.35.1 version 
I update the LEGRAND firmware to the 22 version, but doesn't work

### Zigbee2MQTT version

1.35.1 commit: 3c96204

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF EFR32MG21

### Setup

Linux Debian with mosquitto 

### Debug log

Zigbee2MQTT:info  2024-01-27 08:12:33: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2024-01-27 08:12:33: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"8vg3y-1"}'
Zigbee2MQTT:warn  2024-01-27 08:12:50: Device '0x0004740000278dfb' left the network
Zigbee2MQTT:info  2024-01-27 08:12:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb"},"type":"device_leave"}'
Zigbee2MQTT:warn  2024-01-27 08:12:51: Device '0x0004740000278dfb' left the network
Zigbee2MQTT:info  2024-01-27 08:12:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb"},"type":"device_leave"}'
Zigbee2MQTT:info  2024-01-27 08:12:52: Device '0x0004740000278dfb' joined
Zigbee2MQTT:info  2024-01-27 08:12:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb"},"type":"device_joined"}'
Zigbee2MQTT:info  2024-01-27 08:12:52: Starting interview of '0x0004740000278dfb'
Zigbee2MQTT:info  2024-01-27 08:12:52: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2024-01-27 08:12:54: Successfully interviewed '0x0004740000278dfb', device has successfully been paired
Zigbee2MQTT:info  2024-01-27 08:12:54: Device '0x0004740000278dfb' is supported, identified as: Legrand Cable outlet with pilot wire and consumption measurement (064882)
Zigbee2MQTT:info  2024-01-27 08:12:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Cable outlet with pilot wire and consumption measurement","exposes":[{"access":7,"label":"Device mode","name":"device_mode","property":"device_mode","type":"binary","value_off":"pilot_off","value_on":"pilot_on"},{"access":7,"description":"Controls the target temperature of the heater, with respect to the temperature set on that heater. Possible values: comfort (target temperature = heater set temperature) eco (target temperature = heater set temperature - 3.5°C), frost_protection (target temperature = 7 to 8°C), off (heater stops heating), and the less commonly used comfort_-1 (target temperature = heater set temperature - 1°C), comfort_-2 (target temperature = heater set temperature - 2°C),.","label":"Pilot wire mode","name":"pilot_wire_mode","property":"pilot_wire_mode","type":"enum","values":["comfort","eco","frost_protection","off","comfort_-1","comfort_-2"]},{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Works only when the pilot wire is deactivated","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":5,"description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured apparent power","label":"Power apparent","name":"power_apparent","property":"power_apparent","type":"numeric","unit":"VA"},{"access":7,"description":"Controls the behavior when the device is powered on. Works only when the pilot wire is deactivated","label":"Power-on behavior","name":"power_on_behavior","property":"power_on_behavior","type":"enum","values":["off","previous","on"]},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"064882","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":true,"vendor":"Legrand"},"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2024-01-27 08:12:54: Configuring '0x0004740000278dfb'
Zigbee2MQTT:info  2024-01-27 08:12:55: Successfully configured '0x0004740000278dfb'
Zigbee2MQTT:warn  2024-01-27 08:13:04: Device '0x0004740000278dfb' left the network
Zigbee2MQTT:info  2024-01-27 08:13:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb"},"type":"device_leave"}'
Zigbee2MQTT:warn  2024-01-27 08:13:04: Device '0x0004740000278dfb' left the network
Zigbee2MQTT:info  2024-01-27 08:13:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb"},"type":"device_leave"}'
Zigbee2MQTT:info  2024-01-27 08:13:07: Device '0x0004740000278dfb' joined
Zigbee2MQTT:info  2024-01-27 08:13:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb"},"type":"device_joined"}'
Zigbee2MQTT:info  2024-01-27 08:13:07: Starting interview of '0x0004740000278dfb'
Zigbee2MQTT:info  2024-01-27 08:13:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:warn  2024-01-27 08:13:24: Device '0x0004740000278dfb' left the network
Zigbee2MQTT:info  2024-01-27 08:13:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb"},"type":"device_leave"}'
Zigbee2MQTT:warn  2024-01-27 08:13:24: Device '0x0004740000278dfb' left the network
Zigbee2MQTT:info  2024-01-27 08:13:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000278dfb","ieee_address":"0x0004740000278dfb"},"type":"device_leave"}'

/opt/zigbee2mqtt/lib/zigbee.ts:89
            await device.resolveDefinition();
                         ^
TypeError: Cannot read properties of undefined (reading 'resolveDefinition')
    at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:89:26)
    at Controller.emit (node:events:518:28)
    at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:527:14)
    at Controller.onDeviceJoined (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:586:22)

apizzolato, 2024-02-07:
I have the same indentical problem with Z2M, instead with the same hardware ZHA works correctly.

Benji67, 2024-02-11:
Hi, i have got the same issue. After the update, i can't pair anymore with my legrand outlet.

sjorge, 2024-02-11:
Seems to be a common problem https://github.com/Koenkk/zigbee2mqtt/issues/20664#issuecomment-1937478070

Benji67, 2024-02-11:
It looks like the same issue and this process (https://github.com/Koenkk/zigbee2mqtt/issues/20664#issuecomment-1893561624) works for me :)

srett, 2024-03-03:
Just ran into the same issue. Z2M is crashing hard, thus after restart the device is missing again and the process starts anew, as far as I can tell.

```
 /opt/zigbee2mqtt/lib/zigbee.ts:89
             await device.resolveDefinition();
                          ^
 TypeError: Cannot read properties of undefined (reading 'resolveDefinition')
     at Controller.<anonymous> (/opt/zigbee2mqtt/lib/zigbee.ts:89:26)
     at Controller.emit (node:events:518:28)
     at Controller.selfAndDeviceEmit (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:529:14)
     at Controller.onDeviceJoined (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:588:22)
zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
```

In [zigbee.ts](https://github.com/Koenkk/zigbee2mqtt/blob/2a53b8e59b457e0805318bdb38d2c74de651724d/lib/zigbee.ts#L89), it seems when the await was added in 56f57625b0d5c it should've been added one line below, after device was checked for validity.

srett, 2024-03-03:
I'm starting to suspect I broke my outlet while experimenting. I tried pairing again and again, deleting the DB and whatnot, and now it looks like this:

```
2024-03-03T20:11:36.358Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,199,236,136,177,76,63,21,56,193,164,49,113,234,254,12,69,202,199,236,136,177,76,63,21,56,193,164,49,113,234]
2024-03-03T20:11:36.359Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,199,236,136,177,76,63,21,56,193,164,49,113,234,254,12,69,202,199,236,136,177,76,63,21,56,193,164,49,113,234]
2024-03-03T20:11:36.359Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [199,236,136,177,76,63,21,56,193,164,49,113] - 234
2024-03-03T20:11:36.359Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":60615,"extaddr":"0xa4c138153cb1f499","parentaddr":28977}
2024-03-03T20:11:36.361Z zigbee-herdsman:controller:log Device '0xa4c138153cb1f499' joined
2024-03-03T20:11:36.361Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,199,236,136,177,76,63,21,56,193,164,49,113,234]
2024-03-03T20:11:36.361Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [199,236,136,177,76,63,21,56,193,164,49,113] - 234
2024-03-03T20:11:36.361Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":60615,"extaddr":"0xa4c138153cb1f499","parentaddr":28977}
2024-03-03T20:11:36.361Z zigbee-herdsman:controller:log Device '0xa4c138153cb1f499' joined
2024-03-03T20:11:36.362Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-03-03T20:11:36.362Z zigbee-herdsman:controller:log Device '0xa4c138153cb1f499' accepted by handler
2024-03-03T20:11:36.362Z zigbee-herdsman:controller:log Delete device '0xa4c138153cb1f499' joined, undeleting
2024-03-03T20:11:36.363Z zigbee-herdsman:controller:database:log Writing database to '/opt/zigbee2mqtt/data/database.db'
2024-03-03T20:11:36.554Z zigbee-herdsman:controller:log Device '0xa4c138153cb1f499' is already in database with different networkAddress, updating networkAddress
2024-03-03T20:11:36.555Z zigbee-herdsman:controller:database:log Writing database to '/opt/zigbee2mqtt/data/database.db'
Zigbee2MQTT:info  2024-03-03T21:11:36+01:00: MQTT publish: topic 'z2m/Steckdose TEST', payload '{"last_seen":"2024-03-03T20:11:36.729Z"}'
2024-03-03T20:11:36.733Z zigbee-herdsman:controller:log Interview '0xa4c138153cb1f499' start
2024-03-03T20:11:36.734Z zigbee-herdsman:controller:device:log Interview - start device '0xa4c138153cb1f499'
2024-03-03T20:11:36.735Z zigbee-herdsman:controller:device:log Interview - skip node descriptor request for '0xa4c138153cb1f499', already got it
2024-03-03T20:11:36.735Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":60615,"nwkaddrofinterest":60615}
2024-03-03T20:11:36.735Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,199,236,199,236,36]
2024-03-03T20:11:36.735Z zigbee-herdsman:controller:log Device '0xa4c138153cb1f499' accepted by handler
Zigbee2MQTT:info  2024-03-03T21:11:36+01:00: MQTT publish: topic 'z2m/Steckdose TEST', payload '{"last_seen":"2024-03-03T20:11:36.735Z"}'
2024-03-03T20:11:36.736Z zigbee-herdsman:controller:log Not interviewing '0xa4c138153cb1f499', completed 'false', in progress 'true'
Zigbee2MQTT:info  2024-03-03T21:11:36+01:00: Device 'Steckdose TEST' joined
Zigbee2MQTT:info  2024-03-03T21:11:36+01:00: MQTT publish: topic 'z2m/bridge/event', payload '{"data":{"friendly_name":"Steckdose TEST","ieee_address":"0xa4c138153cb1f499"},"type":"device_joined"}'
Zigbee2MQTT:info  2024-03-03T21:11:36+01:00: Starting interview of 'Steckdose TEST'
Zigbee2MQTT:info  2024-03-03T21:11:36+01:00: MQTT publish: topic 'z2m/bridge/event', payload '{"data":{"friendly_name":"Steckdose TEST","ieee_address":"0xa4c138153cb1f499","status":"started"},"type":"device_interview"}'
2024-03-03T20:11:36.780Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,199,236,199,236,136,177,76,63,21,56,193,164,142,5,254,1,101,5,0,97]
2024-03-03T20:11:36.780Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,199,236,199,236,136,177,76,63,21,56,193,164,142,5,254,1,101,5,0,97]
2024-03-03T20:11:36.780Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [199,236,199,236,136,177,76,63,21,56,193,164,142] - 5
2024-03-03T20:11:36.780Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":60615,"nwkaddr":60615,"ieeeaddr":"0xa4c138153cb1f499","capabilities":142}
2024-03-03T20:11:36.780Z zigbee-herdsman:controller:log Device announce '0xa4c138153cb1f499'
Zigbee2MQTT:info  2024-03-03T21:11:36+01:00: MQTT publish: topic 'z2m/Steckdose TEST', payload '{"last_seen":"2024-03-03T20:11:36.780Z"}'
Zigbee2MQTT:debug 2024-03-03T21:11:36+01:00: Device 'Steckdose TEST' announced itself
Zigbee2MQTT:info  2024-03-03T21:11:36+01:00: MQTT publish: topic 'z2m/bridge/event', payload '{"data":{"friendly_name":"Steckdose TEST","ieee_address":"0xa4c138153cb1f499"},"type":"device_announce"}'
2024-03-03T20:11:36.798Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
2024-03-03T20:11:36.798Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
2024-03-03T20:11:36.798Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
2024-03-03T20:11:36.798Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...]
Zigbee2MQTT:info  2024-03-03T21:11:57+01:00: MQTT publish: topic 'z2m/bridge/event', payload '{"data":{"friendly_name":"Steckdose TEST","ieee_address":"0xa4c138153cb1f499"},"type":"device_leave"}'
```

The device doesn't even interview successfully anymore, it just joins and leaves in an infinite loop; `Error: Interview failed because can not get active endpoints` doesn't sound too great.

Note that I'm not using the same device as in this issue but a [TS011F_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1), but initially it was the same problem, which is why I ended up here...

closed 2024-09-30 by github-actions[bot]