Koenkk/zigbee2mqtt #21037: Error: 2024-01-27 07:32:36Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()

Burki24, 2024-01-27:
### What happened?

When I set 'Measurement poll interval' on device '_TZ3000_rdtixbnu' ([TS0121_plug](https://www.zigbee2mqtt.io/devices/TS0121_plug.html#tuya-ts0121_plug)), I get this error-log: Error 2024-01-27 07:32:36Failed to apply calibration to 'power': 'power_calibration' is not a number, got string (). But I didn't change calibration for power.

After setting 'poll interval' this error message persists and pop up every minute, until I change the parameter 'calibration for power' to '0'.

### What did you expect to happen?

Change the poll interval only, not changing anything other without getting that error message.

### How to reproduce it (minimal and precise)

On every plug-device with powermeassurement, if there were manually changes in settings(specific). Tested on tuya-devices. 

### Zigbee2MQTT version

1.35.1-dev commit: a879382c

### Adapter firmware version

EZSP V8 - 7.0.2.0 build 406

### Adapter

sonoff-zigbee-3-0-usb-dongle-plus-e

### Setup

plain on Pi

### Debug log

Error 2024-01-27 09:02:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:03:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:04:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:06:15Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:06:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:07:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:08:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:09:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:10:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:11:42Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:12:43Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:13:43Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:14:43Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Error 2024-01-27 09:15:43Failed to apply calibration to 'power': 'power_calibration' is not a number, got string ()
Info 2024-01-27 09:15:43MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"b8w1a-6"}'
Debug 2024-01-27 09:15:44Received Zigbee message from 'Wohnzimmer/Sicherheit/Wohnbereich/Radar', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,2,223],"type":"Buffer"},"datatype":2,"dp":9}],"seq":20992}' from endpoint 1 with groupID 0
Info 2024-01-27 09:15:44MQTT publish: topic 'zigbee2mqtt/Wohnzimmer/Sicherheit/Wohnbereich/Radar', payload '{"illuminance_lux":15,"last_seen":1706343343992,"linkquality":212,"target_distance":7.35}'
Debug 2024-01-27 09:15:44Received Zigbee message from 'Buero/Sicherheit/Radar', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,186],"type":"Buffer"},"datatype":2,"dp":9}],"seq":33024}' from endpoint 1 with groupID 0
Info 2024-01-27 09:15:44MQTT publish: topic 'zigbee2mqtt/Buero/Sicherheit/Radar', payload '{"cli":" ","detection_delay":0,"fading_time":20,"illuminance_lux":24,"last_seen":1706343344460,"linkquality":164,"maximum_range":3,"minimum_range":0.3,"presence":true,"radar_sensitivity":8,"self_test":"check_success","target_distance":1.86}'
Debug 2024-01-27 09:15:45Received Zigbee message from 'Wohnzimmer/Sicherheit/Wohnbereich/Radar', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,2,211],"type":"Buffer"},"datatype":2,"dp":9}],"seq":21248}' from endpoint 1 with groupID 0

Koenkk, 2024-01-28:
What is the `power_calibration` value in your `configuration.yaml`?

Burki24, 2024-01-30:
Didn't changed it. So it is not used. the calibration I never changed. It isn't inside configuration.yaml and also not in devices.yaml.

Burki24, 2024-02-01:
@Koenkk 
Same Issue with https://www.zigbee2mqtt.io/devices/SP_240.html.

Zigbee2MQTT V. 1.35.1-dev commit: [0b59a51c](https://github.com/Koenkk/zigbee2mqtt/commit/0b59a51c)
Coordinator-Typ EZSP v8
Coordinator-Version 7.0.2.0 build 406
Coordinator IEEE Adresse 0xe0798dfffebbe7d6
Frontend-Version 0.6.152
Zigbee-herdsman-converters version 18.25.1
Zigbee-herdsman version 0.33.5

Debug:

```
Info 2024-02-01 16:29:40MQTT publish: topic 'zigbee2mqtt/0xa4c138a71702a8fe', payload '{"action":"on","current":0,"energy":0,"linkquality":248,"power":0,"power_on_behavior":"previous","state":"ON","voltage":227}'
Info 2024-02-01 16:29:41MQTT publish: topic 'zigbee2mqtt/0xa4c138a71702a8fe', payload '{"current":0,"energy":0,"linkquality":255,"power":0,"power_on_behavior":"previous","state":"ON","voltage":227}'
Error 2024-02-01 16:29:41Failed to apply calibration to 'power': 'power_precision' is not a number, got string ()
Info 2024-02-01 16:29:41MQTT publish: topic 'zigbee2mqtt/0xa4c138a71702a8fe', payload '{"current":0,"energy":0,"linkquality":248,"power":0,"power_on_behavior":"previous","state":"ON","voltage":227}'
Error 2024-02-01 16:29:42Failed to apply calibration to 'voltage': 'voltage_precision' is not a number, got string ()
Info 2024-02-01 16:29:42MQTT publish: topic 'zigbee2mqtt/0xa4c138a71702a8fe', payload '{"current":0,"energy":0,"linkquality":248,"power":0,"power_on_behavior":"previous","state":"ON","voltage":227}'
Error 2024-02-01 16:29:43Failed to apply calibration to 'current': 'current_precision' is not a number, got string ()
Info 2024-02-01 16:29:43MQTT publish: topic 'zigbee2mqtt/0xa4c138a71702a8fe', payload '{"current":0,"energy":0,"linkquality":252,"power":0,"power_on_behavior":"previous","state":"ON","voltage":227}'
Error 2024-02-01 16:29:44Failed to apply calibration to 'energy': 'energy_calibration' is not a number, got string ()
Info 2024-02-01 16:29:44MQTT publish: topic 'zigbee2mqtt/0xa4c138a71702a8fe', payload '{"current":0,"energy":0,"linkquality":252,"power":0,"power_on_behavior":"previous","state":"ON","voltage":227}'
```
[image]

configuration.yaml:
```

homeassistant: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost:1024
serial:
  port: /dev/ttyACM0
advanced:
  network_key:
    - 102
    - 17
    - 175
    - 4
    - 116
    - 126
    - 187
    - 75
    - 13
    - 76
    - 110
    - 54
    - 65
    - 56
    - 49
    - 58
  pan_id: 55083
  ext_pan_id:
    - 236
    - 84
    - 200
    - 65
    - 99
    - 19
    - 48
    - 171
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
devices:
  '0xa4c13855e2d4363a':
    friendly_name: '0xa4c13855e2d4363a'
  '0xa4c138e3ebf058b7':
    friendly_name: '0xa4c138e3ebf058b7'
  '0x38398ffffeecaf95':
    friendly_name: '0x38398ffffeecaf95'
  '0xa4c138dafc377c8f':
    friendly_name: '0xa4c138dafc377c8f'
    color_sync: true
  '0xa4c138a71702a8fe':
    friendly_name: '0xa4c138a71702a8fe'
    energy_calibration: ''
    power_precision: ''
    voltage_precision: ''
    current_precision: ''
    energy_precision: ''
    state_action: true
```

Koenkk, 2024-02-01:
```
    energy_calibration: ''
    power_precision: ''
    voltage_precision: ''
    current_precision: ''
    energy_precision: ''
```

This is what causes the issue, removing it should fix the problem. I'm not sure how it ended up there in the first place.

Burki24, 2024-02-01:
> ```
>     energy_calibration: ''
>     power_precision: ''
>     voltage_precision: ''
>     current_precision: ''
>     energy_precision: ''
> ```
> 
> This is what causes the issue, removing it should fix the problem. I'm not sure how it ended up there in the first place.

But it wasn't added by myself. So I think there is an issue in Zigbee2MQTT. When I try to change a parameter here:
[image]

There are no values shown in the frontend and in configuration will be added the seen lines above. So for me, I think, there are problems within the converter for this device.

The log, if I try to change something inside 'Settings (specific)':

```
Info 2024-02-01 21:08:08Changed config for device 0xa4c138a71702a8fe
Info 2024-02-01 21:08:08MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"legacy":false},"id":"0xa4c138a71702a8fe","restart_required":false,"to":{"legacy":false,"power_precision":""}},"status":"ok","transaction":"mw2iu-1"}'
Info 2024-02-01 21:08:12Changed config for device 0xa4c138a71702a8fe
Info 2024-02-01 21:08:12MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"legacy":false,"power_precision":""},"id":"0xa4c138a71702a8fe","restart_required":false,"to":{"legacy":false,"power_precision":""}},"status":"ok","transaction":"mw2iu-2"}'
Info 2024-02-01 21:09:09MQTT publish: topic 'zigbee2mqtt/0xa4c138dafc377c8f', payload '{"brightness":254,"color":{"hue":240,"saturation":100},"color_mode":"hs","linkquality":255,"state":"ON"}'
Info 2024-02-01 21:10:41Changed config for device 0xa4c138a71702a8fe
Info 2024-02-01 21:10:41MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"legacy":false,"power_precision":""},"id":"0xa4c138a71702a8fe","restart_required":false,"to":{"legacy":false,"power_precision":""}},"status":"ok","transaction":"mw2iu-3"}'
Info 2024-02-01 21:12:24MQTT publish: topic 'zigbee2mqtt/0xa4c138dafc377c8f', payload '{"brightness":254,"color":{"hue":240,"saturation":100},"color_mode":"hs","linkquality":255,"state":"ON"}'
```

Burki24, 2024-02-01:
> So these lines appear after changing something in the frontend?

I will try this with other devices, to see if it's a device spcific problem or a general problem.

mcarbonneaux, 2024-02-23:
i've the same problem with my [tuya-ts011f_plug_1](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1)  i've set current_precision with value of 1 and energy_precision to 2, and they automaticly setted power_precision to ''.

Mar-85, 2024-03-08:
Recently started getting this issue with all my Tuya smart plugs. And theyve all stopped reporting power consumption via z2m when they used to work flawlessly for months.

jimmydwn, 2024-04-11:
Same issue with my blitzwolf BW-SHP15 smartplug. Got the error message when changing the polling interval. Interval does not seem to change either. Stays at the standard 60 seconds. So I think non of these settings will be working as long as this bug exists.

Will there be a fix?

Koumi460, 2024-05-04:
I have been getting this error with these sockets:
https://www.zigbee2mqtt.io/devices/AU-A1ZBDSS.html#aurora%2520lighting-au-a1zbdss
But also from what I can tell with this switch:
https://www.zigbee2mqtt.io/devices/SSM-U01.html#aqara-ssm-u01
In the logs it is not clear which device is throwing the error, but after reviewing the config file these all had power_precision = "" and when I set them to 0s or 1s the issue went away. 
But more importantly for me, after an update to Z2M version 1.37 all three of the Aurora sockets stopped reporting power consumption. Interesting is that one reports constant 0W power, one reports Null and one says N/A. 
I tried deleting the device and re-adding and it didn't do anything, still showing as Null.
Any help or pointers would be appreciated!

Mar-85, 2024-05-13:
> I have been getting this error with these sockets: https://www.zigbee2mqtt.io/devices/AU-A1ZBDSS.html#aurora%2520lighting-au-a1zbdss But also from what I can tell with this switch: [...]

This definitely helped me with z2m....but requires a z2m reset every now and then to get HA to go grab the values from the plug.

bjornreppen, 2024-07-24:
Created a new installation yesterday and started getting lots of warnings popping up on screen and found multiple instances of *_precision: '' having been automatically added to configuration.yaml for seemingly no good reason.

GKO1, 2024-11-06:
Does somebody have an solution for this topic? I’m also struggeling. I use The NOUS A1 power sockets.

mcarbonneaux, 2024-11-06:
workarond remove void line in configuration file...

closed 2025-01-14 by github-actions[bot]