Koenkk/zigbee2mqtt #21065: Zigbee model: TS011F Zigbee manufacturer: _TZ3000_w0qqde0g Description: ZigBee socket Support Status: Supported Manufacturer: Tuya Model: TS011F not reporting anything except linkquality and on/off

stockno, 2024-01-27:
### Link

https://www.aliexpress.com/item/1005004857105081.html?spm=a2g0o.order_list.order_list_main.16.337c1802IlbDap

### Database entry

{"id":14,"type":"Router","ieeeAddr":"0xa4c13848abf69a42","nwkAddr":39997,"manufId":4417,"manufName":"_TZ3000_w0qqde0g","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_w0qqde0g","powerSource":1,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002281111d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1706348304755,"defaultSendRequestWhen":"immediate"}

### Comments

expoes - it's just on/off and signal strength, I don't know what to do, please help
Zigbee2MQTT Bridge Coordinator version 20210319
Zigbee2MQTT 1.35.1
Home Assistant:
Core 2024.1.5
Supervisor 2023.12.1
Operating System 11.4
Frontend 20240104.0
[image]
[image]
[image]
[image]

### External defintion

```shell
const {onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3000_w0qqde0g',
    description: 'Automatically generated definition',
    extend: [onOff({powerOnBehavior: false}), electricityMeter()],
};

module.exports = definition;
```

stockno, 2024-02-02:
I have already managed to get it working

MacDada, 2024-02-02:
> I have already managed to get it working

How?

stockno, 2024-02-03:
/homeassistant/zigbee2mqtt/configuration.yaml I created

external_converters:
   - socket.js

/homeassistant/zigbee2mqtt/socket.js

I copied this code - I just changed the model of my socket - line 12. Then the electrical measurement it started showing all entities as well:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
{
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_w0qqde0g'}],
        model: 'TS011F_plug',
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1, acVoltageDivisor: 1, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
                acPowerDivisor: 1,
            });
        },
        // This device doesn't support reporting correctly.
        // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
        onEvent: (type, data, device, options) => {
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(globalStore.getValue(device, 'interval'));
                globalStore.clearValue(device, 'interval');
            } else if (!globalStore.hasValue(device, 'interval')) {
                const seconds = options && options.measurement_poll_interval ? options.measurement_poll_interval : 60;
                if (seconds === -1) return;
                const interval = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {/* Do nothing*/}
                }, seconds*1000);
                globalStore.putValue(device, 'interval', interval);
            }
        },
    }
];

closed 2024-09-01 by github-actions[bot]