Koenkk/zigbee2mqtt #21066: Centralite 3400 Keypad not well supported

bonjour81, 2024-01-27:
### What happened?

I have successfully paired a Centralite 3400 keypad which is recognized as 3400-D by Z2M.
It seems to work fine when it's about messages from keypad to z2m; but the other direction, z2m to keypad has no visible effect on keypad (it shall change it's mode for example)

### What did you expect to happen?

Keypad shall react to some mode change by lighting some leds in different colors

### How to reproduce it (minimal and precise)

send message to keypad as described here https://www.zigbee2mqtt.io/devices/3400-D.html#centralite-3400-d

I attach debug log when I send a "disarm" command to the keypad

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20230507

### Adapter

CC2652P

### Setup

Docker on lenovo tiny

### Debug log

2024-01-27T11:12:00.388551697Z Zigbee2MQTT:debug 2024-01-27 12:12:00: Received MQTT message on 'zigbee2mqtt/keypad/set' with data '{"arm_mode":{"mode":"disarm"}}'

2024-01-27T11:12:00.397420940Z Zigbee2MQTT:debug 2024-01-27 12:12:00: Publishing 'set' 'arm_mode' to 'keypad'

2024-01-27T11:12:00.397475491Z   zigbee-herdsman:controller:endpoint CommandResponse 0x000d6f000c45272b/1 ssIasAce.panelStatusChanged({"panelstatus":0,"secondsremain":0,"audiblenotif":0,"alarmstatus":0}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +25s

2024-01-27T11:12:00.397495158Z   zigbee-herdsman:controller:endpoint Request Queue (0x000d6f000c45272b/1): send panelStatusChanged request immediately (sendWhen=active) +1ms

2024-01-27T11:12:00.397506830Z   zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x000d6f000c45272b:792/1 (0,0,1) +25s

2024-01-27T11:12:00.397517853Z   zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":792,"destendpoint":1,"srcendpoint":1,"clusterid":1281,"transid":165,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[25,140,4,0,0,0,0]}} +25s

2024-01-27T11:12:00.397532679Z   zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,24,3,1,1,1,5,165,0,30,7,25,140,4,0,0,0,0,6] +25s

2024-01-27T11:12:00.404753256Z   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +7s

2024-01-27T11:12:00.404797497Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms

2024-01-27T11:12:00.404812356Z   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms

2024-01-27T11:12:00.404941141Z   zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +25s

2024-01-27T11:12:00.404981129Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms

2024-01-27T11:12:00.413510507Z   zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,165,99] +8ms

2024-01-27T11:12:00.413551365Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,165,99] +1ms

2024-01-27T11:12:00.413566236Z   zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,165] - 99 +0ms

2024-01-27T11:12:00.413578534Z   zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":165} +7s

2024-01-27T11:12:00.413828988Z   zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms

MrNiceguy2121, 2024-02-03:
I also recently bought a Centralite 3400 for the purpose of making an alarm and also came to the conclusion that it didn't fully work with Z2M. 
After some searching online I found a post on github with someone using the code below to change the lights of his 3400G (unfortunately I can't find the post).

I added this to my config as an external converter and together with an automation in home assistant it finally changes the lights of the device. 
Only thing left for me is to try and get the sounds to work so that it also gives audible feedback, but I am not a coder so that could take forever. 
 

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const constants = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const arm_mode = {
    key: ['arm_mode'],
    convertSet: async (entity, key, value, meta) => {
        meta.logger.info(`[AlarmKeypad] Converter called for ${JSON.stringify(key)} and ${JSON.stringify(value)}`);
        const isNotification = value.hasOwnProperty('transaction');
        const modeSrc = isNotification ? constants.armNotification : constants.armMode;
        const mode = utils.getKey(modeSrc, value.mode, undefined, Number);
        if (mode === undefined) {
            throw new Error(`Unsupported mode: '${value.mode}', should be one of: ${Object.values(modeSrc)}`);
        }

        if (isNotification) {
            entity.commandResponse('ssIasAce', 'armRsp', {armnotification: mode}, {}, value.transaction);

            // Do not update PanelStatus after confirming transaction.
            // Instead the server should send an arm_mode command with the necessary state.
            // e.g. exit_delay as a result of arm_all_zones
            return;
        }

        globalStore.putValue(entity, 'panelStatus', mode);
        const payload = {panelstatus: mode, secondsremain: 0, audiblenotif: 0, alarmstatus: 0};
        
        meta.logger.info(`[AlarmKeypad] Sending message ${JSON.stringify(payload)}`);
        entity.commandResponse('ssIasAce', 'panelStatusChanged', payload);
    },
};

const definition = {
    zigbeeModel: ['3400-G', '3400'],
    model: '3400-G',
    vendor: 'Centralite',
    description: '3-Series security keypad',
    meta: {battery: {voltageToPercentage: '3V_2100'}},
    fromZigbee: [fz.command_arm_with_transaction, fz.temperature, fz.battery, fz.ias_ace_occupancy_with_timeout],
    toZigbee: [arm_mode],
    exposes: [e.battery(), e.temperature(), e.occupancy(),
        exposes.numeric('action_code', ea.STATE).withDescription('Pin code introduced.'),
        exposes.numeric('action_transaction', ea.STATE).withDescription('Last action transaction number.'),
        exposes.text('action_zone', ea.STATE).withDescription('Alarm zone. Default value 0'),
        e.action(['disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'exit_delay', 'emergency'])],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const clusters = ['msTemperatureMeasurement', 'genPowerCfg', 'ssIasZone', 'ssIasAce'];
        await reporting.bind(endpoint, coordinatorEndpoint, clusters);
        await reporting.temperature(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
    onEvent: async (type, data, device) => {
        if (type === 'message' && data.type === 'commandGetPanelStatus' && data.cluster === 'ssIasAce' && globalStore.hasValue(device.getEndpoint(1), 'panelStatus')) {
            const payload = {
                panelstatus: globalStore.getValue(device.getEndpoint(1), 'panelStatus'),
                secondsremain: 0x00, audiblenotif: 0xC8, alarmstatus: 0x00,
            };
            await device.getEndpoint(1).commandResponse(
                'ssIasAce', 'getPanelStatusRsp', payload, {}, data.meta.zclTransactionSequenceNumber,
            );
        }
    },

};

module.exports = definition;
```

bonjour81, 2024-02-05:
Hi !  thank for your feedback !
I tried the external converter code.

It seems to provide more reactions of the keypad :)

So far: if I send 'arm_all_zones' mode (server to keypad), it will turn the top led and the top middle button (the guy running) in red.
'arm_night_zones' seems to bring no reaction (I would have expected to turn red the moon button).

I will play around with the keypad/server discussion next days (set a mode with pin code / server answer) to check what I can do.

I think the code come from here:  https://github.com/Koenkk/zigbee2mqtt/issues/11828
it does match, except one line in the middle:
`   zigbeeModel: ['3400-G'],  ` vs    `zigbeeModel: ['3400-G', '3400'],`

lukashartman, 2024-02-17:
I've also been trying to connect my Centralite 3400 to Zigbee2MQTT after transitions off ZHA, and I'm having the same issue. However, I'm not able to get the above external converter to work.

closed 2024-09-14 by github-actions[bot]