Koenkk/zigbee2mqtt #21071: Device SIN-4-FP-21_EQU does not respond to command

easyservices, 2024-01-27:
### What happened?

I have 1 SIN-4-FP-21_EQU device on my network. The pairing went well and the device listed in the Zigbee UI does correspond to the supported model (SIN-4-FP-21_EQU) and LQI is 72.

When I send a command for setting up the heater to the "comfort" mode, I have an error as a response.

### What did you expect to happen?

The device should react to order and the heater should be set according to the pilot wire order.

### How to reproduce it (minimal and precise)

Add a SIN-4-FP-21_EQU device to zigbee2mqtt,
send to channel: zigbee2mqtt/0x003c84fffed669c2/set
with payload: {"pilot_wire_mode": "comfort"}

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Setup

docker container on x86_64 (ubuntu OS)

### Debug log

debug 2024-01-27 18:26:50Error: Command 0x003c84fffed669c2/242 manuSpecificNodOnPilotWire.setMode({"mode":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4747,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54703 - 242 - 240 - 64512 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7)

debug 2024-01-27 18:26:54Received Zigbee message from '0x003c84fffed669c2', type 'attributeReport', cluster 'genIdentify', data '{"identifyTime":0}' from endpoint 1 with groupID 0

debug 2024-01-27 18:26:54No converter available for 'SIN-4-FP-21_EQU' with cluster 'genIdentify' and type 'attributeReport' and data '{"identifyTime":0}'

debug 2024-01-27 18:26:55Received Zigbee message from '0x003c84fffed669c2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0

info 2024-01-27 18:26:55MQTT publish: topic 'zigbee2mqtt/0x003c84fffed669c2', payload '{"energy":0,"linkquality":66,"pilot_wire_mode":"frost_protection","power":0,"state":"OFF"}'

debug 2024-01-27 18:26:55Received Zigbee message from '0x003c84fffed669c2', type 'attributeReport', cluster 'manuSpecificNodOnPilotWire', data '{"mode":3}' from endpoint 1 with groupID 0

info 2024-01-27 18:26:55MQTT publish: topic 'zigbee2mqtt/0x003c84fffed669c2', payload '{"energy":0,"linkquality":63,"pilot_wire_mode":"frost_protection","power":0,"state":"OFF"}'

debug 2024-01-27 18:26:59Received Zigbee message from '0x003c84fffed669c2', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,0],"instantaneousDemand":0}' from endpoint 1 with groupID 0
info 2024-01-27 18:26:59MQTT publish: topic 'zigbee2mqtt/0x003c84fffed669c2', payload '{"energy":0,"linkquality":66,"pilot_wire_mode":"frost_protection","power":0,"state":"OFF"}'

error 2024-01-27 18:27:13Publish 'set' 'pilot_wire_mode' to '0x003c84fffed669c2' failed: 'Error: Command 0x003c84fffed669c2/242 manuSpecificNodOnPilotWire.setMode({"mode":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4747,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54703 - 242 - 241 - 64512 - 11 after 10000ms)'

debug 2024-01-27 18:27:13Error: Command 0x003c84fffed669c2/242 manuSpecificNodOnPilotWire.setMode({"mode":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4747,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54703 - 242 - 241 - 64512 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7)

error 2024-01-27 18:27:36Publish 'set' 'pilot_wire_mode' to '0x003c84fffed669c2' failed: 'Error: Command 0x003c84fffed669c2/242 manuSpecificNodOnPilotWire.setMode({"mode":2}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4747,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54703 - 242 - 242 - 64512 - 11 after 10000ms)'

easyservices, 2024-01-29:
It is a regression from the 1.35.0 
I did downgrade to the 1.35.0 and it works fine with this version.

grrosminet, 2024-02-02:
résolu dans 1.35.2

qukhan, 2024-02-10:
I have the exact same problem on version 1.35.3.

I can have the devices join but they will randomly start erroring out within a day or two. The only workaround I have found is to reset the device and have it re-join the network.

grrosminet, 2024-02-10:
1.35.2 works perfectly

qukhan, 2024-02-10:
I just downgraded to 1.35.2 and have the same issue.

easyservices, 2024-02-10:
Good to know. I stay on the 1.35.0 for now as it works fine for me.

Koenkk, 2024-02-10:
can you provide the herdsman debug log with both 1.35.0 (working) and 1.35.3 (not working) when sending this command?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

drpixel972, 2024-02-11:
Hello, 

Here are my two bucks.
My HA system is working with the SONOFF ZigBee 3.0 USB Dongle Plus. I had the issue.
While trying to debug with another couple (computer + conbee adapter ) it works perfectly.
I though that the adapter was the cause. I took my production adapter to my test computer and the issue arise just after pairing the device.
However I restart Z2MQTT to gather the logs ... no issue :-/.
I did the same thing on my production system (Z2MQTT reboot after pairing) ... that's also "fix" the issue :-/.
That said : 
- I had the device since yesterday, so I have no clue on real production system and will the device still respond in a few days ... 
- I have only one device, no clue whether or not I'll need to perform a full restart after each pairing.

Best Regards,

courville, 2024-02-21:
I have the same problem running [1.35.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.3) commit: [fe0742a](https://github.com/Koenkk/zigbee2mqtt/commit/fe0742a)
```
Error 2024-02-21 19:57:49Publish 'set' 'pilot_wire_mode' to 'fils pilote' failed: 'Error: Command 0xb43a31fffed42f45/242 manuSpecificNodOnPilotWire.setMode({"mode":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4747,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3702 - 242 - 209 - 64512 - 11 after 10000ms)'
```

froggy974, 2024-02-23:
same problem for me with 1.35.3-1 , SONOFF ZigBee 3.0 USB Dongle Plus and SIN-4-FP-21_EQU

```Publish 'set' 'state' to '0x90395efffe409949' failed: 'Error: Command 0x90395efffe409949/242 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 41565 - 242 - 24 - 6 - 11 after 10000ms)```

Koenkk, 2024-02-23:
> can you provide the herdsman debug log with both 1.35.0 (working) and 1.35.3 (not working) when sending this command?
> 
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

@froggy974 could you maybe provide this?

froggy974, 2024-02-23:
i upgraded to 1.35.3 and tried to reproduce the issue, but it's working also on last version now :-)

anyway i have attached the 2 log files
[1.35.3 working also 45df7312_zigbee2mqtt_2024-02-23T19-28-09.323Z.log](https://github.com/Koenkk/zigbee2mqtt/files/14389373/1.35.3.working.also.45df7312_zigbee2mqtt_2024-02-23T19-28-09.323Z.log)
[1.35.0 working 45df7312_zigbee2mqtt_2024-02-23T19-24-47.574Z.log](https://github.com/Koenkk/zigbee2mqtt/files/14389374/1.35.0.working.45df7312_zigbee2mqtt_2024-02-23T19-24-47.574Z.log)

courville, 2024-05-03:
> I have the same problem running [1.35.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.3) commit: [fe0742a](https://github.com/Koenkk/zigbee2mqtt/commit/fe0742a) with SONOFF ZigBee 3.0 USB Dongle Plus and on my side it is a SIN-4-FP-21 (without the `_EQU`):
> 
> ```
> Error 2024-02-21 19:57:49Publish 'set' 'pilot_wire_mode' to 'fils pilote' failed: 'Error: Command 0xb43a31fffed42f45/242 manuSpecificNodOnPilotWire.setMode({"mode":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4747,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3702 - 242 - 209 - 64512 - 11 after 10000ms)'
> ```

FYI works for me now on 1.37.0 (did not checked in between when it was really fixed).

easyservices, 2024-05-03:
It works for me also on 1.37.0 (the last working version was the 1.35.3).

closed 2024-05-03 by easyservices