Koenkk/zigbee2mqtt #21072: Endpoint name is not matched properly

grafalex82, 2024-01-27:
### What happened?

I am developing a smart switch that has 3 roughly identical endpoints, named `left`, `right`, and `both` (endpoint match to left or right button of the switch, and a separate endpoint to handle both button press). Here is how endpoints defined in the external converter:

`
const device = {
...
    endpoint: (device) => {
        return {
            "basic": 1,
            "left": 2,
            "right": 3,
            "both": 4
        };
    },
`

Left and Right endpoints work fine. Unfortunately any attempt to work with `both` endpoint controls fail with this message:
`
Received MQTT message on 'zigbee2mqtt/my_test_switch/get' with data '{"state_both":""}'
No converter available for 'state_both' ("")
`

I debugged the issue up to publish.ts, onMQTTMessage() function, up to the place where propertyEndpointRegex regexp is matched. When it has 'state_left' as input the regexp works fine, but 'state_both' is not matched. The problem is that the regexp contains number of predefined endpoint names:

`^(.*?)_(left|right|center|bottom_left|bottom_right|default|top_left|top_right|white|rgb|cct|system|top|bottom|center_left|center_right|ep1|ep2|row_1|row_2|row_3|row_4|relay|usb|l1|l2|l3|l4|l5|l6|l7|l8|l9|l10|l11|l12|l13|l14|l15|l16|l17|l18|l19|l20|l21|l22|l23|l24|th1|th2|th3|th4|th5|th6|th7|th8|th9|th10|button_1|button_2|button_3|button_4|button_5|button_6|button_7|button_8|button_9|button_10|button_11|button_12|button_13|button_14|button_15|button_16|button_17|button_18|button_19|button_20|button_light|button_fan_high|button_fan_med|button_fan_low|heat|cool|water|meter|wifi|no_occupancy_since)$`

So this code works just because `left` and `right` are concidentally on the list.

Unless there is a specific reason to have this regexp implemented as it is now, would it be reasonable to build it in runtime based on actual endpoint names listed in the device converter?

### What did you expect to happen?

Attribute names such as `state_both` parsed normally, based on the fact that `both` is a name of the endpoint.

### How to reproduce it (minimal and precise)

Name an endpoint as 'both', try to operate with its attributes

### Zigbee2MQTT version

1.35

### Adapter firmware version

20200211

### Adapter

CC2652

### Setup

Z2M in docker container, running on x64 Linux computer

Koenkk, 2024-01-28:
Does both control both left and right? This is not quite common for Zigbee devices.

grafalex82, 2024-01-28:
no, both means pressing both buttons of the 2-gang switch. 
The reasons I moved it to a separate endpoint are
1) it has number of settings via extended On/Off Settings Cluster
2) it allows binding to other devices 

Thus. long press of 2 buttons generate a Level Ctrl Move command that can be bound to a shades, for example. And this works separately from left and right buttons that can be configured to other actions.

Koenkk, 2024-01-29:
> But my issue report is more general - endpoint names shall not be predefined on z2m side, but rather allow defining user ones.

I agree, is this something you can fix through a PR?

sjorge, 2024-01-30:
I wonder if this is something we can take from the `endpoint` data on the device definition, all endpoints on a device should be there to map them to the endpointID, well aside from `default` which should probably still be hard coded in z2m.

Koenkk, 2024-01-30:
@sjorge yes I was also thinking that.

grafalex82, 2024-02-01:
> I agree, is this something you can fix through a PR?

Yes, but I need some consulting first.

I studied the existing code, and started with parseTopic() function (publish.ts). 

Let me summarize capabilities of this function. It allows get/set topics of the following forms:
- `<base_topic>/device_name/set`  (endpoint and attribute is set in the payload)
- `<base_topic>/device_name/set/attribute` (default endpoint used)
- `<base_topic>/device_name/endpoint/set` (attribute is set in the payload)
- `<base_topic>/device_name/endpoint/set/attribute` (payload is the value. This option is not described in the documentation, btw)

Also
- The device_name supports simple symbolid names, as well as IEEE address (e.g. 0x1234567890abcdef)
- The device_name supports slash in the name (e.g. 'in/basement/wall_switch_double')
- Endpoint can be either a number, or a symbolic namey

Problem: some of the syntax construct listed above in fact are ambigous:
- the `device_name/endpoint` part is ambigous. It is not clear in advance whether this is a device name + endpoint name, or a device name with slashes. 
- The device name (or device name part) or endpoint name can be `set` or `get`. In this case parsing of such names gets even more complicated

My nearest plans would be to split parseTopic() into 2 functions:
- One parses device name first. This is needed to find the device, and get its endpoints list
- Second parses the rest of the topic - endpoint and attribute name

Am I going in the right direction?

Koenkk, 2024-02-01:
> the device_name/endpoint part is ambigous. It is not clear in advance whether this is a device name + endpoint name, or a device name with slashes.

You are not allowed to use `device/endpoint`, the check is done [here](https://github.com/Koenkk/zigbee2mqtt/blob/06b3395cff4fa938aa29a55112605becfafc83b1/lib/util/utils.ts#L264). So this solves the ambiguity part.

grafalex82, 2024-02-02:
Thank you for the hint, understood.

I have some progress on parsing the topic name without having to know device's endpoint list. But now I somewhat stuck on the place where it matches `attribute_endpoint`-like payload, and here I need an access to the endpoint to name mapping structure from the converter. 

Unfortunately [device definition object here](https://github.com/Koenkk/zigbee2mqtt/blob/06b3395cff4fa938aa29a55112605becfafc83b1/lib/extension/publish.ts#L131) has everything (name, vendor, to/from converters, etc), except for endpoint mapping table [like this](https://github.com/grafalex82/hellozigbee/blob/main/zigbee2mqtt/myswitch.js#L310). Could you please point me to the right way to do this? How fields from the converter appears in the device definition, and why endpoint() is not there?

Koenkk, 2024-02-03:
Maybe I don't understand it correctly, but `endpoint()` is called [one line below](https://github.com/Koenkk/zigbee2mqtt/blob/06b3395cff4fa938aa29a55112605becfafc83b1/lib/extension/publish.ts#L132)

grafalex82, 2024-02-03:
Nevermind, I was confused with simultaneous usage of ZH Device, Z2M Device, and ZHC Device Definition.

I am going to write a function to get a list of endpoint names. Would you like me to add it to the Device class? Or leave it local to Publish extension?

grafalex82, 2024-02-03:
I think I almost done with the fix, going to publish a PR tomorrow. 

The only thing is not working is [this test](https://github.com/Koenkk/zigbee2mqtt/blob/master/test/group.test.js#L398).
```
    it('Should publish state of group when specific state of specific endpoint is changed', async () => {
        const group = zigbeeHerdsman.groups.gledopto_group;
        await resetExtension();

        MQTT.publish.mockClear();
        await MQTT.events.message('zigbee2mqtt/GLEDOPTO_2ID/set', stringify({state_cct: 'ON'}));
        await flushPromises();
        expect(MQTT.publish).toHaveBeenCalledTimes(2);
        expect(MQTT.publish).toHaveBeenCalledWith("zigbee2mqtt/GLEDOPTO_2ID", stringify({"state_cct":"ON"}), {"retain": false, qos: 0}, expect.any(Function));
        expect(MQTT.publish).toHaveBeenCalledWith("zigbee2mqtt/gledopto_group", stringify({"state":"ON"}), {"retain": false, qos: 0}, expect.any(Function));
        expect(group.command).toHaveBeenCalledTimes(0);
    });
```

Although this is currently working, in my opinion this functionality is incorrect. Imagine you have 2 devices on the group
- Device `A` with endpoint `a` on the group
- Device 'B' with endpoint 'b' on the group. At the same time the device also has endpoint 'a' which is NOT a part of the group.

Why a message to `zigbee2mqtt/group_name/set`  with payload `state_a:ON` should toggle the whole group? This payload is definitely valid for a specific device, but the group should not allow anything with the endpoint definition.

What do you think?

Koenkk, 2024-02-04:
> Why a message to zigbee2mqtt/group_name/set with payload state_a:ON should toggle the whole group?

The message is send to the device, not the group: `zigbee2mqtt/GLEDOPTO_2ID/set', stringify({state_cct: 'ON'}`. The reason for this is how z2m handles group [state updates](https://www.zigbee2mqtt.io/guide/usage/groups.html#state-changes)

grafalex82, 2024-02-04:
I found the reason of the failing test. 

Please check out the [PR](https://github.com/Koenkk/zigbee2mqtt/pull/21214).

grafalex82, 2024-02-11:
I think the issue is solved with these 3 PRs:
- https://github.com/Koenkk/zigbee2mqtt/pull/21214
- https://github.com/Koenkk/zigbee2mqtt/pull/21264
- https://github.com/Koenkk/zigbee2mqtt/pull/21325

closed 2024-02-11 by grafalex82