Koenkk/zigbee2mqtt #21074: Cannot read properties of null for update. All devices are requesting OTA in loop whole day

Hadatko, 2024-01-27:
### What happened?

All zigbee devices (OSRAM, Philips) are requesting OTA whole day. Manuall check is failing.
[image]

### What did you expect to happen?

Ask once and then do not spawn so many requests

### How to reproduce it (minimal and precise)

I gues checking for OTA which failed could cause that OTA is asked in loop then

### Zigbee2MQTT version

1.35.1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

skyconnect

### Setup

addon in HA

### Debug log

```
Debug 2024-01-27 20:31:55Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:31:55Device 'chodba_3_svetlo_scena' requested OTA
Debug 2024-01-27 20:31:58Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:31:58Device 'chodba_3_svetlo_scena' requested OTA
Debug 2024-01-27 20:32:01Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:32:01Device 'chodba_3_svetlo_scena' requested OTA
Debug 2024-01-27 20:32:04Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:32:04Device 'chodba_3_svetlo_scena' requested OTA
Debug 2024-01-27 20:32:07Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:32:07Device 'chodba_3_svetlo_scena' requested OTA
Debug 2024-01-27 20:32:11Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"chodba_3_svetlo_scena","transaction":"muh28-1"}'
Info 2024-01-27 20:32:11Checking if update available for 'chodba_3_svetlo_scena'
Debug 2024-01-27 20:32:11OTA: Checking if update available for '0x000d6f001882ac7a' (Zigbee 3.0 DALI CONV LI)
Debug 2024-01-27 20:32:11OTA: Using endpoint '15'
Debug 2024-01-27 20:32:14Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:32:14OTA: Got request '{"fieldControl":1,"manufacturerCode":4364,"imageType":57374,"fileVersion":268723547}'
Debug 2024-01-27 20:32:14OTA: Is new image available for '0x000d6f001882ac7a', current '{"fieldControl":1,"manufacturerCode":4364,"imageType":57374,"fileVersion":268723547}'
Debug 2024-01-27 20:32:14LedvanceOTA: call getImageMeta for Zigbee 3.0 DALI CONV LI
Info 2024-01-27 20:32:14MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"chodba_3_svetlo_scena"},"error":"Failed to check if update available for 'chodba_3_svetlo_scena' (Cannot read properties of null (reading '1'))","status":"error","transaction":"muh28-1"}'
Error 2024-01-27 20:32:14Failed to check if update available for 'chodba_3_svetlo_scena' (Cannot read properties of null (reading '1'))
Debug 2024-01-27 20:32:14TypeError: Cannot read properties of null (reading '1') at getImageMeta (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/ledvance.ts:25:55) at processTicksAndRejections (node:internal/process/task_queues:95:5) at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:398:18) at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:387:29) at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:205:45) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17)
Debug 2024-01-27 20:32:17Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:32:17Device 'chodba_3_svetlo_scena' requested OTA
Debug 2024-01-27 20:32:19Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:32:19Device 'chodba_3_svetlo_scena' requested OTA
Debug 2024-01-27 20:32:22Received Zigbee message from 'chodba_3_svetlo_scena', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268723547,"imageType":57374,"manufacturerCode":4364}' from endpoint 15 with groupID 0
Debug 2024-01-27 20:32:22Device 'chodba_3_svetlo_scena' requested OTA
```

Hadatko, 2024-01-27:
In log above is present also error which happen when i try manually check update

Koenkk, 2024-01-28:
Can you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

closed 2024-08-26 by github-actions[bot]