Koenkk/zigbee2mqtt #21077: Nodon_SIN-4-RS-20_ Calibration

moshiko2312, 2024-01-28:
### Is your feature request related to a problem? Please describe

hi I got the parameter calibration for the SIN-4-RS-20, can you please add this parameter

### Describe the solution you'd like

[image]

### Describe alternatives you've considered

[image]

### Additional context

Thanks

lippertmarkus, 2024-06-11:
Implemented with https://github.com/Koenkk/zigbee-herdsman-converters/pull/7547

closed 2024-06-11 by Koenkk

acasajus, 2024-10-14:
Where did you get the protocol specs? I'd like to take a look 👁️

moshiko2312, 2024-10-17:
From the nodon company